Koenkk/zigbee2mqtt #2578: Working on adding LIGHTIFY Dimming Switch

airdrummingfool, 2019-12-21:
Hi Koen,
I'm working on adding the `LIGHTIFY Dimming Switch`, which has two buttons that can be pressed or held: up/on and down/off.

Press on/up:
```
Received Zigbee message from '0x8418260000e8e507', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
No converter available for '3130' with cluster 'genOnOff' and type 'commandOn' and data '{}'
```

Hold up:
```
Received Zigbee message from '0x8418260000e8e507', type 'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":0,"rate":50}' from endpoint 1 with groupID 0
No converter available for '3130' with cluster 'genLevelCtrl' and type 'commandMoveWithOnOff' and data '{"movemode":0,"rate":50}'
[let go]
Received Zigbee message from '0x8418260000e8e507', type 'commandStop', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 0
No converter available for '3130' with cluster 'genLevelCtrl' and type 'commandStop' and data '{}'
```

Press off/down:
```
Received Zigbee message from '0x8418260000e8e507', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
No converter available for '3130' with cluster 'genOnOff' and type 'commandOff' and data '{}'
```

hold down:
```
Received Zigbee message from '0x8418260000e8e507', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":1,"rate":50}' from endpoint 1 with groupID 0
No converter available for '3130' with cluster 'genLevelCtrl' and type 'commandMove' and data '{"movemode":1,"rate":50}'
[let go]
Received Zigbee message from '0x8418260000e8e507', type 'commandStop', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 0
No converter available for '3130' with cluster 'genLevelCtrl' and type 'commandStop' and data '{}'
```

I haven't worked with dimming switches in Zigbee2mqtt before, so I'm not sure the correct way to set them up. With what I've come up with so far, holding the buttons down seems to hang HomeAssistant; the Docker logs print a bunch of these:
```
homeassistant_1  | 2019-12-20 23:39:41 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.139998328491664] Client exceeded max pending messages [2]: 512
```
Eventually HA seems to recover and continue on.

Here's my first stab at the `devices.js` entry.
```
zigbeeModel: ['LIGHTIFY Dimming Switch'],
model: '73743',
vendor: 'Sylvania',
description: 'Lightify OSRAM Dimming Switch',
supports: 'on/off, brightness',
fromZigbee: [
    fz.genOnOff_cmdOn, fz.genOnOff_cmdOff, fz.cmd_move_with_onoff,
    fz.cmd_move, fz.cmd_stop, fz.temperature, fz.battery_3V],
toZigbee: [tz.light_onoff_brightness],
meta: {configureKey: 1},
configure: async (device, coordinatorEndpoint) => {
    const endpoint = device.getEndpoint(1);
    await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'genPowerCfg']);
    await configureReporting.onOff(endpoint);
    await configureReporting.brightness(endpoint);
},
```

I've set up a `cfg.light_brightness` entity in `lib/extension/homeassistant.js`, but I'm not sure if this is the correct way to represent brightness and on/off from the dimmer.

Thanks for any help/guidance you can privide!

airdrummingfool, 2019-12-21:
(this relates to #2279 )

Koenkk, 2019-12-23:
- I assume that this is the device: https://www.amazon.com/Sylvania-Smart-Home-73743-Lightify/dp/B0196M620Y
- Integration is comparable to https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L2140
- `fromZigbee` should be empty
- All converters that you use should return a payload with `action`.
- You can use this home assistant config: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js#L637

airdrummingfool, 2020-02-22:
Whoops. This is still on my list to do soon, so let's keep this issue open for now.

airdrummingfool, 2020-02-25:
Fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/1007

closed 2020-02-25 by airdrummingfool