Koenkk/zigbee2mqtt #2579: Eurotronics Spirit setpoint timeout

TASSDevon, 2019-12-21:
# Bug Report
## What happened
After a few minutes (I'm not sure how long, could be 30min or so?), When changing the temperature setpoint, the following error occurs:

`zigbee2mqtt:error 2019-12-21T08:07:53: Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 14 - 513 - 4 after 10000ms'
Dec 21 08:07:53 beaglebone npm[20327]: zigbee2mqtt:info  2019-12-21T08:07:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 14 - 513 - 4 after 10000ms'","meta":{"friendly_name":"0x00158d0001ffc997"}}'`

## What did you expect to happen
The setpoint should be updated

## How to reproduce it (minimal and precise)
Pair Eurotronics Spirit with network, changing setpoint works for a while. After an hour or so, try changing the setpoint, it fails with error above.

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: 
{'type': 'zStack12', 'meta': {'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190425}}

The device is routed through an Ikea Tradfi outlet when looking at the network map.

Sensor info:
`local_temperature: 18
occupied_heating_setpoint: 18.5
unoccupied_heating_setpoint: 16
pi_heating_demand: 25
linkquality: 36
current_heating_setpoint: 18.5
eurotronic_error_status: 0
eurotronic_system_mode: 1
system_mode: heat
unit_of_measurement: -
friendly_name: Radiator Veranda Link Quality`

CodeFinder2, 2019-12-24:
I am using 5 such thermostats and never had such issues (over a "not so long range"). None of my devices go to a deep sleep (althought it takes about up to 30sec, typically just a few seconds) until updates are propagated.

Just a guess, but it may be related to bad link quality or too much load on the network. Or probably also due to frequency collisions with other bands (WIFI, bluetooth).

Is it the only device with such issues? Are there other devices nearby (possibly connected with the same/similar sequence of routers) not having such problems?

How much time passes by until you see a change in Z2M when you change the temperature at the device itself?

TASSDevon, 2019-12-24:
I moved the Zigbee channel away from the used WiFi channels (yes I know the channel nrs are not the same :-D)

>>> Is it the only device with such issues? Are there other devices nearby (possibly connected with the same/similar sequence of routers) not having such problems?
Yes, all the other devices (Ikea and Xiaomi) never give this timeout error.

>>> How much time passes by until you see a change in Z2M when you change the temperature at the device itself?

It depends, if this timeout occurs, the setpoint is never changed, hence why I've created a loop to set and check until the setpoint reported is the same as what I have set it to. This morning it was particularly 'hard' apparently to change it, here's a log output:

`Dec 24 08:32:48 beaglebone npm[20327]: zigbee2mqtt:error 2019-12-24T08:32:48: Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 179 - 513 - 4 after 10000ms'
Dec 24 08:33:00 beaglebone npm[20327]: zigbee2mqtt:error 2019-12-24T08:33:00: Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 180 - 513 - 4 after 10000ms'
Dec 24 08:33:12 beaglebone npm[20327]: zigbee2mqtt:error 2019-12-24T08:33:12: Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 181 - 513 - 4 after 10000ms'
Dec 24 08:33:23 beaglebone npm[20327]: zigbee2mqtt:error 2019-12-24T08:33:23: Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 182 - 513 - 4 after 10000ms'
Dec 24 08:33:35 beaglebone npm[20327]: zigbee2mqtt:error 2019-12-24T08:33:35: Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 183 - 513 - 4 after 10000ms'
Dec 24 08:33:47 beaglebone npm[20327]: zigbee2mqtt:error 2019-12-24T08:33:47: Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 184 - 513 - 4 after 10000ms'
Dec 24 08:33:59 beaglebone npm[20327]: zigbee2mqtt:error 2019-12-24T08:33:59: Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 185 - 513 - 4 after 10000ms'
Dec 24 08:34:11 beaglebone npm[20327]: zigbee2mqtt:error 2019-12-24T08:34:11: Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 186 - 513 - 4 after 10000ms'
Dec 24 08:34:23 beaglebone npm[20327]: zigbee2mqtt:error 2019-12-24T08:34:23: Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 187 - 513 - 4 after 10000ms'
Dec 24 08:34:35 beaglebone npm[20327]: zigbee2mqtt:error 2019-12-24T08:34:35: Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 188 - 513 - 4 after 10000ms'

Dec 24 08:35:40 beaglebone npm[20327]: zigbee2mqtt:error 2019-12-24T08:35:40: Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 189 - 513 - 4 after 10000ms'
Dec 24 08:35:52 beaglebone npm[20327]: zigbee2mqtt:error 2019-12-24T08:35:52: Publish 'set' 'current_heating_setpoint' to '0x00158d0001ffc997' failed: 'Error: Timeout - 38687 - 1 - 190 - 513 - 4 after 10000ms'`

zeegbi, 2019-12-25:
I have three Eurotronics Spirit Zigbee thermostats, and I keep receiving the same error on one of the devices. The other 2 are working just fine since the first day. I even swapped their places, the problematic one keeps having the same problem whatever room it is in.

Using zigbee2mqtt version: 1.8.0

TASSDevon, 2019-12-25:
I've updated to 1.8.0 yesterday but the problem keeps occurring (this morning, same story).
Did not test the second valve yet though. So interestingly, it doesn't matter what the link quality is? Was the path it was taking to the coordinator different or the same (if any routers were used).

zeegbi, 2019-12-25:
Yes, my problematic one now have better link quality than the others, but still no change. When I changed the rooms, and re-added it to network, it connected to a different router but still no change in the behavior.

TASSDevon, 2020-01-02:
I've recently also had the exact same behavior with one of my Ikea Tradfri outlets, and at the exact same day, two of my xiaomi sensors who had been going strong for months dropped off the network completely.

I'm starting to suspect I've reached the 'unstable' situation with the CC2531 and that the valve was just the figurative drop of water that spills the bucket. I've bought one of those TI dev boards and will check if the situation is better when using this.

closed 2020-03-10 by stale[bot]