Koenkk/zigbee2mqtt #2582: WXCJKG13LM directly connects without gateway

matlab22, 2019-12-21:
I ordered a WXCJKG13LM 6 way zigbee switch and played around with the buttons. i did not start to pair it, but somehow it controls my other xiaomi devices. This is very odd, because I was running zigbee2mqtt 1.7.1 and switched the channel and used a custom key and pan ID. Now I updated to 1.8.0 and pared it properly. However the switch is still able to directly communicate to the other devices. It can switch on and off them even if I stopped the zigbee-herdsman.
This is a bit scary. How can I avoid it?

matlab22, 2019-12-21:
I think a was able to fix it. While starting the herdsmen I pressed the button on the back of the switch, so it could configure it.
Is there a way to secure the network, so it can not be hijacked by an unauthorized device?

snorro80, 2019-12-27:
I've got the same problem as matlab22. The switch is sending out commands to my other zigbee devices bypassing the router in Domoticz.
No luck in fixing it through matlab22's solution.

matlab22, 2019-12-27:
Were you able to pair it first?
I stopped the service and restarted it, then I could observe that zigbee2mqqtt was trying to configuration the switch. (sudo journalctl -u zigbee2mqtt.service -f)
While it was starting I pressed the button on the back several times, so the switch will not go to sleep

snorro80, 2019-12-29:
Tried it again and managed to get it working without bypassing the router. Thanks matlab22!
Hopefully the press and release function will work soon in Domoticz.

closed 2020-03-05 by stale[bot]