Koenkk/zigbee2mqtt #2583: Force router to be link to an other router?

Bastiencc, 2019-12-21:
Hello !

Sometimes my network map look great all routers (bulbs) are conneted to each others router and to the coordinator, but sometimes (most of time, indeed) there are only connected to the coordinator...

My zigbee network seems to works better when all routers are connected to others routers and to the coordinator.

Is it possible to force router to be connected to each other router ? or maybe is it possible to "rebuild" the network map ?

thanks

zhapwey, 2019-12-21:
My rooters always connect to all rooters and the coordinator they have visible. and create the shortest route for the end devices (this more or less). what do you use as rooter which version of z2m?

Bastiencc, 2019-12-22:
I use z2m 1.8.0 and cc2531 version 20190608 .
my 3 routers are ZNLDP12LM - Xiaomi Aqara smart LED bulb

my map look currently at this , but sometime routers have link to each others:
https://ibb.co/NCFjdzm

closed 2020-02-27 by stale[bot]