Koenkk/zigbee2mqtt #2586: Option to add timestamp to MQTT-Message

J0s3f, 2019-12-21:
# Bug Report
## What happened

MQTT doesn't contain timestamps in the protocol. So if a message is sent with QoS 1 or QoS 2, cached at the message broker and retrieved at a later time, it is impossible to get the initial time it was sent.

## What did you expect to happen
Have an option to add a timestamp to each message (current timestamp from the system running zigbee2mqtt)

## How to reproduce it (minimal and precise)

1. Connect a client with [QoS](https://www.npmjs.com/package/mqtt#about-qos) > 0 and [Persistent Session](https://www.hivemq.com/blog/mqtt-essentials-part-7-persistent-session-queuing-messages/), subscribe to a topic that gets sensor data from zigbee2mqtt
2. disconnect client 
3. configure zigbee2mqtt with QoS>0, start zigbee2mqtt, wait until at least one message is sent to the topic subscribed in 1.
4. reconnect client
5. receive message
6. try to find out the time for the received sensor values
7. you can't

My proposal would be to just add the current unixtime in UTC from the host to each message as JSON property and make this configurable. 

Usecase for me is this: I have some zigbee sensors at home and collect all data of temp/humidity/rain/... sensors in InfluxDB. 

My parents now also have some zigbee stuff setup, with own MQTT broker.
I want to store at least the outside weather data from their place in my InfluxDB. If everything works, no problem. But if the internet connection or VPN fails, I am missing data. The MQTT broker stores this data, but it is useless if received later. (The sensors send based on changes, so if connection was down for 10 hours and there are 6 messages with different temperatures, it is impossible currently to find out which time this temperatures have been measured)

pcb1962, 2019-12-24:
You can use the last_seen option
https://www.zigbee2mqtt.io/information/configuration.html

J0s3f, 2019-12-26:
> You can use the last_seen option
> https://www.zigbee2mqtt.io/information/configuration.html

This seems to be exactly what I need, thanks!

closed 2019-12-26 by J0s3f