Koenkk/zigbee2mqtt #2589: Failing to create new device converter due to carriage return in the name

m4tek, 2019-12-22:
# Bug Report
## What happened
Trying to add support for Saswell TRV SEA801-ZIGBEE. Added definition in the converters file, but it is not being matched due to (apparently) device reporting carriage return in its name

Device reporting as:

Dec 22 11:31:47 rpi3 npm[18640]: zigbee2mqtt:warn  2019-12-22T11:31:47: Received message from unsupported device with Zigbee model 'GbxAXL2
Dec 22 11:31:47 rpi3 npm[18640]: '
Dec 22 11:31:47 rpi3 npm[18640]: zigbee2mqtt:warn  2019-12-22T11:31:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

## What did you expect to happen
Converter entry in the devices.js to be matched against the new device

## How to reproduce it (minimal and precise)
diff devices.js devices.js.save 
5571,5597d5570
< 
<     //SASWELL
<     {
<         zigbeeModel: ['TRV002'],
<         model: 'GbxAXL2',
<         vendor: 'SASWELL',
<         description: 'Radiator valve',
<         supports: 'temperature',
<         fromZigbee: [fz.thermostat_att_report, fz.battery_percentage_remaining],
<         toZigbee: [
<             tz.thermostat_occupied_heating_setpoint, tz.thermostat_local_temperature_calibration,
<             tz.thermostat_setpoint_raise_lower, tz.thermostat_remote_sensing,
<             tz.thermostat_system_mode, tz.thermostat_running_state,
<         ],
< //        meta: {configureKey: 1},
< //        configure: async (device, coordinatorEndpoint) => {
< //            const endpoint = device.getEndpoint(1);
< //            await bind(endpoint, coordinatorEndpoint, [
< //                'genBasic', 'genPowerCfg', 'genIdentify', 'genTime', 'genPollCtrl', 'hvacThermostat',
< //                'hvarUserInterfaceCfg',
< //            ]);
< //            await configureReporting.thermostatTemperature(endpoint);
< //            await configureReporting.thermostatOccupiedHeatingSetpoint(endpoint);
< //            await configureReporting.thermostatPIHeatingDemand(endpoint);
< //        },
<     },
<  

## Debug Info
zigbee2mqtt version: 1.7.1 (commit #b459c35)
CC253X firmware version: 20190619

Koenkk, 2019-12-24:
How does your `database.db` look like?

m4tek, 2019-12-24:
Hey, 

dump of my DB below:

`{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e1a68d","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}},"binds":[]}},"meta":{}}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d000183cb25","nwkAddr":27702,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_86sw1\u0000lu","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,18,25,65535],"outClusterList":[0,3,4,5,18,25,65535],"clusters":{"genBasic":{"dir":{"value":3},"attributes":{}},"genIdentify":{"dir":{"value":3},"attributes":{}},"genGroups":{"dir":{"value":2},"attributes":{}},"genScenes":{"dir":{"value":2},"attributes":{}},"genMultistateInput":{"dir":{"value":3},"attributes":{}},"genOta":{"dir":{"value":3},"attributes":{}},"manuSpecificCluster":{"dir":{"value":3},"attributes":{}}},"binds":[]},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClusterList":[3,4,5,18],"clusters":{"genIdentify":{"dir":{"value":3},"attributes":{}},"genGroups":{"dir":{"value":2},"attributes":{}},"genScenes":{"dir":{"value":2},"attributes":{}},"genMultistateInput":{"dir":{"value":3},"attributes":{}}},"binds":[]},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[3,4,5,12],"clusters":{"genIdentify":{"dir":{"value":3},"attributes":{}},"genGroups":{"dir":{"value":2},"attributes":{}},"genScenes":{"dir":{"value":2},"attributes":{}},"genAnalogInput":{"dir":{"value":3},"attributes":{}}},"binds":[]}},"meta":{}}
{"id":3,"type":"Router","ieeeAddr":"0x000b57fffebb88d0","nwkAddr":56747,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 W opal 1000lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E27 W opal 1000lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214"}}},"binds":[]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214","zclVersion":1,"interviewCompleted":true,"meta":{}}
{"id":4,"type":"Router","ieeeAddr":"0x000b57fffeeb2327","nwkAddr":58476,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 W opal 1000lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attributes":{"10":{},"65533":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI bulb E27 W opal 1000lm","dateCode":"20170302","powerSource":1}},"genIdentify":{"dir":{"value":1},"attributes":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":1},"attributes":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":3},"attributes":{"65533":1,"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attributes":{"16387":1,"65533":1,"onOff":0,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attributes":{"15":0,"16384":254,"65533":1,"currentLevel":254,"remainingTime":0,"onOffTransitionTime":5,"onLevel":255}},"genOta":{"dir":{"value":2},"attributes":{}},"genPollCtrl":{"dir":{"value":2},"attributes":{}},"haDiagnostic":{"dir":{"value":1},"attributes":{"65533":1,"numberOfResets":0,"macRxBcast":0,"macTxBcast":0,"aPSRxBcast":0,"aPSTxBcast":0,"nwkFcFailure":0,"packetBufferAllocateFailures":0,"averageMacRetryPerApsMessageSent":14,"lastMessageLqi":52,"lastMessageRssi":-87}},"lightLink":{"dir":{"value":3},"attributes":{"65533":1}}},"binds":[]}},"meta":{}}
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d00015f1f66","nwkAddr":39880,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_86sw1\u0000lu","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,18,25,65535],"outClusterList":[0,3,4,5,18,25,65535],"clusters":{"genBasic":{"dir":{"value":3},"attributes":{"65281":{"1":2985,"3":29,"4":5032,"5":77,"6":[1,16777226],"10":62180},"modelId":"lumi.sensor_86sw1","appVersion":4}},"genIdentify":{"dir":{"value":3},"attributes":{}},"genGroups":{"dir":{"value":2},"attributes":{}},"genScenes":{"dir":{"value":2},"attributes":{}},"genMultistateInput":{"dir":{"value":3},"attributes":{}},"genOta":{"dir":{"value":3},"attributes":{}},"manuSpecificCluster":{"dir":{"value":3},"attributes":{}},"genOnOff":{"attributes":{"o [... cut ...] {}},"genGroups":{"dir":{"value":2},"attributes":{}},"msTemperatureMeasurement":{"dir":{"value":1},"attributes":{"measuredValue":2266}},"msPressureMeasurement":{"dir":{"value":1},"attributes":{"16":9912,"20":-1,"measuredValue":991}},"msRelativeHumidity":{"dir":{"value":1},"attributes":{"measuredValue":6623}},"manuSpecificCluster":{"dir":{"value":3},"attributes":{}}},"binds":[]}},"appVersion":3,"stackVersion":2,"hwVersion":30,"dateCode":"20161129","swBuildId":"3000-0001","zclVersion":1,"meta":{}}`

m4tek, 2019-12-24:
I guess I've answered my question. Should I use "GbxAXL2\u0000" a model string?

Koenkk, 2019-12-26:
Yes!

closed 2019-12-29 by m4tek

rotragit, 2020-01-11:
@m4tek have you integrated the Saswell TRV SEA801-ZIGBEE? I have the same valves and I would like to know if you get the point before trying to do the job. Thank you :-)

m4tek, 2020-01-19:
@rotragit Heya, sorry I was preoccupied with other things and I did not have time to look here recently.
Frankly there was no progress, as I've found this device to communicate using hex words via a custom cluster (I have number {if not all} of messages coming from the TRV captured).

I tried to get some support on discord DEV channel on how to proceed, but unfortunately nobody was able to give me any hint.

I will open another issue to build this device support.

m4tek, 2020-01-19:
Opened #2781 to discuss further development of support.