Koenkk/zigbee2mqtt #2590: Failed to call 'BridgeConfig' 'onMQTTMessage' [...] failed with status '17' (expected '0')

Juweii, 2019-12-22:
# Bug Report
## What happened
Started zigbee2mqtt and got the following error messages (tons of them):

`error 2019-12-22 22:38:03: Failed to call 'BridgeConfig' 'onMQTTMessage' (Error: SREQ '--> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}' failed with status '17' (expected '0')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:245:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58))
`

Also i got tons of those messages right after the start:
`info  2019-12-22 22:38:03: Set last_seen to ISO_8601_local
info  2019-12-22 22:38:03: Set elapsed to false
info  2019-12-22 22:38:03: Zigbee: allowing new devices to join.
[...]
repeated around 40 times....
[...]
info  2019-12-22 22:38:03: Zigbee: allowing new devices to join.
info  2019-12-22 22:38:03: Set last_seen to ISO_8601_local
info  2019-12-22 22:38:03: Set elapsed to false
`
On mqtt server i have exactly those config options in my tree:
- bridge
   - config: see bottom of the ticket
      - devices: '[array of devices]'
      - elapsed: false
      - last_seen: 'ISO_8601_local'
      - permit_join: true
   - group: ''
      - remove_all: ''
   - networkmap: raw
      - remove_all: ''
   - configure: ''
   - log: '{last log}'
   - state: 'online'
- 0x00....device1
- 0x00....device2
- 0x00....device3

It seems on start every message is repeatet 40 times... Feels like it starts 40 instances of zigbee2mqtt at the same time.

## What did you expect to happen
No errors and if an error occured, just one time and not dozens of the same ones... ;)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:
{"version":"1.8.0","commit":"da4d26a","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190315}},"log_level":"info","permit_join":true}

Koenkk, 2019-12-26:
Error 17 means the CC2531 is out of memory, try upgrading to https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

Juweii, 2019-12-27:
Hello @Koenkk, thank you, will try as soon as i get the chance to do it. Thank you!

closed 2020-03-05 by stale[bot]