Koenkk/zigbee2mqtt #2591: Error after upgrading to 1.8.0

renewoensdregt, 2019-12-23:
# Bug Report
## What happened
Running the addon in Hassio. All runs fine on 1.7.1. After upgrading to 1.8.0 I get the following errors in the logs:

info  2019-12-23 10:31:48: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190109}}'
error 2019-12-23 10:31:51: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'ieeeAddr' of null
    at DeviceReceive.canHandleEvent (/zigbee2mqtt-1.8.0/lib/extension/deviceReceive.js:62:55)
    at DeviceReceive.onZigbeeEvent (/zigbee2mqtt-1.8.0/lib/extension/deviceReceive.js:101:19)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.8.0/lib/controller.js:322:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.8.0/lib/controller.js:226:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/zigbee2mqtt-1.8.0/lib/zigbee.js:57:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/zigbee2mqtt-1.8.0/node_modules/zigbee-herdsman/dist/controller/controller.js:446:22)
    at Generator.next (<anonymous>)
    at /zigbee2mqtt-1.8.0/node_modules/zigbee-herdsman/dist/controller/controller.js:8:71)
error 2019-12-23 10:31:51: Not connected to MQTT server!

and Zigbee does not start up, see log.
When rolling back to 1.7.1, all works again.

## What did you expect to happen

## How to reproduce it (minimal and precise)
upgrade to 1.8.0 (tried several times, even removing addon and re-installing.

## Debug Info
zigbee2mqtt version: 1.8.0
CC2530 firmware version: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190109}}'

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3994402/log.txt)

Koenkk, 2019-12-26:
Can you share the log when running with log level debug?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

renewoensdregt, 2019-12-27:
Hi Koen,
First of all thanks for your time and efforts! Really appreciate your work!
See below the debug log.
on a sidenote, I am confused about the configuration.yaml. Is that still needed or is it part of the configuration of the Hassio addon? to me they look the same, but I updated in both and I see in the log that debug mode was activated as I see debug lines.

`./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2019-12-27T09:33:57: PM2 log: Launching in no daemon mode
2019-12-27T09:33:58: PM2 log: App [npm:0] starting in -fork mode-
2019-12-27T09:33:58: PM2 log: App [npm:0] online
> zigbee2mqtt@1.8.0 start /zigbee2mqtt-1.8.0
> node index.js
zigbee2mqtt:info  2019-12-27 09:34:04: Logging to console and directory: '/share/zigbee2mqtt/log/2019-12-27.09-34-04'
zigbee2mqtt:debug 2019-12-27 09:34:05: Loaded state from file /share/zigbee2mqtt/state.json
zigbee2mqtt:info  2019-12-27 09:34:05: Starting zigbee2mqtt version 1.8.0 (commit #unknown)
zigbee2mqtt:info  2019-12-27 09:34:05: Starting zigbee-herdsman...
zigbee2mqtt:debug 2019-12-27 09:34:05: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/ttyUSB0"}}'
zigbee2mqtt:info  2019-12-27 09:34:07: zigbee-herdsman started
zigbee2mqtt:info  2019-12-27 09:34:07: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190109}}'
zigbee2mqtt:debug 2019-12-27 09:34:07: Zigbee network parameters: {"panID":6755,"extendedPanID":"0xdddddddddddddddd","channel":11}
zigbee2mqtt:debug 2019-12-27 09:34:07: Received Zigbee message from '0x00158d00024f2d8e', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":0}' from endpoint 2 with groupID 0
zigbee2mqtt:error 2019-12-27 09:34:07: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'ieeeAddr' of null
    at DeviceReceive.canHandleEvent (/zigbee2mqtt-1.8.0/lib/extension/deviceReceive.js:62:55)
    at DeviceReceive.onZigbeeEvent (/zigbee2mqtt-1.8.0/lib/extension/deviceReceive.js:101:19)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.8.0/lib/controller.js:322:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.8.0/lib/controller.js:226:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/zigbee2mqtt-1.8.0/lib/zigbee.js:57:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/zigbee2mqtt-1.8.0/node_modules/zigbee-herdsman/dist/controller/controller.js:446:22)
    at Generator.next (<anonymous>)
    at /zigbee2mqtt-1.8.0/node_modules/zigbee-herdsman/dist/controller/controller.js:8:71)
zigbee2mqtt:error 2019-12-27 09:34:07: Not connected to MQTT server!
zigbee2mqtt:error 2019-12-27 09:34:07: Cannot send message: topic: 'homeassistant/switch/0x00158d00024f2d8e/switch/config', payload: '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x00158d00024f2d8e/set","state_topic":"zigbee2mqtt/0x00158d00024f2d8e","json_attributes_topic":"zigbee2mqtt/0x00158d00024f2d8e","name":"0x00158d00024f2d8e_switch","unique_id":"0x00158d00024f2d8e_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00024f2d8e"],"name":"0x00158d00024f2d8e","sw_version":"Zigbee2mqtt 1.8.0","model":"Mi power plug ZigBee (ZNCZ02LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}
zigbee2mqtt:error 2019-12-27 09:34:07: Not connected to MQTT server!
zigbee2mqtt:error 2019-12-27 09:34:07: Cannot send message: topic: 'homeassistant/sensor/0x00158d00024f2d8e/power/config', payload: '{"unit_of_measurement":"W","icon":"mdi:factory","value_template":"{{ value_json.power }}","state_topic":"zigbee2mqtt/0x00158d00024f2d8e","json_attributes_topic":"zigbee2mqtt/0x00158d00024f2d8e","name":"0x00158d00024f2d8e_power","unique_id":"0x00158d00024f2d8e_power_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00024f2d8e"],"name":"0x00158d00024f2d8e","sw_version":"Zigbee2mqtt 1.8.0","model":"Mi power plug ZigBee (ZNCZ02LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}
zigbee2mqtt:error 2019-12-27 09:34:07: Not connected to MQTT server!
zigbee2mqtt:error 2019-12-27 09:34:07: Cannot send message: topic: 'homeassistant/sensor/0x00158d00024f2d8e/linkquality/config', payload: '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d00024f2d8e","json_attributes_topic":"zigbee2mqtt/0x00158d00024f2d8e","name":"0x00158d00024f2d8e_linkquality","unique_id":"0x00158d00024f2d8e_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00024f2d8e"],"name":"0x00158d00024f2d8e","sw_version":"Zigbee2mqtt 1.8.0","model":"Mi power plug ZigBee (ZNCZ02LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}
zigbee2mqtt:error 2019-12-27 09:34:13: Failed to start zigbee
zigbee2mqtt:error 2019-12-27 09:34:13: Exiting...
zigbee2mqtt:error 2019-12-27 09:34:13: Error: SRSP - UTIL - ledControl after 6000ms
[... cut ...] :"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/ttyUSB0"}}'
zigbee2mqtt:info  2019-12-27 09:34:37: zigbee-herdsman started
zigbee2mqtt:info  2019-12-27 09:34:37: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190109}}'
zigbee2mqtt:debug 2019-12-27 09:34:37: Zigbee network parameters: {"panID":6755,"extendedPanID":"0xdddddddddddddddd","channel":11}`

renewoensdregt, 2019-12-27:
Just FYI, I leave a debug session from the restored 1.7.1 version where we see all runs fine. 
the main difference I see is in the beginning...in 1.8.0 it states SOCAT is DISABLED and not started which does not appear in 1.7.1. 

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4005332/log.txt)

Koenkk, 2019-12-27:
Thanks, can you also share your database.db and configuration.yaml? (Dont forgot to remove urls and passwords from it when posting).

renewoensdregt, 2019-12-27:
Hi Koen,
Attached 2 files : one is the zigbee2mqtt folder pre upgrade (when all is working) and one post upgrade to 1.8.0 when I have the error ( i have replaced user and pw).

[zigbee2mqtt - pre upgrade.zip](https://github.com/Koenkk/zigbee2mqtt/files/4006040/zigbee2mqtt.-.pre.upgrade.zip)
[zigbee2mqtt - post upgrade.zip](https://github.com/Koenkk/zigbee2mqtt/files/4006041/zigbee2mqtt.-.post.upgrade.zip)

The only confusing thing : In the addon in Hassio there are settings I do not see in the configuration.yaml (socat related), so I refer again to this file which are the hassio addon config settings  : https://github.com/Koenkk/zigbee2mqtt/files/3994411/addon.parameters.txt

Koenkk, 2020-01-03:
Sorry for taking so long, but you issue should be fixed now in the latest dev branch (use edge for hassio), can you confirm?

renewoensdregt, 2020-01-04:
Sorry Koen, but that did not seem to help. 
See below and attached the same files you asked before (user and pw removed):
[zigbee2mqtt - edge.zip](https://github.com/Koenkk/zigbee2mqtt/files/4021702/zigbee2mqtt.-.edge.zip)
`Debug information

-----
-----
{
  "data_path": "/share/zigbee2mqtt",
  "devices": "[]",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://homeassistant",
    "user": "xxxxxx",
    "password": "xxxxxx"
  },
  "serial": {
    "port": "/dev/ttyUSB0",
    "disable_led": true
  },
  "advanced": {
    "log_level": "debug",
    "rtscts": false,
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blacklist": []
  },
  "ban": [],
  "whitelist": [],
  "queue": {},
  "socat": {
    "enabled": false,
    "master": "pty,raw,echo=0,link=/dev/ttyZ2M,mode=777",
    "slave": "tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5",
    "restartdelay": 1.0,
    "initialdelay": 1.0,
    "options": "-d -d",
    "log": false
  }
./run.sh: line 24: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
}Debug information end
[Info] Socat is DISABLED and not started
2020-01-04T09:40:32: PM2 log: Launching in no daemon mode
2020-01-04T09:40:33: PM2 log: App [npm:0] starting in -fork mode-
2020-01-04T09:40:33: PM2 log: App [npm:0] online
> zigbee2mqtt@1.8.0-dev start /app
> node index.js
zigbee2mqtt:info  2020-01-04 09:40:38: Logging to console and directory: '/share/zigbee2mqtt/log/2020-01-04.09-40-38'
zigbee2mqtt:debug 2020-01-04 09:40:39: Loaded state from file /share/zigbee2mqtt/state.json
zigbee2mqtt:info  2020-01-04 09:40:39: Starting zigbee2mqtt version 1.8.0-dev (commit #2451d536f2c9e446d396df55c9493f256ab9352e)
zigbee2mqtt:info  2020-01-04 09:40:39: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-01-04 09:40:39: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/ttyUSB0"}}'
zigbee2mqtt:info  2020-01-04 09:40:41: zigbee-herdsman started
zigbee2mqtt:info  2020-01-04 09:40:41: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190109}}'
zigbee2mqtt:debug 2020-01-04 09:40:41: Zigbee network parameters: {"panID":6755,"extendedPanID":"0xdddddddddddddddd","channel":11}
zigbee2mqtt:error 2020-01-04 09:40:47: Failed to start zigbee
zigbee2mqtt:error 2020-01-04 09:40:47: Exiting...
zigbee2mqtt:error 2020-01-04 09:40:47: Error: SRSP - UTIL - ledControl after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.8.0-dev start: `node index.js`
npm ERR!
 Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.8.0-dev start script.
npm ERR!
 This is probably not a problem with npm. There is likely additional logging output above.`

Koenkk, 2020-01-07:
This looks like a different issue, can you set `disable_led` to `false` in the configuration?

renewoensdregt, 2020-01-07:
Thanks Koen! That resolved it! 
Maybe because the cc2390 doesn´t have a led it needs that setting. It used to work, but for some reason stopped!
Thanks again. Just transferred some to your Paypal! Enjoy the beer or Coffee!

closed 2020-01-07 by renewoensdregt

Koenkk, 2020-01-07:
@renewoensdregt great, thanks!

Arminhosseini, 2020-01-15:
Hi,
I have the same problem (the first one) and did your changes in controller.js but still, I have the problem I created a new issue that I attached my log file there.
#2741