Koenkk/zigbee2mqtt #2593: Unable to repair aqara door sensor

jnavarrodev, 2019-12-23:
I wanted to fix my zigbee map because a lot of devices appears with no connections with any router/coordinator so I read that I had to repair them.

I began with an aqara window/door sensor. I deleted from z2m, then reset the device (long press to the button) and then, with pair mode enable, press again the button for 5 seconds and pair them.

The problem is that the device is not going to pair mode anymore (or that is what I think). If I keep the button press, the led starts blinking, but as soon as I stop, the led does the same. I put it near the coordinator, but just nothing happens. I try a lot of times, after resetting z2m, remove the battery from the device for a while, etc...but nothing works. 

Any suggestion? What is wrong with the device?

davet2001, 2019-12-23:
I have the same door sensors.

Based on your description, you may need to remove the device from zigbee2mqtt before you can re-pair it.

https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

jnavarrodev, 2019-12-24:
Yes, I remove it after trying to re-pair. Something strange  was that I could not delete it, and I had to do a force-remove. But the device was working fine and responding when opening and closing the windows.

CodeFinder2, 2019-12-26:
So if this is solved, you should close your issue. ;-)

jnavarrodev, 2019-12-29:
I could not solve it. I paired the device with the xiaomi gateway and works fine. But with zigbee2mqtt it is impossible. 
I also tried to create a new instance of my docker z2m service with everything empty, and also did not work.

davet2001, 2020-01-06:
In several cases for me, the only solution seems to be:

- Stop zigbee2mqtt.
- Delete the entry from database.db
- Unplug and replug the CC2531.
- Press the reset button on the CC2531.
- Start zigbee2mqtt again.

(I don't know whether all of the above are needed.)  I am not fully sure but it seems the USB device retains some information about the network until you power cycle it.

After that, the network should have fully 'forgotten' about the device, and see it as a new device when you start the pairing process again.

You can see this by monitoring the log with journalctl -f -u zigbee2mqtt.service or similar.

jnavarrodev, 2020-01-12:
I was gonna try something you say but I had not test yet: To press de reset button. Then, I realized that my cc2530 + cc2591 has no reset button... :(
So I just keep it unplug for a couple minutes, and then try again...and it worked! I am not sure why, but I was able to pair it again. 
So lesson learned. Not unpair a device again, ever.

Thanks!

closed 2020-01-12 by jnavarrodev

OzGav, 2021-05-25:
I just want to add a note for future finders of this issue. I also had the issue with the blue LED blinking constantly while the button was held and then going our as soon as it was released. In the end I did what @davet2001 suggested above although that did not immediately fix the problem so it may have done nothing BUT then I took out the battery, held the reset button down and whilst still holding it down reinserted the battery. This instantly re-paired it. Hopefully it will stay that way...

zelial, 2024-07-22:
FTR I had this same issue when migrating my 20+ Aqara sensors from deconz to z2m. Very frustrating experience, took 5-10 pairing attempts (with dongle reboots) to get a sensor paired. Then I upgraded my dongle firmware (from EZSP to ember) via https://darkxst.github.io/silabs-firmware-builder/ (and switched z2m to this adapter) and the issue was gone. With all my devices - I had to repair them all. But it worked on first attempt as it should!

famzah, 2024-08-04:
It may be pure luck, but I was able to pair my sensor after I enabled [debug logging](https://www.zigbee2mqtt.io/guide/configuration/logging.html#debugging) for zigbee2mqtt (and restarted it).

Before that I tried all other suggestions here. FTR, the sensor had been working for a long time but stopped after I rebooted the Linux server for maintenance.

famzah, 2024-08-19:
I was able to re-pair my Aqara temperature/humidity sensor after I changed the battery.

Even though the sensor was able to report data regularly, the battery didn't provide sufficient power for the pairing interview with the coordinator. Once I replaced the battery, it was obvious that the old battery was very weak because with the new one, the blue LED was much stronger and easier to see.

FTR, with the old battery it seemed that the device was about to pair because it was blinking several times as expected but the coordinator picked up nothing.

MIKAspyed, 2025-02-12:
I also couldn't connect my door sensor. The blue LED only blinked when I pressed the button.
I fixed my aqara door sensor this way:
1. Remove battery
2. Open Zigbee2mqtt and click Permit Join
3. Press and hold the button on the sensor and insert the battery

It will also be useful to check if the device is deleted from the file /config/.storage/core.config_entries 
This helped me when I couldn't reconnect the gateway.