Koenkk/zigbee2mqtt #2595: Loosing link quality since update 1.7.1-1.8.0

Ormerod2, 2019-12-23:
#Bug Report 
Since Update 1.71 I keep having  issues with link quality. It was running fine before with good linkquality. Additionally I am having issues with osram plugs as well. Only when close to coordinator, they are running good. 

## What happened
All devices, most Xiaomi, are reporting lower link quality or lost connection 
[image]

## What did you expect to happen
Remaining in the same quality as before

## How to reproduce it (minimal and precise)
Data’s are persistent. 

## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version: Dec 23 20:55:12 openHABianPi npm[22465]: zigbee2mqtt:info  2019-12-23 20:5                                                                        5:12: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version"                                                                        :"1.8.0","commit":"da4d26a","coordinator":{"type":"zStack12","meta":{"tran                                                                        sportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":                                                                        20190515}},"log_level":"info","permit_join":true}'

Additional information: running my own setup of hardware: 
[image]

But haven’t touched hardware for months 

Feel free to close issue if not related to software.

zhapwey, 2019-12-23:
hello could you say if that SHIELD for the adapter is yours or what is it ??

Ormerod2, 2019-12-23:
Yes it is mine. Working on a release version right now. It come pretty handy with an 100mw amplifier. 1 coordinator for the entire house

Ormerod2, 2019-12-24:
Found issue with antenna. Seems my vacuum robot pulled it of a little

Sorry for bordering

closed 2019-12-24 by Ormerod2