Koenkk/zigbee2mqtt #2596: New device creates entity with no friendly name

marcgarciamarti, 2019-12-23:
New device paired successfully. I appended the configuration parts in the main HA configuration.yaml file. The entity created within HA derives from the ID and not from the friendly name, unlike my other 11 devices.

This happened as a result of pairing a brand new Xiaomi ZNCZ02LM

zigbee2mqtt version: 1.8.0
CC253X firmware version: 20190223

marcgarciamarti, 2020-01-01:
paired another Xiaomi ZNCZ02LM and I experience the same behaviour. 

This is my devices.yaml file 

```
~ $ cat /share/zigbee2mqtt/devices.yaml 
'0x000b57fffeda5dae':
  friendly_name: light11000lmikea
  retain: true
'0x00158d00022f35ca':
  friendly_name: plug1xiaomi
  retain: false
'0x00158d0003a51a6d':
  friendly_name: plug2xiaomi
  retain: false

```

and this is what Home Assistant shows for one of the devices working one way, and the other working in a different way

[image]
[image]

marcgarciamarti, 2020-01-06:
@Koenkk any ideas? Any additional information I may provide?

Thanks a million!!

carldebilly, 2020-01-06:
On the first time, Home Assistant will name it automatically using the "friendly name" and will keep that name. If you rename it in Z2M before the first discovery by HASS, you'll end up with that name instead of the 0xXXXX physical address.

By the way, it's safe to change it to what you want in HASS, just remember to update all scripts & configuration using it.

marcgarciamarti, 2020-01-07:
thanks for your answer @carldebilly. I'm not sure I understand 100% though. This is what I usually do when I pair a new zigbee device:
1. I verify that z2m traces indicate the device is paired
2. I edit the configuration.yaml file in zigbe2mqtt folder and define the friendly name there
3. I reboot the add-on
4. I then edit the global HA configuration.yaml and include in it the device specific configuration parts (I do not use mqtt autodiscover)
5. I reboot HA
6. New entities appear in HA if everything works ok

Given this, could you help me understand how to avoid ending up with the physical name?

thank you so much in advance!

nldroid, 2020-01-07:
The moment that you pair a device it doesn't have a friendly name yet but it's already communicated to Home Assistant with the device id. You can just delete the device in Home Assistant and it will be added with the new name.

To prevent this from happening is probably not easier. If Home Assistant is not running while pairing, it won't be added. You can start Home Assistant after renaming the device in Z2M. You'll have to make sure that the MQTT messages aren't retained in your MQTT broker.

marcgarciamarti, 2020-01-07:
> The moment that you pair a device it doesn't have a friendly name yet but it's already communicated to Home Assistant with the device id. You can just delete the device in Home Assistant and it will [...]

If I understand your first statement above, I could simply remove the entities in HA and new entities would be recreated using the friendly name? is that really the case or am I misunderstanding something?

nldroid, 2020-01-07:
Yes. See here for more info about how it works in Home Assistant:

https://community.home-assistant.io/t/easiest-way-to-remove-mqtt-device/130466

marcgarciamarti, 2020-01-12:
Hi @nldroid can you confirm that the information provided above also applies to setups without MQTT autodiscovery?

marcgarciamarti, 2020-01-19:
> Yes. See here for more info about how it works in Home Assistant:
> 
> https://community.home-assistant.io/t/easiest-way-to-remove-mqtt-device/130466

I just wanted to say that you hit the nail right on its head. I indeed removed the previously created entity, and the system created one using the friendly name right after. Thank you so much for being so precise!!

nldroid, 2020-01-20:
So the issue can be closed?

marcgarciamarti, 2020-01-20:
> So the issue can be closed?

Yes! I know now how to fix it although I’m not sure why did it ended up like that in the first place. If somebody could clarify why, I would appreciate it.

Knowing that I have MQTT discover disabled, Do I need to restart z2m and then let HA see the newly created entity, and then reboot HA? Or reboot HA first?

closed 2020-02-01 by marcgarciamarti

Morphy99, 2020-07-15:
Sorry to resurrect an old issue but I can see there is an workflow issue with getting the entities named as per the "friendly names" in Home Assistant. I'm looking to automate joining and renaming devices as much as possible and would like to do all this from lovelace in HA. 

To add a device I've automated:
1. Enabling joining (MQTT publish)
2. Visual feedback of the joining (MQTT sensor) 
3. Renaming (MQTT publish from HA)
4. Disable Joining (MQTT publish)

To correctly name the entities in HA I'd need to do something like this:
1. Enabling joining (MQTT publish)
**2. Disable homeassistant discovery (Not available via MQTT?)**
3. Visual feedback of the joining (MQTT sensor) 
4. Renaming (MQTT publish from HA)
**5. Enable homeassistant discovery (Not available via MQTT?)
6. Publish device updates from Z2M to HA without restarting Z2M**

Having looked into it 2 and 5 are not possible via MQTT, could this be implemented?
6 I need to look up and test the correct MQTT topic if possible at all. Restarting a container isn't too much of a problem for me however.

Koenkk, 2020-07-15:
@Morphy99 disabling/enabling Home Assistant will be possible with the new api (see TODO list: https://github.com/Koenkk/zigbee2mqtt/issues/3281#issue-593587723)