Koenkk/zigbee2mqtt #2597: Iris iL0_7 motion sensor failed to interview

jpoconn87, 2019-12-24:
# Bug Report
## What happened
I recently picked up two Iris iL0_7 motion sensors and I'm having issues getting them to work. The sensors pair initially but fail the interview and never work properly. I've tried restarting and removing and repairing but nothing seems to help. I know the sensors are functional because they pair to Smartthings.

On a side note, the iL0_7 sensor supports temperature and humidity measurement as well. Is it possible to enable these readings?

This is a copy of the logs during the pairing process with debugging on. Please let me know what else I can provide to help solve this problem!
https://pastebin.com/3TnwkAZW 

## Debug Info
zigbee2mqtt version: Hassio edge version 1.8.0-dev
CC253X firmware version: source routing 20190619

domas902, 2019-12-24:
Dobrý den jsem zmateny vůbec nevím co mám dělat

jasperro, 2019-12-24:
It is possible the issue is a custom `network_key`, remove it if you have it and re-pair.

jpoconn87, 2019-12-24:
No custom network keys enabled on my setup, but thanks for the idea! I just had it removed from the network for almost a day and tried again with the same results. It shows up with the proper components (link quality, occupancy, low_battery) but nothing ever triggers and the device is unresponsive on the network after pairing.

fiercedeitylink, 2019-12-29:
Similar problem here with the Iris 3320-L contact sensor. Most of the time, I get "failed to interview", but when I got it to pair/interview successfully, it doesn't publish anything to zigbee2mqtt/devicename.

Logs:
Successful pairing and interview: https://pastebin.com/TDLWA7st
Successful pairing but interview failed: https://pastebin.com/tcbSwPi7
Failed pairing and interview: https://pastebin.com/66Gd5GDP

I tried a clean install of both the latest stable version and the dev branch and I couldn't get them to work. Weird thing is that I was able to pair two of them without any issues. Using a CC2531 running 20190619. I also have two SmartThings IM6001-MPP01 multipurpose sensors that have similar issues getting them to connect successfully (couldn't get either of them to work, but they work with the SmartThings hub along with the Iris sensors).

Koenkk, 2020-01-06:
Can you post the herdsman debug logging of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

fiercedeitylink, 2020-01-06:
Oh oops, I didn't see that.
Iris 3320-L: https://pastebin.com/kPJ3V36a
SmartThings IM6001-MPP01: https://pastebin.com/zrFMK7ny

I tried it with dev and 1.8.0, and I get a lot of "node descriptor failed" and then "Error: Interview failed because can not get active endpoints".

jpoconn87, 2020-01-07:
I'm on Hassio and it doesn't seem like the log is saving the herdsman debugging information. Here's logs from two different pairing attempts. The herdsman logs are in a different time zone than my normal logs for some reason too, these attempts were only a few minutes from each other.

log file from hassio share folder: https://pastebin.com/xUxREGY8 

Copy/pasted log from addon page while running: https://pastebin.com/5CVN6GSE

Koenkk, 2020-01-08:
@fiercedeitylink as you have the smart things hub, could you sniff the pairing of both sensors when connecting them with the smartthings hub and when pairing them to the latest zigbee2mqtt dev? In this way I can compare both and see where the differences are. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

fiercedeitylink, 2020-01-08:
Ok, I tried but I was having a lot of issues. Hopefully it contains something useful. https://www.dropbox.com/s/9wzn33yax59g6fh/zigbee%20wireshark%20logs.zip?dl=0

I was able to add both the Iris and SmartThings sensor to SmartThings, but I couldn't get Zigbee2MQTT to pair with the SmartThings multipurpose sensor at all. The sensor appeared to be broadcasting "beacon request" repeatedly according to Wireshark and Z2M didn't show "device connected" or "interview started" at all. This is new since I posted, so maybe it's something with my setup.

I was, however, able to get Iris to pair and interview successfully with Z2M, but it was still doing the thing where it doesn't update its state or publish the state to MQTT.

jpoconn87, 2020-01-10:
I reinstalled Hassio and the iris sensor is responding as expected now. It actually still fails the interview, but the occupancy sensor is working fine. The iL07_1 doesn't have a tamper sensor as far as I can tell, but the Zigbee2mqtt seems to be looking for one. Could that be causing the failure?

Koenkk, 2020-01-12:
@fiercedeitylink 

I don't see anything pairing related in any of the smartthings logs, are you sure these are the logs at the moment when you pair the device with smartthings?

fiercedeitylink, 2020-01-13:
I'm p sure, I even made sure to open/close the contact sensor a few times, but it didn't show any activity. The SmartThings app, did show it opening/closing, however. I can try again today/tomorrow, it just takes a while for me to turn off/unplug 14 lightbulbs. I think it's time for me to reset everything and start over, too...

fiercedeitylink, 2020-01-14:
I did it without removing all of the light bulbs, but it appeared work this time. I got device announcement/Samjin (and Ember for the Iris sensor) and ZCL OnOff.

Adding SmartThings multipurpose sensor to SmartThings hub: https://www.dropbox.com/s/n4lmuu1exsuwon0/add%20smartthings%20to%20smartthings4.pcapng?dl=0

Adding Iris contact sensor to SmartThings hub: https://www.dropbox.com/s/td4wop10b1vg4jz/add%20iris%20to%20smartthings3.pcapng?dl=0

Adding SmartThings sensor to Zigbee2MQTT (failed to interview): https://www.dropbox.com/s/qtgy7bd9lbx9ddw/add%20smartthings%20to%20z2m.pcapng?dl=0

Same thing (failed to interviewed) but "successfully configured"):
https://www.dropbox.com/s/ssldl98nlm0f0ja/add%20smartthings%20to%20z2m%202.pcapng?dl=0

Hope it helps.

Koenkk, 2020-01-16:
> Adding SmartThings multipurpose sensor to SmartThings hub: https://www.dropbox.com/s/n4lmuu1exsuwon0/add%20smartthings%20to%20smartthings4.pcapng?dl=0

This looks like a zigbee2mqtt log, doesn't seem to be from smartthings

> Adding Iris contact sensor to SmartThings hub: https://www.dropbox.com/s/td4wop10b1vg4jz/add%20iris%20to%20smartthings3.pcapng?dl=0

At packet ID 121 you see the device is leaving, after that not further activity. Did the device work after this pairing? (it looks strange to me). 

[image]

fiercedeitylink, 2020-01-16:
I think I remember doing permit_join false, but I did it again just in case:
Multipurpose sensor added to SmartThings:
https://www.dropbox.com/s/hgejhwdk9fvgni5/add%20smartthings%20to%20smartthings5.pcapng?dl=0
Iris added to SmartThings:
https://www.dropbox.com/s/om9y7xvra503px8/add%20iris%20to%20smartthings4.pcapng?dl=0

And I thought that it worked, because I open/closed it a few times after pairing.

Koenkk, 2020-01-16:
When sniffing are you sure that you are sniffing on the correct channel? Also when sniffing smartthings make sure to power down your complete zigbee2mqtt network (unplug coordinator and power down all routers, e.g. bulbs).

Reason I'm asking:
> Multipurpose sensor added to SmartThings: 

Still seems to be a zigbee2mqtt sniffing log 

> Iris added to SmartThings:

Doesn't contain any pairing info (as far as I can see), nor do I see open/close commands you describe.

fiercedeitylink, 2020-01-17:
Oh ok, that makes more sense. It was on channel 19.
https://www.dropbox.com/s/gw7eyy4zdxvl921/add%20smartthings%20to%20smartthings.pcapng?dl=0
https://www.dropbox.com/s/8f2p0wx5gb3eohl/add%20iris%20to%20smartthings.pcapng?dl=0

Hopefully, these work otherwise I can try to turn off all the light bulbs/routers tomorrow.

Koenkk, 2020-01-17:
Thanks, that was exactly what I needed.

Can you try replacing `node_modules/zigbee-herdsman/dist/controller/model/device.js` with https://gist.github.com/Koenkk/a45bdd9da8f8df31b10490663dae78ab and pair again? 

Please provide the wireshark log together with the debug log.

fiercedeitylink, 2020-01-17:
I got the SmartThings sensor to connect, but now I'm having trouble getting the Iris sensor to connect.

Here are the logs for the SmartThings sensor:
https://pastebin.com/0YUbDhbh
https://www.dropbox.com/s/7ul2xngrxv63tc3/add%20smartthings%20to%20z2m%20new.pcapng?dl=0

I'll edit this if I can get the Iris sensor to connect. Z2M and Wireshark don't show anything, so it might be the sensor itself. Weird because I wasn't having any problems before, and it connects quickly to the SmartThings hub.

Koenkk, 2020-01-17:
@fiercedeitylink I see there is still an exception while pairing (I made a mistake in the `device.js` I've provided, Please try with the new one; https://gist.github.com/Koenkk/a45bdd9da8f8df31b10490663dae78ab

Is the smartthings sensor working as expected after these changes?

fiercedeitylink, 2020-01-17:
Neat thanks, it works now.

Logs in case you need them:
https://pastebin.com/ECcLXNTR
https://www.dropbox.com/s/5u498ixkv53ph2q/add%20smartthings%20to%20z2m%20new%202.pcapng?dl=0

Koenkk, 2020-01-17:
Great that it works now, I also want to get rid of the `interview_failed` error but for that I need some more info. Can you try with the following device.js https://gist.github.com/Koenkk/a45bdd9da8f8df31b10490663dae78ab and provide the logs again?

I'm also curious if it works with the iris sensor now, perhaps you smartthings network was still powered when trying to pair it to zigbee2mqtt?

fiercedeitylink, 2020-01-17:
This one gives me an error at line 318: setTimeout(() => {yield endpoint.read('ssIasZone', ['iasCieAddr', 'zoneState']);}, 2000);
https://pastebin.com/sfc1qej3

Koenkk, 2020-01-17:
Updated again: https://gist.github.com/Koenkk/a45bdd9da8f8df31b10490663dae78ab

fiercedeitylink, 2020-01-17:
First try didn't seem to work: https://www.dropbox.com/s/w9ceuoq0t1ezpz9/add%20smartthings%20to%20z2m%20new%203.pcapng?dl=0

Second try worked: https://www.dropbox.com/s/qk7rhi4j5z771pu/add%20smartthings%20to%20z2m%20new%204.pcapng?dl=0

Z2M log for both: https://pastebin.com/ayETnrj3 I force_removed it after the first try.

Still struggling with the Iris sensor. I tried just connecting it to the master branch and it didn't work either, but I'll keep trying with the dev branch and this device.js.

Koenkk, 2020-01-18:
@fiercedeitylink great, I've pushed the changes to the latest zigbee2mqtt dev branch, interview should succeed now, can you try?

fiercedeitylink, 2020-01-18:
Neat, I got one of the SmartThings sensors to pair and work properly but it still said failed to interview. My other SmartThings sensor failed to interview as well but didn't work. I wish I had Wireshark set up and capturing, but I can try and redo it tomorrow and see.

The Iris sensors pair again for some reason, but it still doesn't show anything when the contact is open/closed.
```zigbee2mqtt:info  2020-01-18 16:12:38: Device '0x000d6f000ef61069' joined
zigbee2mqtt:info  2020-01-18 16:12:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6f000ef61069"}}'
zigbee2mqtt:info  2020-01-18 16:12:38: Starting interview of '0x000d6f000ef61069'
zigbee2mqtt:info  2020-01-18 16:12:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6f000ef61069"}}'
zigbee2mqtt:debug 2020-01-18 16:12:38: Device '0x000d6f000ef61069' announced itself
zigbee2mqtt:debug 2020-01-18 16:12:52: Received Zigbee message from '0x000d6f000ef61069', type 'readResponse', cluster 'genBasic', data '{"modelId":"3320-L","manufacturerName":"CentraLite","powerSource":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-01-18 16:12:52: No converter available for '3320-L' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"3320-L","manufacturerName":"CentraLite","powerSource":3}'
zigbee2mqtt:debug 2020-01-18 16:12:53: Received Zigbee message from '0x000d6f000ef61069', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"stackVersion":2}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-01-18 16:12:53: No converter available for '3320-L' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1,"stackVersion":2}'
zigbee2mqtt:debug 2020-01-18 16:12:54: Received Zigbee message from '0x000d6f000ef61069', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"����������������"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-01-18 16:12:54: No converter available for '3320-L' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"����������������"}'
zigbee2mqtt:debug 2020-01-18 16:12:59: Saving state to file /app/data/state.json
zigbee2mqtt:info  2020-01-18 16:13:02: Successfully interviewed '0x000d6f000ef61069', device has successfully been paired
zigbee2mqtt:info  2020-01-18 16:13:02: Device '0x000d6f000ef61069' is supported, identified as: Iris Contact and temperature sensor (3320-L)
zigbee2mqtt:info  2020-01-18 16:13:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000d6f000ef61069","model":"3320-L","vendor":"Iris","description":"Contact and temperature sensor","supported":true}}'
zigbee2mqtt:info  2020-01-18 16:13:02: Configuring '0x000d6f000ef61069'
zigbee2mqtt:info  2020-01-18 16:13:06: Successfully configured '0x000d6f000ef61069'

Koenkk, 2020-01-18:
Good, do you also have the herdsman debug logging? (I need that to see why interview fails).

fiercedeitylink, 2020-01-20:
@Koenkk sorry, I only have two CC2531 so I always have to switch one between my working setup and test setup.

Here are the logs for the two SmartThings sensors added to Z2M (herdsman debug logs and Wireshark): https://www.dropbox.com/s/8nri82c9rxpwuj2/z2m%20debug.zip?dl=0

Having trouble again getting the Iris sensor to appear at all. Seems like it only works when I get the green/red blinking LED and not blinking blue. I'll update if I can figure it out.

Koenkk, 2020-01-20:
@fiercedeitylink this doesn't seem to be with the latest dev branch (logging is a bit different there which I don't see in your logs). Note that after pulling the latest dev branch you need to reinstall the dependencies. Remove the node_modules folder and execute `npm install`.

fiercedeitylink, 2020-01-20:
@Koenkk oh ok, that would explain a lot, oops.  On my main setup, it successfully interviews the SmartThings sensor:

```info  2020-01-20 15:07:41: Device '0x286d970001006006' joined
info  2020-01-20 15:07:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x286d970001006006"}}'
info  2020-01-20 15:07:41: Starting interview of '0x286d970001006006'
info  2020-01-20 15:07:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x286d970001006006"}}'
debug 2020-01-20 15:07:41: Device '0x286d970001006006' announced itself
debug 2020-01-20 15:07:41: Received Zigbee message from '0x286d970001006006', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":33,"extendedstatus":0}' from endpoint 1 with groupID 0
debug 2020-01-20 15:07:41: Received Zigbee message from '0x286d970001006006', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2374}' from endpoint 1 with groupID 0
debug 2020-01-20 15:07:43: Received Zigbee message from '0x286d970001006006', type 'readResponse', cluster 'genBasic', data '{"modelId":"multi","manufacturerName":"Samjin"}' from endpoint 1 with groupID 0
info  2020-01-20 15:07:43: MQTT publish: topic 'homeassistant/sensor/0x286d970001006006/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x286d970001006006","json_attributes_topic":"zigbee2mqtt/0x286d970001006006","name":"0x286d970001006006_temperature","unique_id":"0x286d970001006006_temperature_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x286d970001006006"],"name":"0x286d970001006006","sw_version":"Zigbee2mqtt 1.9.0-dev","model":"Multipurpose sensor (2018 model) (IM6001-MPP01)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-01-20 15:07:43: MQTT publish: topic 'homeassistant/binary_sensor/0x286d970001006006/contact/config', payload '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","state_topic":"zigbee2mqtt/0x286d970001006006","json_attributes_topic":"zigbee2mqtt/0x286d970001006006","name":"0x286d970001006006_contact","unique_id":"0x286d970001006006_contact_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x286d970001006006"],"name":"0x286d970001006006","sw_version":"Zigbee2mqtt 1.9.0-dev","model":"Multipurpose sensor (2018 model) (IM6001-MPP01)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-01-20 15:07:43: MQTT publish: topic 'homeassistant/sensor/0x286d970001006006/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x286d970001006006","json_attributes_topic":"zigbee2mqtt/0x286d970001006006","name":"0x286d970001006006_battery","unique_id":"0x286d970001006006_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x286d970001006006"],"name":"0x286d970001006006","sw_version":"Zigbee2mqtt 1.9.0-dev","model":"Multipurpose sensor (2018 model) (IM6001-MPP01)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-01-20 15:07:43: MQTT publish: topic 'homeassistant/sensor/0x286d970001006006/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x286d970001006006","json_attributes_topic":"zigbee2mqtt/0x286d970001006006","name":"0x286d970001006006_linkquality","unique_id":"0x286d970001006006_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x286d970001006006"],"name":"0x286d970001006006","sw_version":"Zigbee2mqtt 1.9.0-dev","model":"Multipurpose sensor (2018 model) (IM6001-MPP01)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
debug 2020-01-20 15:07:43: Received Zigbee message from '0x286d970001006006', type 'readResponse', cluster 'genBasic', data '{"powerSource":3,"zclVersion":2}' from endpoint 1 with groupID 0
debug 2020-01-20 15:07:43: Received Zigbee message from '0x286d970001006006', type 'readResponse', cluster 'genBasic', data '{"appVersion":17}' from endpoint 1 with groupID 0
debug 2020-01-20 15:07:43: Received Zigbee message from '0x286d970001006006', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 1 with groupID 0
debug 2020-01-20 15:07:44: Received Zigbee message from '0x286d970001006006', type 'readResponse', cluster 'genBasic', data '{"swBuildId":""}' from endpoint 1 with groupID 0
debug 2020-01-20 15:07:44: Received Zigbee message from '0x286d970001006006', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00124b000be67540","zoneState":1}' from endpoint 1 with groupID 0
debug 2020-01-20 15:07:45: Received Zigbee message from '0x286d970001006006', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00124b000be67540","zoneState":1}' from endpoint 1 with groupID 0
info  2020-01-20 15:07:45: Successfully interviewed '0x286d970001006006', device has successfully been paired
info  2020-01-20 15:07:45: Device '0x286d970001006006' is supported, identified as: SmartThings Multipurpose sensor (2018 model) (IM6001-MPP01)
[... cut ...]
info  2020-01-20 15:07:45: Configuring '0x286d970001006006'
info  2020-01-20 15:07:46: Successfully configured '0x286d970001006006'
debug 2020-01-20 15:08:00: Received Zigbee message from '0x286d970001006006', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":32,"extendedstatus":0}' from endpoint 1 with groupID 0
info  2020-01-20 15:08:00: MQTT publish: topic 'zigbee2mqtt/0x286d970001006006', payload '{"contact":true,"tamper":false,"battery_low":false,"linkquality":49}'

Koenkk, 2020-01-21:
@fiercedeitylink awesome!

closed 2020-03-28 by stale[bot]