Koenkk/zigbee2mqtt #2599: Ikea Float Panel MAC no ack' (233)

mullischlumpf, 2019-12-24:
zigbee2mqtt version: 1.8.0 
Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}' 

Hello,

I have a strange behaviour with  my Ikea Float Panel. I paired the device successfully and everything seems fine. But when i publish a Message to the device i get an error (See log below). 

I already removed the device from the database file manually and readd the Float Panel but get the same error. 

I had my coordinator for pairing only a few centimeters away from the Float Panel. So for now i have no Idea what i could do next... all other devices workes fine

Here is the log:

zigbee2mqtt:info  2019-12-24 10:23:53: Device '0x000b57fffe9e4f52' joined
stdout
10:23:53
zigbee2mqtt:info  2019-12-24 10:23:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000b57fffe9e4f52"}}'
stdout
10:23:53
zigbee2mqtt:info  2019-12-24 10:23:53: Starting interview of '0x000b57fffe9e4f52'
stdout
10:23:53
zigbee2mqtt:info  2019-12-24 10:23:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000b57fffe9e4f52"}}'
stdout
10:24:03
zigbee2mqtt:info  2019-12-24 10:24:03: Successfully interviewed '0x000b57fffe9e4f52', device has successfully been paired
stdout
10:24:03
zigbee2mqtt:info  2019-12-24 10:24:03: Device '0x000b57fffe9e4f52' is supported, identified as: IKEA FLOALT LED light panel, dimmable, white spectrum (30x90 cm) (L1528)
stdout
10:24:03
zigbee2mqtt:info  2019-12-24 10:24:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000b57fffe9e4f52","model":"L1528","vendor":"IKEA","description":"FLOALT LED light panel, dimmable, white spectrum (30x90 cm)","supported":true}}'
stdout

zigbee2mqtt:error 2019-12-24 10:25:27: Publish 'set' 'state' to '0x000b57fffe9e4f52' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'
stdout
10:25:27
zigbee2mqtt:info  2019-12-24 10:25:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x000b57fffe9e4f52' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'","meta":{"friendly_name":"0x000b57fffe9e4f52"}}'

And here the entry in the database:

{"id":52,"type":"Router","ieeeAddr":"0x000b57fffe9e4f52","nwkAddr":60623,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"FLOALT panel WS 30x90","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":544,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"FLOALT panel WS 30x90","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170331","swBuildId":"1.2.217"}}},"binds":[]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170331","swBuildId":"1.2.217","zclVersion":1,"interviewCompleted":true,"meta":{}}

jasperro, 2019-12-24:
Are you using a custom `network_key`?, that could be the issue.

mullischlumpf, 2019-12-24:
Yes i changed the network_key because i switched from a cc2531 to CC26X2R1 and i had to change the key to get zigbee2mqtt working.
So the only way is switch to default key and readd all devices???

Would be great if there is a way to avoid repairing all 42 devices :-/

jasperro, 2019-12-24:
I guess so, but it's a pain to repair all those devices. I don't know if there is a way to avoid repairing everything, and even then I don't know if that will fix your problem.

mullischlumpf, 2019-12-25:
Hmmm what a hassle just because of one device :( but otherwise i have to find a way to get it working again... i think i will try it and hope it will help. Thanks for your reply. I will update you on this :-)

mullischlumpf, 2019-12-26:
I could fix that Problem. It was not the Network key... i had to switch the channel and after repairing everything the ikea float panel works again 👍

closed 2019-12-26 by mullischlumpf