Koenkk/zigbee2mqtt #2602: i can't configure xiaomi smoke detector

kitopopo, 2019-12-24:
# i can't configure xiaomi smoke detector
## I am trying to send messages to my xiaomi smoke detector through mosquitto_sub but I do not get the message to reach its destination, I have tried all these options:

sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x00158d0xxxxxxx/set {"selftest": ""}
sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x00158d0xxxxxxxx/set/ {"selftest": ""}
sudo mosquitto_pub -h 127.0.0.1 -t -m zigbee2mqtt/0x00158xxxxxxx/set {"sensitivity": "low"}
sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x00158d00xxxxxx/set -m {"sensitivity": "low"}
sudo mosquitto_pub -h localhost -t zigbee2mqtt/0x00158d00xxxxxx/get/ -m { "state": "" }

None of them work, I get the error: Error: Unknown option '{'.

Could you tell me what the correct structure would be? I also tried to turn off an xiaomi smart plug switch and I get the same errors:

sudo mosquitto_pub -h localhost -t zigbee2mqtt/0x00158d00026xxxxxx/set -m "{ "state: OFF" }"

I await your news. Thank you in advance and thank you very much for your work and dedication

## What did you expect to happen

Let the messages reach their destination

## How to reproduce it (minimal and precise)

typing the message in the Raspberry4 command console. 

## Debug Info
zigbee2mqtt version: 1.8.0 (raspberry pi 4)
CC253X firmware version: 20190608.zip

jasperro, 2019-12-24:
`mosquitto_pub -h localhost -t "zigbee2mqtt/0x00158d00026xxxxxx/set" -m "{ \"state: OFF\" }"` should work. You need to escape the quotes.

kitopopo, 2019-12-25:
hi dear friend,

I test the following message:

mosquitto_pub -h 192.168.1.60 -t "zigbee2mqtt/0x00158d000xxxxx/set" -m "{ \"state: OFF\" }" with the quotes and without the quotes but i receive this log messages:

Publish 'set' 'state' to 'zigbee2mqtt/0x00158d000xxxxx' failed: 'Error: Cluster 'genOnOff' has no command '{"state: off"}''

MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'zigbee2mqtt/0x00158d000xxxxx' failed: 'Error: Cluster 'genOnOff' has no command '{\"state: off\"}''","meta":{"friendly_name":"zigbee2mqtt/0x00158d000xxxxx"}}'

any solution? i don't understand.

Thanks in advanced

Best regards

jasperro, 2019-12-25:
Have you tried the sensitivity too? `sudo mosquitto_pub -h 127.0.0.1 -t -m zigbee2mqtt/0x00158xxxxxxx/set {\"sensitivity\": \"low\"}`

kitopopo, 2019-12-25:
Hi dear friend,

I test the following messages:

- sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x00158d000411xxxx/set -m "{\"sensitivity\": \"high\"}"    --> Works correctly

-  sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x00158d000411xxxx/set -m "{\"selftest\": \"\"}"
--> Works correctly

the documentation should be updated in the following link:

https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/JTYJ-GD-01LM_BW.md#sensitivity

After make the self test the smoker detector led powers on 1 second in red color and after powers on 0.5 seconds in orange colour is correctly? Does should it be green? The selfftest does the selftest run automatically daily? I wait your news

I thank you very much for your answers and dedication, I use zigbe2mqtt for many devices, it is a very good project. Merry Christmas

closed 2019-12-27 by kitopopo