Koenkk/zigbee2mqtt #2604: GLEDOPTO once again

andreasbrett, 2019-12-24:
Sorry to bother you again with GLEDOPTO but my GL-C-008 won't work anymore after updating to latest dev (I'm on commit 66b3bcb). I read the previous issues here on GitHub but non of them solved my problem. I have a RGBW GL-C-008 but my LED strip is only RGB which is why I only connected the RGB wires of the GLEDOPTO to it.

I removed it from the network (even force-removing) and re-paired it but nothing works. I can't turn it off nor change color. Database looks like this:

{"id":2,"type":"Router","ieeeAddr":"0x00124b0019c60150","nwkAddr":7020,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO","epList":[10,11,13],"endpoints":{"10":{"profId":49246,"epId":10,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GLEDOPTO","manufacturerName":
"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.2"}}},"binds":[]},"11":{"profId":49246,"epId":11,"devId":
528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{},"binds":[]},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":
{},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.2","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1577216014956}

It's the exact same behaviour I had November 2018 which you fixed with a new commit that was specific to the endpoint configuration of my GL-C-008 model (RGBW, not RGB+CCT).

Here's the old issue:
https://github.com/Koenkk/zigbee2mqtt/issues/608#issuecomment-442427462

andreasbrett, 2019-12-26:
Looks like this commit made z2m incompatible with my GLEDOPTO controller. https://github.com/Koenkk/zigbee-herdsman-converters/pull/661

Koenkk, 2019-12-27:
The problem is that gledopto messed up their modelIDs in earlier version (7 carry the 8 modelID, 8 should always be a rgb + ct controller, 7 rgbw). Does manually changing the modelID to GL-C-007 in database.db fix the issue?

andreasbrett, 2019-12-27:
> The problem is that gledopto messed up their modelIDs in earlier version (7 carry the 8 modelID, 8 should always be a rgb + ct controller, 7 rgbw). Does manually changing the modelID to GL-C-007 in database.db fix the issue?

No, doesn't fix the issue.

andreasbrett, 2019-12-29:
Issue can be closed once https://github.com/Koenkk/zigbee-herdsman-converters/pull/857 has been merged.

closed 2020-01-07 by Koenkk