Koenkk/zigbee2mqtt #2605: Unable to pair WSDCGQ01LM

m-pavel, 2019-12-24:
# Bug Report
## What happened
I'm unable to pair xiaomi temp/hum sensor WSDCGQ01LM. Just nothing in the logs. I'm continue pressing join button on sensor for 5 min. Battery is ok (tied some). Meantime IKEA button joined without any errors. 
Log is empty even if debug level is on (of course there are unrelated debug messages). 
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.8.0-dev 
CC253X firmware version: CC2531_DEFAULT_20190608.zip

jameswarner62, 2019-12-25:
I can only seem to pair these directly to the coordinator so is it close? Normally it's a 5 second hold to a pair which gives a flash, then a pause - then either one long pause for failed pairing  or 3 short flashes for success...

m-pavel, 2019-12-26:
It's 1 cm far from coordinator antenna. After 5sec. press it makes 3 short flashes and 1 flash after 1s pause. Nothing in the logs meantime. I can continue pressing connect button every sec. as it was recommended in some instruction, or I can leave it as is result is the same - nothing in the logs.
If I flash stick with sniffer fw I can see sensor sending requests. I'm waiting for a second stick and going to look what is happening between them.

tdn131, 2019-12-26:
@m-pavel please check your;
MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.8.0","commit":"da4d26a","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'

I noticed that permit_join was set to false on 1.8.0-dev, even though it was enabled. 

I also can't repair xiaomi devices. I have no idea whats wrong. I don't see any lines in the logs regarding the sensors. For me it's WSDCGQ11LM.

pcb1962, 2019-12-26:
Do you perhaps have another zigbee hub of some sort nearby (Ikea, Hue, Xiaomi etc)? 
3 short flashes usually means it has paired ok, maybe it's paired with something other than your z2m hub.

holgerebert, 2020-01-11:
Same problem here: i have already 2 of them running without problems, but i can't pair none of 4 new sensors. Nothing in the log, even if starting with DEBUG=* 

Any suggestions?

closed 2020-03-18 by stale[bot]

jackwilsdon, 2020-05-11:
I'm still seeing this problem on zigbee2mqtt 1.13.0 - there is nothing in the log in debug mode when attempting to pair (and I've ensured `permit_join` is `true`).

razvanmdobre, 2020-07-03:
I have the same issue with zigbee2mqtt 1.14.1 - I already have two sensors paired long time ago. Now, I got 4 new sensors and I cannot pair them. Just like reported before, absolutely nothing in zigbee2mqtt log.
I unplugged the Ikea gateway, I stopped the 2.4GHz wifi but no success.