Koenkk/zigbee2mqtt #2607: Updated Securifi Peanut not reporting power levels

jpoconn87, 2019-12-25:
I recently updated the firmware on my Peanut plug but all the power levels are still reading as 0. I know the update worked because I can see levels when I have the plug paired to Smartthings or the Almond router. Does anyone have one of these plugs that's actually reporting energy levels? Any suggestions on how to get mine reporting? 

## Debug Info
zigbee2mqtt version: Hassio Edge version 1.8.0-dev
CC253X firmware version: 20190619 - source routing

jpoconn87, 2019-12-29:
Looks like I'm getting power information but the converter isn't working? I'm very new to this so I'm not sure where to begin.

https://pastebin.com/39f5GdkT

jpoconn87, 2020-01-04:
The mystery continues. I did the same update to my friend's Peanut and he can see power levels in his hassio environment. We're both using the same edge version, installed fresh in the last few days. No custom devices.js enabled. 

Also odd is I noticed my iris 3210 smart plug isn't reporting power levels either. Could there be an issue with my converters file? It seems to be affecting any device that measures power levels. I've tried removing and repairing a dozen times now probably. Deleting any trace of the devices from states.json, database, and every other place I can find. I even installed MQTT Explorer and deleted the entries for them from the broker.

jpoconn87, 2020-01-06:
Update: the iris 3210 plug is reporting power levels now, but still nothing from the peanut. I think I missed deleting something from the states.json for the iris plug previously. I tried the same process of removing the device and deleting all the residual entries in the system for both devices. When I paired them again the next day, the iris plug is reporting power but the Peanut stayed how it was.

Koenkk, 2020-01-08:
Can you share your `database.db` file?

jpoconn87, 2020-01-10:
I think this was an isolated issue. I did a fresh install of Hassio yesterday and all the devices I was having trouble with seem to be working as expected now. Thanks for responding, and thank you for this amazing project!!

closed 2020-01-10 by jpoconn87