Koenkk/zigbee2mqtt #2611: reporting error on ZNLDP12LM - Xiaomi Aqara smart LED bulb

SargonofAssyria, 2019-12-26:
zigbee2mqtt version: 1.8.0
CC253X firmware version: 20190608

Logging of this bulb shows:
info  2019-12-26 16:35:32: Overlooplamp (0x00158d0002b7d689): ZNLDP12LM - Xiaomi Aqara smart LED bulb (Router)
info  2019-12-26 16:35:33: Succesfully setup reporting for '0x00158d0002b7d689' - 1 - genOnOff
info  2019-12-26 16:35:33: Succesfully setup reporting for '0x00158d0002b7d689' - 1 - genLevelCtrl
info  2019-12-26 16:35:33: Succesfully setup reporting for '0x00158d0002b7d689' - 1 - lightingColorCtrl
error 2019-12-26 16:35:43: Failed to setup reporting for '0x00158d0002b7d689' - Error: Timeout - 41629 - 1 - 7 - 258 - $
    at Timeout._onTimeout (/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)

What about the last message? Bulb is working normally. I have 2 of them, both the same error.

Koenkk, 2019-12-31:
Can you share your `database.db`?

SargonofAssyria, 2019-12-31:
This is part of the database No. 4 is the Bulb.
`{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0012023155","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,110],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{}}
{"id":2,"ieeeAddr":"0x00158d00026e25d9","nwkAddr":12934,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65281":{"1":2985,"3":27,"4":17320,"5":26,"6":[0,1],"8":516,"10":40985,"100":1}}}},"binds":[]}},"interviewCompleted":false,"meta":{}}
{"id":3,"ieeeAddr":"0x00158d000236a71a","nwkAddr":45079,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"msIlluminanceMeasurement":{"attributes":{"measuredValue":0}},"msOccupancySensing":{"attributes":{"occupancy":1}},"genBasic":{"attributes":{"65281":{"1":2985,"3":30,"4":17320,"5":41,"6":[0,6],"10":0,"11":0,"100":0}}}},"binds":[]}},"interviewCompleted":false,"meta":{}}
{"id":4,"type":"Router","ieeeAddr":"0x00158d0002b7d689","nwkAddr":41629,"manufId":4447,"manufName":"LUMI","powerSource":"DC Source","modelId":"lumi.light.aqcn02","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,4,3,5,10,258,13,19,6,1,1030,8,768,1027,1029,1026],"outClusterList":[25,10,13,258,19,6,1,1030,8,768],"clusters":{"genBasic":{"attributes":{"65281":{"3":35,"5":169,"7":"0x0000000000000000","8":278,"9":2051,"10":0,"100":0,"101":127,"102":313},"modelId":"lumi.light.aqcn02","manufacturerName":"LUMI","powerSource":4,"zclVersion":1,"appVersion":22,"stackVersion":2,"hwVersion":1,"dateCode":"09-30-2018","swBuildId":"1.22\u0000"}},"genOnOff":{"attributes":{"61440":117440565,"onOff":0}},"genLevelCtrl":{"attributes":{"61440":10657024,"currentLevel":127}},"lightingColorCtrl":{"attributes":{"61440":10657024,"colorTemperature":313}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0012023155","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0012023155","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b0012023155","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x00124b0012023155","endpointID":1}]}},"appVersion":22,"stackVersion":2,"hwVersion":1,"dateCode":"09-30-2018","swBuildId":"1.22\u0000","zclVersion":1,"interviewCompleted":true,"meta":{}} `

Koenkk, 2020-01-07:
Thanks, fixed in latest dev.

closed 2020-01-07 by Koenkk

SargonofAssyria, 2020-01-08:
Tnx for the excellent work!