Koenkk/zigbee2mqtt #2613: Gledopto GL-C-008 ignores fade up transition X seconds brighthess up

wojciechczyz, 2019-12-26:
# Bug Report
## What happened
When using script or direct manipulating using Home Assistant / Developer Tools / Services
and manipulating light to fade into brighter ligth such as:

light.turn_on
with:
entity_id: light.0x00124b001d43cbed_light
brightness: 1
kelvin: 2300
transition: 0

and then

light.turn_on
with:
entity_id: light.0x00124b001d43cbed_light
brightness: 255
kelvin: 2300
transition: 20

transition is not well executed. It jumps into destination colour directly.

## What did you expect to happen

smooth transition from 1 to 255 taking 20 seconds.

## How to reproduce it (minimal and precise)

Go to Home Assistant / Developer Tools / Services
for service light.turn_on
and your light

light.turn_on
with:
entity_id: light.0x00124b001d43cbed_light
brightness: 1
kelvin: 2300
transition: 0

and then

light.turn_on
with:
entity_id: light.0x00124b001d43cbed_light
brightness: 255
kelvin: 2300
transition: 20

## Debug Info
zigbee2mqtt version:
1.8.0
CC253X firmware version:
source 1.2

closed 2020-03-03 by stale[bot]

myradon, 2020-03-03:
I think Gledopto is effected by a BUG concerning transitions. I use Z2M as well aloing with Home Assistant and Node-RED. I found out when I set e.g transition of 5s to other RGB-color, Gledopto just jumps steps. It doesn't turn off while HA and NR claim it's off.  

On the interwebs you can find more topics about Gledopto odd behaviour. Do you use Node-RED? Maybe as a workaround you could do a delay-/timer-node and do small steps with 2s transitions to final brightness Or write your own Function node with setTimeout to trigger every 2s an increment (node.send() ); It's shit but I'm actually going to try it. Might else move to ESPHome on a ESP8266 to replace Gledopto.

wojciechczyz, 2020-03-03:
It may have been related to problems with zigbee board. Thank you for help