Koenkk/zigbee2mqtt #2614: friendly_name vs device_id

runningman84, 2019-12-27:
It would be great to use the device_id as identifier in home assistant but use the friendly_name just as an additional attribute. This used to work just fine but I migrated my zigbee2mqtt instance and now all devices are lost in homeassistant and new devices are created with their friendly_name as device id:

before
sensor.0x00158d0001833b20_temperature
after
sensor.kitchen_temperature

Is this an expected behaviour?

## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version:

Koenkk, 2019-12-27:
This is indeed expected behaviour, when you migrate zigbee2mqtt you can copy your configuration.yaml to avoid this.

runningman84, 2019-12-27:
The funny thing is that I copied all files but still the sensors are renamed

runningman84, 2019-12-28:
It would be great to have the option to have sensors with their unique device id and use the friendly_name just as a frriendly name as home assistant.

runningman84, 2019-12-30:
Today I renamed all sensors in the configuration.yaml. According to the zigbee2mqtt logs everything is published with the new names.

But in home assistant the device id is still an old id but the friendly name is new (it matches the name in the configuration.yaml).

I had to remove the file ./.storage/core.entity_registry in order to force home assistant to use the new names...

closed 2020-02-28 by runningman84