Koenkk/zigbee2mqtt #2617: permit_join inconsistent

dirstel, 2019-12-27:
# Bug Report
## What happened
I set up a fresh Zigbee2MQTT Container via docker, using latest-dev.

in configuration.yaml permit_join is set to true.
This seems to be honored, but the published config (via MQTT) says permit_join is false

`Using '/app/data' as data directory
> zigbee2mqtt@1.8.0-dev start /app
> node index.js
zigbee2mqtt:info  2019-12-27 16:01:25: Logging to console and directory: 'data/log/2019-12-27.16-01-25'
zigbee2mqtt:info  2019-12-27 16:01:25: Starting zigbee2mqtt version 1.8.0-dev (commit #f069a7e)
zigbee2mqtt:info  2019-12-27 16:01:25: Starting zigbee-herdsman...
zigbee2mqtt:info  2019-12-27 16:01:26: zigbee-herdsman started
zigbee2mqtt:info  2019-12-27 16:01:26: Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'
zigbee2mqtt:info  2019-12-27 16:01:26: Currently 0 devices are joined:
zigbee2mqtt:warn  2019-12-27 16:01:26: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2019-12-27 16:01:26: Allowing new devices to join.
zigbee2mqtt:warn  2019-12-27 16:01:26: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2019-12-27 16:01:26: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2019-12-27 16:01:26: Connecting to MQTT server at mqtt://server-1.home:1883
zigbee2mqtt:info  2019-12-27 16:01:26: Connected to MQTT server
zigbee2mqtt:info  2019-12-27 16:01:26: MQTT publish: topic 'home-new/bridge/state', payload 'online'
zigbee2mqtt:info  2019-12-27 16:01:26: MQTT publish: topic 'home-new/bridge/config', payload '{"version":"1.8.0-dev","commit":"f069a7e","coordinator":{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}},"log_level":"info","permit_join":false}'`

## Debug Info
zigbee2mqtt version: dev.latest
CC2652 firmware version: 20191106

tdn131, 2019-12-28:
I also noticed this in the latest 1.8.0-dev. I am using a CC2531 though. permit_join is true, but in the line it's permit_join false. This has to be because of the changes from 1.8.0 to 1.8.0dev, since 1.8.0 doesn't have this.

AlmightyFrog, 2019-12-31:
I can't confirm it. I am using docker image latest which is 1.8.0 currently, and also there i have inconsistency. Even 1.7.1 and 1.7.0 seem to have it
```
app_1  | Using '/app/data' as data directory
app_1  | 
app_1  | > zigbee2mqtt@1.7.0 start /app
app_1  | > node index.js
app_1  | 
app_1  | zigbee2mqtt:info  2019-12-31T05:29:26: Logging to directory: '/app/data/log/2019-12-31.06-29-26'
app_1  | zigbee2mqtt:info  2019-12-31T05:29:26: Starting zigbee2mqtt version 1.7.0 (commit #551a58a)
app_1  | zigbee2mqtt:info  2019-12-31T05:29:26: Starting zigbee-herdsman...
app_1  | zigbee2mqtt:info  2019-12-31T05:29:28: zigbee-herdsman started
app_1  | zigbee2mqtt:info  2019-12-31T05:29:28: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
app_1  | zigbee2mqtt:info  2019-12-31T05:29:28: Currently 0 devices are joined:
app_1  | zigbee2mqtt:warn  2019-12-31T05:29:28: `permit_join` set to  `true` in configuration.yaml.
app_1  | zigbee2mqtt:warn  2019-12-31T05:29:28: Allowing new devices to join.
app_1  | zigbee2mqtt:warn  2019-12-31T05:29:28: Set `permit_join` to `false` once you joined all devices.
app_1  | zigbee2mqtt:info  2019-12-31T05:29:28: Zigbee: allowing new devices to join.
app_1  | zigbee2mqtt:info  2019-12-31T05:29:28: Connecting to MQTT server at mqtt://x.x.x.x
app_1  | zigbee2mqtt:info  2019-12-31T05:29:28: Connected to MQTT server
app_1  | zigbee2mqtt:info  2019-12-31T05:29:28: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
app_1  | zigbee2mqtt:info  2019-12-31T05:29:28: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.7.0","commit":"551a58a","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":false}'
```

Strangely I can also not add any new devices. This is just a reinstall of the server, before it worked, now it does not anymore...

Koenkk, 2020-01-07:
Is this issue still present in the latest dev? For me it's working fine now.

[image]

Wojciechem, 2020-01-30:
Hello, I seem to have the same issue:
```
zigbee2mqtt:info  2020-01-30 22:44:03: Logging to console and directory: '/app/data/log/2020-01-30.22-44-03'
zigbee2mqtt:info  2020-01-30 22:44:03: Starting zigbee2mqtt version 1.9.0 (commit #287f843)
zigbee2mqtt:info  2020-01-30 22:44:03: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-01-30 22:44:04: zigbee-herdsman started
zigbee2mqtt:info  2020-01-30 22:44:04: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:info  2020-01-30 22:44:04: Currently 0 devices are joined:
zigbee2mqtt:warn  2020-01-30 22:44:04: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-01-30 22:44:04: Allowing new devices to join.
zigbee2mqtt:warn  2020-01-30 22:44:04: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-01-30 22:44:04: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-01-30 22:44:04: Connecting to MQTT server at mqtt://localhost
zigbee2mqtt:info  2020-01-30 22:44:04: Connected to MQTT server
zigbee2mqtt:info  2020-01-30 22:44:04: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-01-30 22:44:04: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.9.0","commit":"287f843","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":false}'
```

sh4jonny, 2020-01-30:
> Is this issue still present in the latest dev? For me it's working fine now.
> 
> [image]

what kind of usb stick you are use ?

Koenkk, 2020-01-31:
Found the issue, fixed, thanks all!

closed 2020-01-31 by Koenkk