Koenkk/zigbee2mqtt #2618: Sylvania 73739 Interview failed

backcountrymountains, 2019-12-27:
I've got some Osram Sylvania Lightyify RGBW BR30 bulbs but I can't pair them.  They should be supported as [Sylvania 73739](https://www.zigbee2mqtt.io/devices/73739.html). It gets put in the database as only a router after the following interview failure:

Log:
```
debug 2019-12-26 22:10:23: Device '0x7cb03eaa00a5c039' announced itself
error 2019-12-26 22:10:42: Failed to interview '0x7cb03eaa00a5c039', device has not successfully been paired
info  2019-12-26 22:10:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x7cb03eaa00a5c039"}}'
debug 2019-12-26 22:10:45: Saving state to file /share/zigbee2mqtt/state.json
```
Database:
```
{"id":8,"type":"Router","ieeeAddr":"0x7cb03eaa00a5c039","nwkAddr":62903,"manufId":48042,"epList":[3],"endpoints":{"3":{"profId":260,"epId":3,"devId":258,"inClusterList":[0,3,4,5,6,8,768,64527],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LIGHTIFY BR RGBW","manufacturerName":"OSRAM","powerSource":1}}},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":1577423422998}
```
But then, a minute later, the light sends it's config info:
```
debug 2019-12-26 22:12:46: Received Zigbee message from '0x7cb03eaa00a5c039', type 'readResponse', cluster 'genBasic', data '{"modelId":"LIGHTIFY BR RGBW","manufacturerName":"OSRAM","powerSource":1}' from endpoint 3 with groupID 0
```
After that happens, I can use the light. 

The problem is that the correct information isn't stored in the database and the light won't work after a restart of zigbee2mqtt.

Is the interview failure a timeout thing that I can change? Is there a way to force it into the database? How do I make it work? What am I doing wrong?

Thanks.

## Debug Info
zigbee2mqtt version: Starting zigbee2mqtt version 1.8.0-dev (commit #2451d536f2c9e446d396df55c9493f256ab9352e)
CC253X firmware version: 1.2

stampeder, 2020-01-22:
I have two 74731's bought together in a starter kit. One pairs without issue when factory reset, the other is a brick. Won't pair no matter what I do.
Comments?

backcountrymountains, 2020-03-11:
Not sure @stampeder 
I've updated my z2m to 1.11.0. I just tried pairing again yesterday and it worked perfectly. Maybe there has been an update that fixed whatever bug I had?
Anyway, closing because this is no longer a problem for me.

closed 2020-03-11 by backcountrymountains