Koenkk/zigbee2mqtt #2620: brightness_move for groups

eilmi, 2019-12-27:
Hi Koen,
I just updated to zigbee2mqtt 1.8.0 and I was wondering if it´s possible to implement the new command "brightness_move" also in groups because right now it`s only working for single devices
(tried it with Ikea Gu10 spots)

Koenkk, 2019-12-31:
Added!

closed 2019-12-31 by Koenkk

sergiigladchuk, 2020-02-13:
Hi all,
I just tested this `brightness_move` command for my grouped lights. All works fine, the only problem is when i send command `{"brightness_move" : "stop"}`  the Z2M is not publishing state message with current brightness for the group. I have checked it for single light and message is being publish in this case. I use latest master branch.
@Koenkk is this by design or impossible to implement for the groups, or I need some special setup in my groups.yaml?

Koenkk, 2020-02-13:
@sergiigladchuk fixed in latest dev branch (try in a few hours), thanks for reporting.