Koenkk/zigbee2mqtt #2621: Cant devices and cant add after i turned the power off

Spyky, 2019-12-27:
Everything worked fine until i had to do something to my electric installation.
Turned off the raspberry the correct way and turned it on again after the power was up.

After this the Xiaomi sensors i  use with zigbee2mqtt are not in the device section anymore.
In hardware zigbee2mqtt is still there and i can remove and add it.
If i try to add again a xiaomi device it will not pick it up. 

test -w /dev/ttyACM0 && echo success

How do i find out that the sniffer picks up the connection signal from de xiaomi?

Spyky, 2020-01-02:
After reinstall error messages are gone, but still can't pair.

Spyky, 2020-01-15:
> After reinstall error messages are gone, but still can't pair.

I now see message bellow
Jan 15 16:10:47 raspberrypi npm[20901]: zigbee2mqtt:error 2020-01-15 16:10:47: E                            rror: The module '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@se                            rialport/bindings/build/Release/bindings.node'
Jan 15 16:10:47 raspberrypi npm[20901]: was compiled against a different Node.js                             version using
Jan 15 16:10:47 raspberrypi npm[20901]: NODE_MODULE_VERSION 79. This version of                             Node.js requires
Jan 15 16:10:47 raspberrypi npm[20901]: NODE_MODULE_VERSION 64. Please try re-co                            mpiling or re-installing

What to do? Do i have do a reinstall of an older version...strange.

Koenkk, 2020-01-16:
Try: `cd /opt/zigbee2mqtt && rm -rf node_modules && npm install`.

Spyky, 2020-01-16:
> cd /opt/zigbee2mqtt && rm -rf node_modules && npm install

Hello Koen. Thnks for the response. Did it but still the same problem.
Jan 16 17:09:21 raspberrypi npm[4780]: zigbee2mqtt:error 2020-01-16 17:09:21: Error: The module '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node'
Jan 16 17:09:21 raspberrypi npm[4780]: was compiled against a different Node.js version using
Jan 16 17:09:21 raspberrypi npm[4780]: NODE_MODULE_VERSION 79. This version of Node.js requires
Jan 16 17:09:21 raspberrypi npm[4780]: NODE_MODULE_VERSION 64. Please try re-compiling or re-installing

Jan 16 17:09:21 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jan 16 17:09:21 raspberrypi systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Jan 16 17:09:21 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.

Spyky, 2020-01-25:
Couldn't find the solution. So i put a backup back and it worked again.

closed 2020-01-25 by Spyky