Koenkk/zigbee2mqtt #2624: Zigbee bridge is not reporting connected devices after complete re-install

Wamor, 2019-12-27:
Hello,

I had to do a complete re-install of my software on the Raspberry Pi (domoticz, zigbee2mqtt, etc).
When I start domoticz I see that the following message is published:
MQTT message: zigbee2mqtt/bridge/config/devices 
I then would expect to receive a complete list of paired devices from my ZigBee-bridge but nothing is reported.
When I send the above message with MQTT.fx I see a full list of devices reported and I also can switch on and off some devices with domoticz.
When I want to switch on and off the same device after restarting domoticz I got the following error-message:
Device Zigbee2MQTT - 0xd0cf5efffecb6ed5 does not have registered zigbee2mqtt device.
Sending MQTT message: zigbee2mqtt/bridge/config/devices with MQTT.fx makes the device work again. It looks like domoticz is not receiving a list with connected devices after restart.
This is bothering me now for some time and I got no clue what is happening here.
Can somebody maybe give me some advise what is happening here and to solve this.

Regards,

Wamor

turboproc, 2019-12-30:
Something similar happening over here. Had for a long time 2 Ikea devices paired which work fine. Just added a Philips HUE bulb. According to the debug output it has been paired:

zigbee2mqtt:info  2019-12-30 16:48:35: Currently 3 devices are joined:
zigbee2mqtt:info  2019-12-30 16:48:35: 0x000d6fffxxxxxxx (0x000d6ffffxxxxxx): E1603/E1702 - IKEA TRADFRI control outlet (Router)
zigbee2mqtt:info  2019-12-30 16:48:35: 0x000b57fffexxxxxxx (0x000b57fxxxxxx): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
zigbee2mqtt:info  2019-12-30 16:48:35: 0x0017880104xxxxxxx (0x0017880104xxxxxx): 9290011370B - Philips Hue white A60 bulb E27 (Router)

BUT in terms of publishing zigbee2mqtt only publishes the first 2 devices through MQTT.

To make it workse, if I manually (publih "toggle" vi a command line), the HUE bulb will happely switch on and off.

Any idea what's happening?

Wamor, 2019-12-30:
Hi,

I did some extra testing today and discovered that everything work fine if I manually restart Domoticz after starting my Raspberry Pi. All devices are reported by MQTT and detected by Domoticz.
May be some timing issues here?

Regards,

Wamor

turboproc, 2019-12-30:
Did some further debugging and noticed that an error is thrown during joining:    zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":40977,"nwkaddr":40977,"ieeeaddr":"0x0017880104a50220","capabilities":142} +2m
  zigbee-herdsman:controller:log Device announce '0x0017880104a50220' +2m
  zigbee-herdsman:adapter:zStack:znp:error Error while parsing to ZpiObject 'TypeError: Cannot read property 'updateLastSeen' of undefined
  zigbee-herdsman:adapter:zStack:znp:error     at Controller.onDeviceAnnounce (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:269:16)
  zigbee-herdsman:adapter:zStack:znp:error     at ZStackAdapter.emit (events.js:189:13)
  zigbee-herdsman:adapter:zStack:znp:error     at ZStackAdapter.onZnpRecieved (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:424:22)
  zigbee-herdsman:adapter:zStack:znp:error     at Znp.emit (events.js:189:13)
  zigbee-herdsman:adapter:zStack:znp:error     at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:80:18)
  zigbee-herdsman:adapter:zStack:znp:error     at Parser.emit (events.js:189:13)
  zigbee-herdsman:adapter:zStack:znp:error     at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/parser.js:47:26)
  zigbee-herdsman:adapter:zStack:znp:error     at Parser._transform (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/parser.js:26:14)
  zigbee-herdsman:adapter:zStack:znp:error     at Parser.Transform._read (_stream_transform.js:190:10)
  zigbee-herdsman:adapter:zStack:znp:error     at Parser.Transform._write (_stream_transform.js:178:12)' +0ms

In reading the source code it looks like the device is not created (Also missing the  'device joined' message). Looks like the buHUE bulb is skipping this part.

turboproc, 2019-12-31:
This issue has been solved more or lessfor me  as covered in https://github.com/Koenkk/zigbee-herdsman/issues/118

closed 2020-02-29 by Koenkk