Koenkk/zigbee2mqtt #2625: Hue smart button doesn't pair with CC1352P-2

kitcorey, 2019-12-28:
# Bug Report
## What happened
I'm using the latest dev docker image with a CC1352P-2 adapter and I'm not able to add a hue smart button (9290022230) to the network.   I was able pair it to a hue bridge and update to the latest firmware (2.30.0_r30777). 

Pairing looks like it is successful in the logs, but pressing the button doesn't light the indicator LED or send any updates to mqtt.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: zigbee2mqtt:latest-dev
CC1352P-2 firmware version: CC1352P_2_20191106.zip
Debug logs: https://pastebin.com/MYwDMTiW

magnayn, 2020-01-10:
I don't think this is specific to the CC1352; I have a CC2531 and it exhibits the same behaviour (pairs, but button does nothing).

I _think_ it's to do with something more complex being required. I wrote it up here:
https://zigbee2mqtt.discourse.group/t/philips-hue-smart-button-not-working/751

I'm a bit stumped as to what to do next..

Koenkk, 2020-01-11:
@magnayn the configure probably fails because the device is sleeping, try waking up the device by pressing it while the configure happens (on start of zigbee2mqtt or when doing it manually via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure)

magnayn, 2020-01-11:
I got a response in the log that config isn't required (I reset the device itself, it wa definitely awake with a flashing light):

```
zigbee2mqtt:debug 2020-01-11 15:37:35: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data 'SmartButton'
winston:file written true false
winston:file logged 220841 debug 2020-01-11 15:37:35: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data 'SmartButton'

zigbee2mqtt:warn  2020-01-11 15:37:35: Skipping configure of 'SmartButton', device does not require this.
winston:file written true false
winston:file logged 220935 warn  2020-01-11 15:37:35: Skipping configure of 'SmartButton', device does not require this.
```

My configuration.yaml lists it, so pairing seemed to work:
```
devices:
  '0x286d97000108a3d1':
    friendly_name: 'MotionSensor'
  '0x00178801060766a4':
    friendly_name: 'SmartButton'
```

magnayn, 2020-01-11:
(The initial setup logs showed this, not sure if it is of any use)

```
zigbee-herdsman:controller:log Permit joining +1m
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0} +1m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228] +1m
  serialport/stream _write 10 bytes of data +1m
  serialport/binding-abstract write 10 bytes +1m
  serialport/bindings/unixWrite Starting write 10 bytes offset 0 bytesToWrite 10 +1m
  serialport/bindings/unixWrite write returned: wrote 10 bytes +1ms
  serialport/bindings/unixWrite Finished writing 10 bytes +0ms
  serialport/stream binding.write write finished +1ms
  serialport/bindings/poller received "readable" +1m
  serialport/binding-abstract read +12ms
  serialport/bindings/unixRead Starting read +1m
  serialport/bindings/unixRead Finished read 6 bytes +0ms
  serialport/stream binding.read finished { bytesRead: 6 } +11ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82] +1m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +1m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  serialport/stream _read reading { start: 16159, toRead: 49377 } +0ms
  serialport/binding-abstract read +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +1ms
  serialport/bindings/poller received "readable" +8ms
  serialport/binding-abstract read +9ms
  serialport/bindings/unixRead Starting read +8ms
  serialport/bindings/unixRead Finished read 8 bytes +0ms
  serialport/stream binding.read finished { bytesRead: 8 } +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0} +1m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  serialport/stream _read reading { start: 16167, toRead: 49369 } +1ms
  serialport/binding-abstract read +1ms
  serialport/bindings/unixRead Starting read +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +1ms
```

magnayn, 2020-01-11:
I'm somewhat confused by the zigbee 3.0 vs zigbee 1.2 stuff - maybe this is part of the issue?

I have a 1.2 firmware on my CC2531 (I tried the 3.0.x firmware and it did not seem to be able to pair with anything at all).

Koenkk, 2020-01-13:
This is probably not part of the issue, I think the device requires the additional configure which you mentioned in https://zigbee2mqtt.discourse.group/t/philips-hue-smart-button-not-working/751.

After adding this to `devices.js` you should be able to configure it, but make sure while configuring it the device is awake (by pressing e.g. a button on it).

magnayn, 2020-01-20:
Ah, I think pairing works (the title of this ticket is probably not quite right), it's just unresponsive after that (same as OP: "Pairing looks like it is successful in the logs, but pressing the button doesn't light the indicator LED or send any updates to mqtt.").

It feels like yes, it needs some sort of configuration. I can do the software coding bit, but I'm drawing a blank as to what that device needs in order to work correctly...

magnayn, 2020-01-25:
So, digging and trying to learn as I go. The database shows this:

```json
 "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 2096,
      "inClusterList": [
        0,
        1,
        3,
        64512,
        4096
      ],
      "outClusterList": [
        25,
        0,
        3,
        4,
        6,
        8,
        5,
        4096
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "ROM001",
            "manufacturerName": "Philips",
            "powerSource": 3,
            "zclVersion": 2,
            "appVersion": 2,
            "stackVersion": 1,
            "hwVersion": 1,
            "dateCode": "20190619",
            "swBuildId": "2.21.0_r29784"
          }
        }
      },
      "binds": []
    }
  },
```

I have tried to make it configure by adjusting the devices.js thus:

```js

  {
        zigbeeModel: ['ROM001'],
        model: '9290022230',
        vendor: 'Philips',
        description: 'Hue smart button',
        supports: 'action',
        fromZigbee: [fz.command_on, fz.command_off_with_effect, fz.SmartButton_skip, fz.battery_percentage_remaining],
        toZigbee: [],
         meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint1 = device.getEndpoint(1);
            await bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);

        },
```

But nothing seems to happen. I hard-reset the button (hold down reset for 10 secs) - it announces itself (enough to pair) but then nothing futher.

Any hints?

Koenkk, 2020-01-26:
Is this the device: https://www2.meethue.com/en-us/p/hue-smart-button/046677553715?

We need to figure out how to configure it, do you have the original hue bridge to sniff traffic?  (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

magnayn, 2020-01-30:
It is, I don't have a hue bridge unfortunately so I can't sniff.

a patch for this project (
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2077)
diffs.txt
<https://github.com/dresden-elektronik/deconz-rest-plugin/files/3876449/diffs.txt>
seems
to imply it's close to a RWL02 device - but, that seems to use endpoint 2,
where I tried
doing that I got a null pointer (could it be an active endpoint that's not
declared by it?)

Koenkk, 2020-01-30:
Can you try with 

```js
{
    zigbeeModel: ['ROM001'],
    model: '9290022230',
    vendor: 'Philips',
    description: 'Hue smart button',
    supports: 'action',
    fromZigbee: [fz.command_on, fz.command_off_with_effect, fz.SmartButton_skip, fz.battery_percentage_remaining],
    toZigbee: [],
    meta: {configureKey: 4},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint1 = device.getEndpoint(1);
        await bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);

        const endpoint2 = device.getEndpoint(2);
        const options = {manufacturerCode: 0x100B, disableDefaultResponse: true};
        await endpoint2.write('genBasic', {0x0031: {value: 0x000B, type: 0x19}}, options);
        await bind(endpoint2, coordinatorEndpoint, ['manuSpecificPhilips', 'genPowerCfg']);
        await configureReporting.batteryPercentageRemaining(endpoint2);
    },
},
```

magnayn, 2020-01-31:
I tried various things like pressing the button to make it awake,
resetting, forcing configure but I get the following:

zigbee2mqtt:error 2020-01-31 11:57:52: Failed to configure 'SmartButton',
attempt 4 (Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout._onTimeout
(/home/magnayn/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7))

Koenkk, 2020-01-31:
Please try with the latest dev branch, I've improved the error logging there to get a better understanding where things go wrong.

magnayn, 2020-02-01:
I've got a fuller log, which I'll attach in a moment.

The similar log line though is now:

  zigbee-herdsman:adapter:zStack:unpi:parser <--
[254,3,69,161,253,98,0,120] +52ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext
[254,3,69,161,253,98,0,120] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 161 -
[253,98,0] - 120 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - bindRsp -
{"srcaddr":25341,"status":0} +52ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
zigbee2mqtt:error 2020-02-01 11:48:05: Failed to configure 'SmartButton',
attempt 4 (TypeError: Cannot read property 'write' of undefined
    at Object.configure
(/home/magnayn/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:1738:25)
    at async DeviceConfigure.configure
(/home/magnayn/zigbee2mqtt/lib/extension/deviceConfigure.js:95:13))
winston:file written true false
  serialport/stream _read reading { start: 1607, toRead: 63929 } +1ms
  serialport/binding-abstract read +54ms
winston:file logged 16404 error 2020-02-01 11:48:05: Failed to configure
'SmartButton', attempt 4 (TypeError: Cannot read property 'write' of
undefined
    at Object.configure
(/home/magnayn/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:1738:25)
    at async DeviceConfigure.configure
(/home/magnayn/zigbee2mqtt/lib/extension/deviceConfigure.js:95:13))

magnayn, 2020-02-01:
[debug.log](https://github.com/Koenkk/zigbee2mqtt/files/4142496/debug.log)

Koenkk, 2020-02-01:
Are you sure you are using my updated code from https://github.com/Koenkk/zigbee2mqtt/issues/2625#issuecomment-580403474 (it uses endpoint 1 instead of 2 now. I'm asking because it fails at `await endpoint1.write` (because `endpoint1` = `null`), but this is impossible as it is already used before.

magnayn, 2020-02-01:
Ah, sorry - I hadn't noticed that the patch had changed.

I'm happy to report though that it seems to work! 🎆 

Thankyou for this, the hue button is a really neat little controller which will signficantly improve my home WAF :)

Koenkk, 2020-02-01:
@magnayn great, so the button works as expected now?

Btw which button is this? https://static2.shortcutlabs.com/wp-content/uploads/2017/09/fille.png or https://www.assets.signify.com/is/image/PhilipsLighting/113464f0f77442aeb7e6aaa000ea3b06?wid=1280&hei=1280&$jpglarge$

magnayn, 2020-02-01:
Yes - it reports both action (on, off) and fade up/down, and link quality
(I haven't seen battery yet but I haven't been looking too closely)

Yes - it's that button (the exact one is this:
https://www.amazon.co.uk/Philips-Button-Wireless-Control-Assistant/dp/B07SQZYYKL)
- it may be a very slightly different SKU as the baseplate is square (UK)
rather than the rectangle (US) style, but the button itself looks the same.

Koenkk, 2020-02-01:
Awesome, I will add it to the dev branch :smile:

closed 2020-02-01 by Koenkk