Koenkk/zigbee2mqtt #2626: Bticino living now smart

CatMonster, 2019-12-28:
Hi,
In Italy there is this new product which is al lot promising from Bticino, especially if you want to retrofit a not so much old electrical system by adding neutral wires where is needed, instead of re-doing all new wiring.

https://catalogo.bticino.it/prbt/bticino/sistemi-per-edifici-residenziali/living-now/living-now-smart

The problem is that the app is limited and would be cool to integrate it inside home assistant.

Questions of a zigbee noob:
- Any support for this devices is possible?
- How they can be put them into pairing mode? On promotional videos it seems that their gateway is required. I bought this module https://catalogo.bticino.it/BTI-K4531C-IT, which is basically a relay and a load meter, and can be connected to zigbee network; tapping on it, shaking or pressing the button, hasn't triggered nothing on zigbee2mqtt, not a unsupported message even.

nldroid, 2019-12-28:
Are you sure it's zigbee? I see two series on the site: living now and myhome. The myhome has some zigbee in the description. Living now hasn't. Living now has RF on the switch. I associate that with 433 Ghz switches.

CatMonster, 2019-12-28:
My 

> Are you sure it's zigbee? I see two series on the site: living now and myhome. The myhome has some zigbee in the description. Living now hasn't. Living now has RF on the switch. I associate that with 433 Ghz switches.

My home is an old system and runs on bus. The new one, living now smart is based on zigbee as they said.

CatMonster, 2019-12-28:
Here a simple demo of the system: https://www.youtube.com/watch?v=F8DhsWFWS4U.
The guy connects all modules to the gateway without the app and touching any device, only by pressing an external button.
Auto translations are not too broken, only for first couple minutes.

nldroid, 2019-12-29:
I find it very hard to find any confirmation that it's Zigbee. I see some Zigbee mentions on their site but not in combination with "Living Now"
The best thing you can do is to try to sniff the packages: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

keyita, 2019-12-30:
Hi, I can confirm it is Zigbee
I have joined today a couple of K3004C to my network with really base configuration following
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Since Bticino is part of Legrand group, see #2550 and #2399 work by @cemonneau.

cemonneau, 2019-12-30:
Hi, 
Thanks for your feedback, I never heard the name of the brand before but it's good to know the work should be compatible. Do they have other model id/names @keyita ? I could add that to my pushs and PR.

keyita, 2019-12-31:
Hi @cemonneau,

Thanks, that would be great and really nice by you!
Since I am new with zigbee2mqtt, please correct me if I am not providing the right information:

I have paired the device by adding the new device as 
 `zigbeeModel: [' Light switch with neutral\u0000\u0000\u0000\u0000\u0000'],`
and 
`onEvent: async (type, data, device) => {
            await legrand.pairing_security_event(type, data, device);
 },`
My understanding is that this is the only attribute relevant for matching the device?

Maybe the following line from my database.db should be usefull too:
`{"id":12,"type":"Router","ieeeAddr":"0x000474##########","nwkAddr":######,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Light switch with neutral\u0000\u0000\u0000\u0000\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,6,4,5,15,64513],"outClusterList":[0,64513,25],"clusters":{"genBasic":{"attributes":{"modelId":" Light switch with neutral\u0000\u0000\u0000\u0000\u0000","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":2,"appVersion":0,"stackVersion":65,"hwVersion":1,"dateCode":" \u000020191002\u0000\u0000\u0000\u0000\u0000","swBuildId":"00b\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000 \u0002\u0000\u0001\u0000 \u0002\u0000\u0002\u0000!\u0012\u0000\u0000\u0000\u0000��\u0003\u0000\u0010\u0002\u0001\u0004\u0000\u0018\u0002\u0000��!"}},"genBinaryInput":{"attributes":{"presentValue":0}},"genIdentify":{"attributes":{"identifyTime":60}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b00XXXXe028","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00XXXXe028","endpointID":1},{"cluster":15,"type":"endpoint","deviceIeeeAddress":"0x00124b00XXXXe028","endpointID":1}]}},"appVersion":0,"stackVersion":65,"hwVersion":1,"dateCode":" \u000020191002\u0000\u0000\u0000\u0000\u0000","swBuildId":"00b\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000 \u0002\u0000\u0001\u0000 \u0002\u0000\u0002\u0000!\u0012\u0000\u0000\u0000\u0000��\u0003\u0000\u0010\u0002\u0001\u0004\u0000\u0018\u0002\u0000��!","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1577673957321}`

I am still playing around with toZigbee and fromZigbee configurations: It is a two way switch and I did not achieve yet a fully working configuration.
product link: [https://catalogo.bticino.it/app/webroot/low_res/491606_492825_LE10356AC.pdf](url)
https://catalogo.bticino.it/app/webroot/low_res/491606_492825_LE10356AC.pdf

the last part of the link "LE10356AC" seem familiar with some codes of the Céliane serie, but i think there are no 1-to-1 matches on models id.

cemonneau, 2020-01-01:
Hi @keyita 
I looked briefly at the PDFs and your code.
The part I was refering to was ```
zigbeeModel: [' Light switch with neutral\u0000\u0000\u0000\u0000\u0000'],
```
so that's good. Have you tested to send commands ? What are the physical "buttons" are characteristics (rgb LED ? on/off ? ).

I think it's identical to the product "Dimmer switch w/o neutral" we have here in france/"raw" Legrand.

Could you test that code (please ensure you have the modification from https://github.com/Koenkk/zigbee-herdsman-converters/pull/829/files applied and replace the part by the one bellow).
It should enable on/off commands, and LED control (mainly).

```
        zigbeeModel: [
            ' Dimmer switch w/o neutral\u0000\u0000\u0000\u0000\u0000',
            ' Light switch with neutral\u0000\u0000\u0000\u0000\u0000'
        ],
        model: '067771',
        vendor: 'Legrand',
        // led blink RED when battery is low
        description: 'Wired switch without neutral',
        supports: 'on, off',
        fromZigbee: [
            fz.identify,
            fz.on_off,
        ],
        toZigbee: [
            tz.on_off,
            tz.legrand_settingAlwaysEnableLed,
            tz.legrand_settingEnableLedIfOn,
            tz.legrand_settingEnableDimmer,
            tz.legrand_identify,
        ],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'genLevelCtrl', 'genBinaryInput']);
        },
        onEvent: async (type, data, device) => {
            await legrand.pairing_security_event(type, data, device);
        },
    },
```

Please report back once you have tested and I will gladly add the zigbeeModel to my push

cemonneau, 2020-01-01:
EDIT: With the name I dont think your product support dimming. I have (yet) not tested that part that's why the code above should work even if the zigbeeModel I have should support it (but requires an OTA with the official gateway)

keyita, 2020-01-01:
Hi @cemonneau,

This was the complete code I added to devices.js in my really first attempt to send commands when I found your pushes some days ago, I did in fact copy your configuration from the dimmer switch, disabling dimmer part :
`{
        zigbeeModel: [' Light switch with neutral\u0000\u0000\u0000\u0000\u0000'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
        model: 'K3004C', // Vendor model number, look on the device for a model number
        vendor: 'BTicino', // Vendor of the device (only used for documentation and startup logging)
        description: 'Light switch with neutral', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
        supports: 'on, off, led color', // Actions this device supports (only used for documentation)
        fromZigbee: [
            fz.identify,
            fz.on_off,
            //fz.state,
        ],
        toZigbee: [
            tz.on_off,
            tz.legrand_settingAlwaysEnableLed,
            tz.legrand_settingEnableLedIfOn,
            tz.legrand_identify,
        ],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'genBinaryInput']);
        },
        onEvent: async (type, data, device) => {
            await legrand.pairing_security_event(type, data, device);
        },
    },`

and configuration.yaml:
  ```
'0x0004740000XXXXXX':
    friendly_name: 'switchk'
    permanent_led: true
```

after restart the led from red turned to green, and blinked blu during actions):

So I think I have already tested that configuration.

```
info  2019-12-30 03:45:17: switchk (0x0004740000XXXXXX): K3004C - BTicino Light switch with neutral (Router)
info  2019-12-30 03:45:54: MQTT publish: topic 'zigbee2mqtt1/switchk', payload '{"linkquality":52,"state":"ON","action":"identify"}'
info  2019-12-30 03:45:54: MQTT publish: topic 'zigbee2mqtt1/switchk', payload '{"linkquality":34,"state":"OFF"}'
```

The device has a single command button that commutes the phase(L) between the 2 outputs (two-way-switch/va-et-vient) so one of the 2 output is always 'ON', and a little reset/paring button (not accessible when the cover is mounted/during normal operation) + LED. As you can see the device always reports back the state of OFF (but the light is ON)

I am now downloading the ZBOSS utilities to sniff some packets and try to understand what the devices is sending. I hope it sends the output active (1 or 2) and something usefull to understand if there is any watt consumption (means the connected light is on).

Looking at [https://www.legrand.fr/pro/catalogue/42556-version-celiane-with-netatmo/interrupteur-filaire-connecte-avec-option-variateur-celiane-with-netatmo-sans-neutre-5w-a-300w-compensateur-titane] video (paused at 2:07) looks also the 067771 has two outputs.

My setup is like the one described at https://www.legrand.fr/catalogue/interrupteur/interrupteur-ou-va-et-vient-simple-avec-plaque-mosaic-blanc video (from 0:26 ca).

Thanks, I am really appreciating your support.

cemonneau, 2020-01-01:
Really good feedback @keyita 

Warning, long answer ^^

What are you using as coordinator ? (Cc2531?) And which version too (1.2 or 3.0). I'm collecting info on versions for the doc as some things seems to not work (or harder) on some versions/firmware. Can you test another channel too ? (Only 11 working on my side on all products I have). 
If you could test those 3 transformers too (even if it's not the primary use) 
            tz.legrand_settingAlwaysEnableLed (led always on)
            tz.legrand_settingEnableLedIfOn (led on when using it : on shutter while its moving, on a switch when the power is on, that's what I found yet, feedback welcome on 2 way switches)
            tz.legrand_identify (send an action to the led : permanent during 4 seconds, blink 3 or 4 times, and the color too for the animation).
Devices I have also have the reset button behind but nothing on Legrand manuals, had to try. Unmount is easy when you know how, I will probably make a video.
Do you have the official gateway too and did you receive an OTA on your devices ? (Will be useful for the doc). I have it and some devices received one (I don't put everything on the test network) it once broke something, seems more stable with the 3.0 version (see some of the latest comment I made on the Legrand issue).

I don't have 2 way switches (only wireless ones on production) but looking at your logs the on/off state is reported (so endpoint 1 is used for the "main" state). The 2nd state might be on endpoint 2, but it may be required to subscribe to it too (don't know how even if I red most of the code). I'm not too used to 2 way on the electric side so I will have to check if you need help on that. 

When you will have zboss, if you can push 1 second the reset button, check if it sends a permit join request or something similar. Some devices do that, others sends an identify. 

Side question, do you have more devices ?

keyita, 2020-01-03:
Hi @cemonneau,

long question... long answer ^^

It seems I had a bad setup while repositioning my coordinator during testing, I added a 30cm usb cable and now everything is working as expected, out of the box

Both my K4003C are responding correctly to ON OFF commands and reports back correct state even when switching ON/OFF from non smart switches  (two way wiring), no editing to your code needed for this, no additional endpoint needed.

So it really work! Thanks!

Led feedback:
tz.legrand_settingAlwaysEnableLed (led always on): it does what described on your comment but the color is GREEN in my case, ON only if in pairing mode
tz.legrand_settingEnableLedIfOn: On when power is On, perfect! color=BLUE
tz.legrand_identify: I did manage to setup sniffing, so I added as comment the reported payload:

```
                const effects = {
                    'blink3': 0x00,     // reported as Blink, it does blink 3 times :), OK
                    'fixed': 0x01,      // reported as Breathe, OK
                    'blinkgreen': 0x02, // reported as Okay, it blinks 1 time green, OK
                    'blinkblue': 0x03,  // reported as Unknown, it blinks 5 times blue, OK
                };
                // only works for blink3 & fixed
                const colors = {
                    'default': 0x00,    // OK = Blue
                    'red': 0x01,        // OK
                    'green': 0x02,      // OK
                    'blue': 0x03,       // OK
                    'lightblue': 0x04,  // OK
                    'yellow': 0x05,     // half yellow half red
                    'pink': 0x06,       // NO fallback to 0
                    'white': 0x07,      // half white half red
                };

```
With an exception for pink, all is working (white and yellow work, but are really bad looking).
On Living Now devices the reset button is easy to access: the device comes "naked", you have to mount the cover (sold separately). It sends a permit join request as you suggested, led turns green (with permit_join at false otherwise no noticeable).
```
ZigBee Device Profile, Permit Join Request
    Sequence Number: 152
    Duration: 180
    Significance: 1
```
Version info: 1.8.0-dev / docker
Coordinator: CC2531 / Stack: 3.0
Channel: 11
```
MQTT publish: topic 'zigbee2mqtt1/bridge/config', payload '{"version":"1.8.0-dev","commit":"unknown","coordinator":{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}},"log_level":"info","permit_join":false}'
```
I do not have the official gateway, most of my lights are controlled by shelly2.5,but since I have still some lights to cover in order to make them all smart I bought this device because it fits with no additional space needed and it extend the coverage of the zigbee network ( both on hallway ). 
So I did not recive any OTA, but if I am not wrong the database.db line (from my 2nd post) contains 20191002, and a swBuild 

All other zigbee devices I own are sensor endopoints and a spare CC2531 as router ( now used as sniffer device, with those switches as router I will keep it as sniffer), but now I will probably buy some more from Bticino, to add some wireless switches.
I have also tested and confirm working the following
```
zigbeeModel: [
            ' Dimmer switch w/o neutral\u0000\u0000\u0000\u0000\u0000',
            ' Light switch with neutral\u0000\u0000\u0000\u0000\u0000'
        ],
```

I wanted to set the led blue on always (I am not on battery and all my switches on hallway have led) , so I have added this converter that worked for me, led is blue:
```
legrand_settingEnableLedIfOff: {
        key: ['led_when_off'],
        convertSet: async (entity, key, value, meta) => {
            // enable or disable the LED (blue) also when light is OFF
            const enableLedIfOn = value === 'ON' || !!value;
            const payload = {3: {value: enableLedIfOn, type: 16}};
            await entity.write('manuSpecificLegrandDevices', payload, options.legrand);
        },
    },
```
I would really appreciate if you can include also this.

Thanks!

keyita, 2020-01-03:
@cemonneau, seems legrand_settingEnableLedIfOff works only if followed by a legrand_settingAlwaysEnableLed, now is always blue, turns green while pairing then blue again when pairing is off

cemonneau, 2020-01-03:
Hi @keyita 

I will see to include what's need in my PR in the next few days.
Regarding the LED, it's state is supposed to be green when the network is in pairing mode, whatever config is set. "tz.legrand_settingAlwaysEnableLed" is supposed to set the LED in permanent blue when permit_join=false so legrand_settingEnableLedIfOff should not be added BUT tz.legrand_settingAlwaysEnableLed should work. 

The problem is the written attribute is not the right one but Legrand/Netatmo seems to not be constant on their attribute values, I will see to add probably something like "tz.legrand_settingAlwaysEnableLed_1" and "tz.legrand_settingAlwaysEnableLed_3", feedback welcome @Koenkk on the nomenclature.

Other than that I think it's good, just needs to add that to my PR

cemonneau, 2020-01-03:
@keyita 
Could you add to your devices.js file the following part (for your device)
```
toZigbee: [
{
	key: ['legrandRawSettings'],
	convertSet: async (entity, key, value, meta) => {
		const options = {manufacturerCode: 0x1021, disableDefaultResponse: true};
		const payload = {};
		payload[value.attribute] = {value: value.value, type: value.type};
		await entity.write('manuSpecificLegrandDevices', payload, options);
	},
},
]
```
And then test with mqtt with payloads like
```
{"legrandRawSettings":{"attribute":0,"value":REPLACE_HERE,"type":16}}
```
It's a custom transformer I have to test setting attributes. 
Attribute is the attribue number  
Value is the value we set  
Type is the type we set (16 or 9 currently)  
At the same time use zboss to look at the raws response and see which attribute reports ok or not & existing or wrong value type.
If you can find the physical difference it would be great too, I would add the support (if you find more attributes).
Currently you saw the LED attributes but the 0 is used on some devices to set (enable) the dimming (requires an OTA update on some). I dont think 2 way switches have dimming (or anything else), but well, we never know.

keyita, 2020-01-03:
Hi @cemonneau ,

tested legrandRawSettings,  with attribute 0
```
Status Record
    Status: Unsupported Attribute (0x86)
    Attribute: 0x0000

```
I started by resetting one of the device and my bad, you were right with legrand_settingEnableLedIfOff there is no need of legrand_settingAlwaysEnableLed_3 your original legrand_settingAlwaysEnableLed works.

cemonneau, 2020-01-09:
@keyita setting fix will be in that PR https://github.com/Koenkk/zigbee-herdsman-converters/pull/891

closed 2020-01-12 by Koenkk

blackwiz4rd, 2020-01-14:
Hi guys, amazing work. Just a note, the correct model number he is referring to is K4003c and not K3004C. Looking forward to install a living now system with zigbee2mqtt!
Thanks ;)

ster1um, 2020-01-17:
What is the right model?
From supported devices the photo is K4531C!
The K3004C do not exist.

ster1um, 2020-01-17:
is the right model?
From supported devices the photo is K4531C!
The K3004C do not exist.

keyita, 2020-01-18:
@blackwiz4rd @Koenkk - K4003C is the right model name, my typo.
@ster1um you are right, the image on support is referring to the model of the OP (K4531C)
I do not own that model, I own the K4003C (https://catalogo.bticino.it/BTI-K4003C-IT)
I have joined the topic because there was a doubt regarding "bticino living now" being zigbee at all.

When you sniff packets with wireshark,  Legrand is recognised as the manufacter:
    [Extended Source: Legrand_00:XX:XX:XX:XX (00:04:74:00:XX:XX:XX:XX)]

K4531C will be probably recognised when you join it as one of the legrand model codes already supported by @cemonneau, as he recently added:

```
         zigbeeModel: [
             ' Connected outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000' +
             '\u0000\u0000\u0000\u0000\u0000',
         ],
         model: '067775',
         vendor: 'Legrand',
         description: 'Power socket with power consumption monitoring'

```
This should cover also the model in the OP, it would be nice if someone with K4531C can try.
BR.

blackwiz4rd, 2020-01-20:
> @blackwiz4rd @Koenkk - K4003C is the right model name, my typo.
> @ster1um you are right, the image on support is referring to the model of the OP (K4531C)
> I do not own that model, I own the K4003C [...]

> @blackwiz4rd @Koenkk - K4003C is the right model name, my typo.
> @ster1um you are right, the image on support is referring to the model of the OP (K4531C)
> I do not own that model, I own the K4003C [...]

Thanks for pointing that out. I will try to add support for the 4027C, which is used for shutters. I already saw a similar product, which is already supported by Legrand. I will report soon when I buy it.

cemonneau, 2020-01-20:
@blackwiz4rd ping me if you need help. I'm getting used to how some things may be with Legrand. I'm also collecting data for the wifi

ChumKiu, 2020-02-06:
Hi all,
I'm just starting today to work with this switch and after some time I've found this thread.

The device is actually supported but I kindly ask to better understand the things, at my first attempt I've logged the ZigBee communication with Wireshark and after the _Active Endpoint Request_ and related _Active Endpoint Response_ the system send to the switch the _Node Descriptor Request_ instead the _Simple Descriptor Request_ that is used to discover the endpoint structure... then I got the message that the device was not supported from zigbee2mqtt.

Instead try the steps used to add support for new device I've see this thread and then I've updated the zigbee2mqtt to the latest version and now the switch is correctly recognised (green led is on) and now from Wireshark I can see that the _Simple Descriptor Request_ is sent just after the _Active Endpoint Request_ as expected, so actually the question is solved but I kindly ask some details about the zigbee2mqtt sequence of operations.

My approach with every new device (supported or not) is to look at the Wireshark log during the commissioning stage in order to know how the device behave and I've noticed during the test that the _Simple Descriptor Request_ was not issued as usual, could you please let me know where this sequence of operation is addressed? I'm try to better understand all the things around this nice library in order to support more devices, is this related some other issue and I'm looking at the wrong place?

Thank for your time!

keyita, 2020-02-10:
Hi @ChumKiu,

Led behavior during pairing mode works as you described.
When not in pairing mode, depends on your settings: default is what you described.
If you enable tz.legrand_settingEnableLedIfOn then the led will stay on (blue) when the light is on.
If you enable tz.legrand_settingAlwaysEnableLed then the led will stay on also when the light is off.

I am away from the switches, but this should work
> topic: zigbee2mqtt/[FRIENDLY_NAME]/set 
> payload: {"led_when_on":"ON", "permanent_led":"ON"}

Beside that, K4003C is the model you are referring right?

BR.

blackwiz4rd, 2020-03-05:
@keyita
Hi, I need some help to setup the K4003C. Zigbee2mqtt doesn't seem to recognize it. The K4003C works by pressing it manually and the led has a fixed red color. However, it doesn't show up in the logs. I am sure that my zigbee2mqtt setup (raspberry+CC2531) is correct, because I have one OSRAM led connected to it and it works properly.
PS: The first time I have turned on the general current, the light turned on and off a few times by itself and during that period Zigbee2mqtt had recognized the led (I have read the logs).
I can post the logs here or in a new issue.

keyita, 2020-03-08:
Hi @blackwiz4rd 

Try to reset the switch while zigbee2mqtt is in pairing mode.
If nothing happen try by switching off and than back on the main (general current) first.

I remember reading that "every powered devices from Legrand ask every other devices on the network since how long there are poweron, and if the value is too high, it leaves the network.
I think it's a security to force people to shutdown their electric installation when doing the installation."
https://github.com/Koenkk/zigbee2mqtt/issues/2399#issuecomment-562859323

Reset button location is under the cover.
https://www.catalogo.bticino.it/app/webroot/low_res/491606_492825_LE10356AC.pdf
Image on top, point 2

BR.

blackwiz4rd, 2020-03-10:
> Hi @blackwiz4rd
> 
> Try to reset the switch while zigbee2mqtt is in pairing mode.
> If nothing happen try by switching off and than back on the main (general current) first.
> 
> I remember reading that "every powered devices from Legrand ask every other devices on the network since how long there are poweron, and if the value is too high, it leaves the network.
> I think it's a security to force people to shutdown their electric installation when doing the installation."
> [#2399 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/2399#issuecomment-562859323)
> 
> Reset button location is under the cover.
> https://www.catalogo.bticino.it/app/webroot/low_res/491606_492825_LE10356AC.pdf
> Image on top, point 2
> 
> BR.

Thanks for the suggestions. Here is what I did for successful pairing:

1. Stopped zigbee2mqtt
2. Turned off lights only from the main socket (this allows my raspberry to keep power on)
3. Turned on zigbee2mqtt with `permit_join: true`
4. Turned on lights from the main socket
5. Pairing is successful (I suggest to keep on every other Zigbee device for the first time pairing)

marcgarciamarti, 2020-03-15:
@CatMonster have you by any chance integrated the K4003C in a Bticino LivingLight series wall switch?

blackwiz4rd, 2020-03-15:
> @CatMonster have you by any chance integrated the K4003C in a Bticino LivingLight series wall switch?

I did, you just need to follow what has been written above for pairing.

marcgarciamarti, 2020-03-15:
@blackwiz4rd I'm on the verge of crying. I've been searching for a decent solution for months (years even) to no avail... amazing!!! I hope it does not become a nightmare running neutral wires around the place :( Xiaomi has a product that can run without neutral :(
is there any battery powered unit? that would be great too!!

blackwiz4rd, 2020-03-15:
> @blackwiz4rd I'm on the verge of crying. I've been searching for a decent solution for months (years even) to no avail... amazing!!! I hope it does not become a nightmare running neutral wires [...]

There isn't a solution from bticino without neutral, however if you are close to a socket it is quite easy to run the wires ;).

marcgarciamarti, 2020-03-21:
Hello, I regret to confirm that the K4003C is unfortunately incompatible with Bticino Livinglight series. I was hoping it would be easy to integrate it into my current wall switches line but it does not work :(

blackwiz4rd, 2020-03-21:
> Hello, I regret to confirm that the K4003C is unfortunately incompatible with Bticino Livinglight series. I was hoping it would be easy to integrate it into my current wall switches line but it does not work :(

Sorry, I did a huge mistake when reading your question. It is not compatible with the Livinglight if you wish to keep the same plate, supports and so on. But, it is compatible with the same wiring system, that is what I meant.

maxcanna, 2020-07-01:
@marcgarciamarti If you have the (more common) living light serie you can use [L4003C](https://catalogo.bticino.it/BTI-L4003C-IT) (pay attention to the first letter identifying the product serie)

marcgarciamarti, 2020-07-01:
> @marcgarciamarti If you have the (more common) living light serie you can use [L4003C](https://catalogo.bticino.it/BTI-L4003C-IT) (pay attention to the first letter identifying the product serie)

Thanks @maxcanna !!
I will look into that! I was not aware of this product at all. Is that compatible with zigbee2mqtt??? Do you use this product?
Thanks a million!!

maxcanna, 2020-07-02:
> Thanks @maxcanna !!
> I will look into that! I was not aware of this product at all. Is that compatible with zigbee2mqtt??? Do you use this product?
> Thanks a million!!

I'm waiting for my zigbee stick to be delivered. I plan to use some of these. I expect full compatibility with the K4003C since the manufacturer is the same. It should be a matter of different enclosure. Anyhow I'd add support for it in case it doesn't work out of the box

marcgarciamarti, 2020-07-15:
> I'm waiting for my zigbee stick to be delivered. I plan to use some of these. I expect full compatibility with the K4003C since the manufacturer is the same. It should be a matter of different [...]

@maxcanna have you managed to test this yet?

maxcanna, 2020-07-15:
> @maxcanna have you managed to test this yet?

I received my zigbee stick on Monday. Now I have to buy a switch to test the compatibility

maxcanna, 2020-07-25:
@marcgarciamarti I tried my L4003C with z2m and it works out of the box. I updated the docs and the name of the device in z2m to reflect this.

maxcanna, 2020-07-25:
@keyita did you use this device as simple switch? I tried it and using `OFF` as payload it doesn't toggle the output (turning off the single connected output). I have to send `TOGGLE` or `ON` again to make it switch output. This way command is not equipollent. I.e. if I send `ON` multiple times I expect it to be on even if it already was but it'd toggle.

I saw that you reported a similar behaviour:

> The device has a single command button that commutes the phase(L) between the 2 outputs (two-way-switch/va-et-vient) so one of the 2 output is always 'ON', and a little reset/paring button (not [...]

(how) did you solve?

marcgarciamarti, 2020-07-27:
> @marcgarciamarti I tried my L4003C with z2m and it works out of the box. I updated the docs and the name of the device in z2m to reflect this.

Would you believe I can't find thus article in Spain? Google just keeps digging up retailers in Italy... I've just contacted bticino in Spain and they are clueless. Multinational nonsenses.

marcgarciamarti, 2020-07-27:
> @keyita did you use this device as simple switch? I tried it and using `OFF` as payload it doesn't toggle the output (turning off the single connected output). I have to send `TOGGLE` or `ON` again [...]

This issue is marked as closed. I don't think you will get any developer attention if you are experiencing any kind of problems.

maxcanna, 2020-07-27:
> Note that this issue is marked as closed. I don't think you will get any developer attention if you are experiencing any kind of problems.

Yeah, I know. Point is I don't know wherever else to ask 😞

keyita, 2020-07-27:
> @keyita did you use this device as simple switch? I tried it and using `OFF` as payload it doesn't toggle the output (turning off the single connected output). I have to send `TOGGLE` or `ON` again [...]

Hi, sorry for late reply.

I own two BTicino connected switches, one wired as simple switch one wired as 2way switch.
In my early testing I had my USB stick connected directly to my laptop, this setup was giving me unexpected behaviours.
By adding a 30cm USB cable zigbee stick stabilized my connection.
After testing I have also reset the devices,as I was testing unsupported commands.

Both switch reports and respond to commands correctly.
I am using them with home assistant.

If you think it helps I can share my payloads.

regards.

maxcanna, 2020-07-27:
> Hi, sorry for late reply.
> 
> I own two BTicino connected switches, one wired as simple switch one wired as 2way switch.
> In my early testing I had my USB stick connected directly to my laptop, this setup was giving me unexpected behaviours.
> By adding a 30cm USB cable to the zigbee stick, stabilized my connection.
> After testing I have also reset the devices as I was playing with unsupported commands.
> 
> Both switch reports and respond to commands correctly.
> I am using them with home assistant.
> 
> If you think it helps I can share my payloads.
> 
> regards.

No worries. 🙂 Today I wired one as two way switch and it worked absolutely fine as you reported. I am curious about using it as single way switch. Did you try it this way too? Does it work? Or I have to buy the single switch/dimmer?

keyita, 2020-07-28:
Yes,

one of my switches is setup as single switch.
Worked out of the box without need of any special configuration.

maxcanna, 2020-07-28:
> Yes, one of my switches is setup as single switch. Worked out of the box without need of any special configuration.

Oh great, so it's smart enough to detect how's wired. 🎉  I tested it with a multimeter before installing it and it gave strage results. TY again

wb441, 2020-08-04:
> Hello, I regret to confirm that the K4003C is unfortunately incompatible with Bticino Livinglight series. I was hoping it would be easy to integrate it into my current wall switches line but it does not work :(

@marcgarciamarti Good news! Bticino launched Zigbee-compatible modules for the LivingLight series. I give the references for the white ones: N4003C (on/off), N4411C (on/off/dim) and N4027C (shutter). 
I ordered them directly from Italy (not distributed in my country), they fit perfectly. I did have to add support for them, but it was a copy paste of existing configuration of Legrand (e.g. for N4411C I copied from Legrand model 067771 ).
Conclusions:
Don't bother with the N4003C. In normal use, the relay is delayed by about a second which feels like poor quality.
The N4411C isn't a lot more expensive and responds immediately to a button press. Optionally you can also activate the dimmer function which isn't for large numbers of bulbs, but will do in most cases.

All in all, it works really well!

marcgarciamarti, 2020-08-04:
> @marcgarciamarti Good news! Bticino launched Zigbee-compatible modules for the LivingLight series. I give the references for the white ones: N4003C (on/off), N4411C (on/off/dim) and N4027C [...]

Thanks for reporting on your experience. I was putting lots of efforts in getting a N4003C exported into my country, but the hefty shipping costs put me off. Where did you get yours from? where are you located? I would be interested in the white series too as my current switches are white.
Could you kindly share your working configuration?

maxcanna, 2020-08-04:
> @marcgarciamarti Good news! Bticino launched Zigbee-compatible modules for the LivingLight series. I give the references for the white ones: N4003C (on/off), N4411C (on/off/dim) and N4027C (shutter).

I wrote it here https://github.com/Koenkk/zigbee2mqtt/issues/2626#issuecomment-652324935 😉  

> I ordered them directly from Italy (not distributed in my country), they fit perfectly. I did have to add support for them, but it was a copy paste of existing configuration of Legrand (e.g. for N4411C I copied from Legrand model 067771 ).

N4003C was already working fine. What did you have to do? 🤔 
I already updated the converter too Koenkk/zigbee-herdsman-converters#1432

> Conclusions:
> Don't bother with the N4003C. In normal use, the relay is delayed by about a second which feels like poor quality.
> The N4411C isn't a lot more expensive and responds immediately to a [...]

You can't use N4003C in place of N4411C since the first is a two way switch and the latter isn't

marcgarciamarti, 2020-08-04:
> Would you believe I can't find thus article in Spain? Google just keeps digging up retailers in Italy... I've just contacted bticino in Spain and they are clueless. Multinational nonsenses.

Purchasing this online and exporting it into my country costs me half of the item price. When I said "Multinational nonsenses" I should have said "Multinational cr*p". Unbelievable. Can somebody give me a tip on how to avoid these hefty shipping costs? Does anybody know of any multinational retailer with commercial presence throughout Europe that could possibly ease the shipping costs?

cemonneau, 2020-08-04:
@marcgarciamarti I just what it would cost if I was shipping with my company from france to spain cost almost 18€ (2kg). Damm that's high, for something that cost around 32€ I feel your pain. Even some of my clients (not in that area) are paying a bit less due to their volume but it's still around thise price. 
You could "maybe" use some online services where it's people doing the transport, I know there is some "p2p" transport companies.

wb441, 2020-08-04:
> 
> You can't use N4003C in place of N4411C since the first is a two way switch and the latter isn't

@maxcanna Correct, I just realized it didn't make any difference for me as I only have one switch per light, but you are definitely right that in a hallway with more than one switch, N4003C is the only possible option. I believe I read somewhere that they are able to detect current, therefore only one N4003C should do.

When talking about adding support, I was referring to N4411C which as of yesterday still needed the quick copy-paste of another Legrand dimmer. Apart from that it's smooth sailing :)

marcgarciamarti, 2020-08-13:
> @marcgarciamarti I just what it would cost if I was shipping with my company from france to spain cost almost 18€ (2kg). Damm that's high, for something that cost around 32€ I feel your pain. Even [...]

I seem to have managed to get ahold of these two models. @wb441 and @maxcanna could you please share your current working configuration so that I can hit the ground running? Can I just use whatever is displayed in z2m documentation? Bad news is I need to run a neutral wire I guess :(

maxcanna, 2020-08-13:
> I seem to have managed to get ahold of these two models. @wb441 and @maxcanna could you please share your current working configuration so that I can hit the ground running? Can I just use whatever is displayed in z2m documentation? 

Yes, buy a CC2531 stick and flashed (if not already flashed). Plug into the pi (or whatever else you're using).  Write your minimal z2m config (you can tune it later but you need to set the right port at least) and start it. Once it's running enable pairing. Now you can wire the switch. On first "boot" they'll automatically join the network so you don't even need to reset them. Once you paired them all you can move them into the wall and finish the setup.

> Bad news is I need to run a neutral wire I guess :(

That's for sure

marcgarciamarti, 2020-08-13:
> Yes, buy a CC2531 stick and flashed (if not already flashed). Plug into the pi (or whatever else you're using).  Write your minimal z2m config (you can tune it later but you need to set the right [...]
I already have some 15 zigbee devices running on z2m. When I asked I was referring to the configuration piece of the yaml specific for these 2 products. If you could help me with that, I would appreciate it.
> 
> 
> 
> > Bad news is I need to run a neutral wire I guess :(
> 
> 
> 
> That's for sure
How do you pair these? Do you do a temporarily wiring near the RPI or can you already put them in the final location?
I would swear that at some point (years ago) I read that one could take the USB stick away from the RPI for pairing purposes... I'm I daydreaming?

maxcanna, 2020-08-13:
Obviously the stick must be plugged and z2m started and join enabled. No need to be very near. Don't put the switch into the wall, configure them using a wire and a plug. It's a lot easier. There's no need to write device specific yaml.

marcgarciamarti, 2020-09-12:
> @marcgarciamarti Good news! Bticino launched Zigbee-compatible modules for the LivingLight series. I give the references for the white ones: N4003C (on/off), N4411C (on/off/dim) and N4027C [...]

Hi there! I just managed today to tackle this with my recent purchase of a 4411. After wiring it following the vendor instructions it seems like the module making the add-on throw warning messages of it not being yet supported:
```
info  2020-09-12 13:47:22: MQTT publish: topic 'zigbee2mqtt/occupancys1xiaomi', payload '{"battery":100,"linkquality":70,"occupancy":true,"voltage":3025}'
info  2020-09-12 13:47:22: Device '0x00047400007e1d74' joined
info  2020-09-12 13:47:22: Starting interview of '0x00047400007e1d74'
info  2020-09-12 13:47:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00047400007e1d74"},"type":"device_connected"}'
info  2020-09-12 13:47:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00047400007e1d74"},"type":"pairing"}'
info  2020-09-12 13:47:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00047400007e1d74"},"type":"device_announced"}'
info  2020-09-12 13:47:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00047400007e1d74"},"type":"device_announced"}'
info  2020-09-12 13:48:19: Successfully interviewed '0x00047400007e1d74', device has successfully been paired
warn  2020-09-12 13:48:19: Device '0x00047400007e1d74' with Zigbee model ' Dimmer switch with neutral' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
info  2020-09-12 13:48:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x00047400007e1d74","supported":false},"type":"pairing"}'
warn  2020-09-12 13:48:29: Received message from unsupported device with Zigbee model ' Dimmer switch with neutral'
warn  2020-09-12 13:48:29: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
info  2020-09-12 13:48:52: MQTT publish: topic 'zigbee2mqtt/occupancys1xiaomi', payload '{"battery":100,"linkquality":70,"occupancy":false,"voltage":3025}'
warn  2020-09-12 13:49:34: Received message from unsupported device with Zigbee model ' Dimmer switch with neutral'
warn  2020-09-12 13:49:34: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
info  2020-09-12 13:51:18: MQTT publish: topic 'zigbee2mqtt/plug2xiaomi', payload '{"consumption":39.35,"linkquality":70,"power":0.5,"state":"ON","temperature":47,"voltage":null}'
info  2020-09-12 13:51:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00047400007e1d74"},"type":"device_announced"}'
warn  2020-09-12 13:52:02: Received message from unsupported device with Zigbee model ' Dimmer switch with neutral'
warn  2020-09-12 13:52:02: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-09-12 13:52:06: Received message from unsupported device with Zigbee model ' Dimmer switch with neutral'
warn  2020-09-12 13:52:06: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-09-12 13:52:14: Received message from unsupported device with Zigbee model ' Dimmer switch with neutral'
warn  2020-09-12 13:52:14: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-09-12 13:52:24: Received message from unsupported device with Zigbee model ' Dimmer switch with neutral'
warn  2020-09-12 13:52:24: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```
after a fresh restart of the add-on I see this:
```
Zigbee2MQTT:info  2020-09-12 14:16:24: wallswitch_bticion_n4411c_1 (0x00047400007e1d74): Not supported (Router)
``
is this expected at all?

maxcanna, 2020-09-12:
> is this expected at all?

Yes, since N4411C is not (yet) supported. See the [provided instructions](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html) to add support for it.

> I purchased the 4003 model but I have not yet given it a try.

I use N4003C and it works fine out of the box.

[14 more replies]