Koenkk/zigbee2mqtt #2627: TRADFRI Dimmer gone wild

kisseler, 2019-12-28:
I successfully integrated a [Tradfri Dimmer IKEA ICTC-G-1](https://www.zigbee2mqtt.io/devices/ICTC-G-1.html) und binded it to a Hue Lamp. It starts to work, but using the dimmer a few times, it starts firing houndres of MQTT messages. The cause of the problem is obviously not the device itself (so no continuos zigbee traffic), because remove the battery does not stop the messages.

Could it be a problem of zigbee2mqtt not getting any kind of "stop message" or confirmation? Ony restart of zigbee2mqtt helps stopping that.

While I'm writing this, I recorded about 8000 MQTT messages with the same topics of my dimmer...

[image]

Koenkk, 2020-01-03:
Please read the notes from https://www.zigbee2mqtt.io/devices/E1744.html#recommendation, however I agree that the current integration sucks, I will refactor it soon (it should work OK out of the box).

kisseler, 2020-01-13:
Thanks, I added that configuration and it solved the problem of multiple messages.

Another problem stays reliability. I binded the dimmer test wise with directly to a hue go lamp. It let's me turn on the lamp directly (for realy dimming the messages range is wrong), but it does not report any status via mqtt anymore. So I suspect, the dimmer disconnected from the zigBee network and only communicates directly to the lamp. Is this behaviour expected when a remote is binded? Or is that a problem of reliability/stability?

About reliability we were discussing in [this thread](https://github.com/Koenkk/zigbee2mqtt/issues/2705), and for me it's a big issue. From my experience so far, I unfortunately cannot reccomend using z2m in a productive environment (although I do so).

Koenkk, 2020-01-16:
This is actually expected, for this you need to use the reporting feature: https://www.zigbee2mqtt.io/information/report.html

Clem-, 2020-02-03:
> Please read the notes from https://www.zigbee2mqtt.io/devices/E1744.html#recommendation, however I agree that the current integration sucks, I will refactor it soon (it should work OK out of the box).

I had the exact same issue and I think it would worth adding the recommendation of E1744 device documentation to the Tradfri Dimmer ICTC-G-1 one?

Koenkk, 2020-02-04:
@Clem- added

closed 2020-04-11 by stale[bot]

dmak, 2020-12-29:
The issue exists in Z2M v1.16.1: even after IKEA dimmer ICTC-G-1 was removed the battery, Z2M continues to send several events per second to MQTT (I provide logs for 2 seconds):
```
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"rotate_left","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'rotate_left'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"rotate_left","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'rotate_left'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"rotate_left","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'rotate_left'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"rotate_left","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'rotate_left'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"rotate_left","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:41: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'rotate_left'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"rotate_left","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'rotate_left'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"rotate_left","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'rotate_left'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"rotate_left","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'rotate_left'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"rotate_left","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'rotate_left'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"rotate_left","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","brightness":2,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-12-29 20:35:42: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'rotate_left'
```
`log_level=debug` for above log. I conclude that Z2M "generates" five identical messages blocks per second (rotation started + rotation finished) without any stop condition. It continued for several hours till I have restated Z2M because of growing log.

Koenkk, 2020-12-29:
@dmak try disabling the legacy integration: https://www.zigbee2mqtt.io/devices/ICTC-G-1.html#legacy-integration

dmak, 2020-12-29:
Thanks, @Koenkk, your advice had helped. However I see that the message always has `"brightness":223, "rate":70` independent on what direction and how quickly I rotate the dimmer, `"linkquality"` is nevertheless changing.

Koenkk, 2020-12-30:
By default the new integration doesn't provide a simulated brightness, you can active it via the simulated brightness option: https://www.zigbee2mqtt.io/devices/ICTC-G-1.html#device-type-specific-configuration. The rate will not be updated anymore, you have to remove it manually from `data/state.json`.

dmak, 2020-12-31:
I have applied the following configuration:
```
'0x000b57fffe1cf13e':
  friendly_name: dimmer
  legacy: false
  debounce: 0.2
  debounce_ignore: [ action ]
  simulated_brightness: true
```
and brightness is now back. I am a bit concerned that brightness jumps from 20 to 255 in this example:
```
info  2020-12-30 23:06:11: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"brightness_move_down","action_rate":195,"battery":74,"brightness":20,"linkquality":7,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 23:06:11: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","battery":74,"brightness":20,"linkquality":7,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 23:06:11: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'brightness_move_down'
info  2020-12-30 23:06:11: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.01,"battery":74,"brightness":255,"linkquality":23,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 23:06:11: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","battery":74,"brightness":255,"linkquality":23,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 23:06:11: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'brightness_move_to_level'
```
and a series of many generated events per second with the same brightness:
```
info  2020-12-30 15:40:42: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"brightness_stop","battery":60,"brightness":0,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:42: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","battery":60,"brightness":0,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:42: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'brightness_stop'
info  2020-12-30 15:40:42: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action_rate":70,"battery":60,"brightness":0,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:43: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"battery":60,"brightness":0,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:43: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"battery":60,"brightness":0,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:43: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"battery":60,"brightness":0,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:43: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"battery":60,"brightness":0,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:43: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"battery":60,"brightness":0,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:44: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"battery":60,"brightness":0,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:44: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"battery":60,"brightness":0,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:44: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"battery":60,"brightness":0,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:44: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"battery":60,"brightness":0,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:44: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"battery":60,"brightness":0,"linkquality":84,"update":{"state":"idle"},"update_available":false}'
```
and for example this case when brightness changes in sequence 20→20→235... I suspect it is hardly possible to achieve reasonable smoothness with this dimmer:
```
info  2020-12-30 15:40:01: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'brightness_move_down'
info  2020-12-30 15:40:01: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"battery":60,"brightness":20,"linkquality":63,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:12: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"brightness_move_down","action_rate":195,"battery":60,"brightness":20,"linkquality":57,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:12: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","battery":60,"brightness":20,"linkquality":57,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:12: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'brightness_move_down'
info  2020-12-30 15:40:12: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"brightness_move_down","action_rate":70,"battery":60,"brightness":20,"linkquality":55,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:12: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","battery":60,"brightness":20,"linkquality":55,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:12: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'brightness_move_down'
[... cut ...]
info  2020-12-30 15:40:13: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"brightness_move_down","action_rate":195,"battery":60,"brightness":235,"linkquality":55,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:13: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"action":"","battery":60,"brightness":235,"linkquality":55,"update":{"state":"idle"},"update_available":false}'
info  2020-12-30 15:40:13: MQTT publish: topic 'zigbee2mqtt/dimmer/action', payload 'brightness_move_down'
```

Koenkk, 2020-12-31:
> and brightness is now back. I am a bit concerned that brightness jumps from 20 to 255 in this example:

This happens when you rotate the remote quickly, you can see that it requests the 255 brightness: `{"action":"brightness_move_to_level","action_level":255`

Note that the value is published on an interval basis which you can configure: https://www.zigbee2mqtt.io/devices/ICTC-G-1.html#device-type-specific-configuration

> I suspect it is hardly possible to achieve reasonable smoothness with this dimmer:

IMHO the only correct way to achieve smoothness is to execute brightness_move/brightness_stop commands on the bulb (in this way you ask the bulb to start dimming down instead of setting an absolute brightness). But this ofc depends if your bulb supports this (most Zigbee ones do). See https://www.zigbee2mqtt.io/devices/F7C033.html#movingstepping