Koenkk/zigbee2mqtt #2631: Recover and repair from Backup

ayushjain4484, 2019-12-29:
HI,

I am running zigbee2mqtt on my raspberry pi 3 B+,
I recently had issue with with rasbian OS.
I reinstall Os on New card and then installed zigbee2mqtt, previously working setup is with my on old card.
just wanted to know is there any way i can recreate my previous network without repairing all devices again as size of my network was very large and are all located in remote locations which are hard to get access to repair.

Thanks in advance.

Kryzek, 2019-12-29:
Copy zigbee2mqtt/**data** -directory (and it's contents) from old SD-card to new card and your configuration (MQTT settings, network keys etc) and devices database should be working after you start Z2M.

ayushjain4484, 2019-12-29:
I tried coping all file to data dir, but seems to be failling after that:

Please find below logs for more details
Dec 29 16:06:42 raspberrypi npm[28043]: > zigbee2mqtt@1.8.0 start /opt/zigbee2mqtt
Dec 29 16:06:42 raspberrypi npm[28043]: > node index.js
Dec 29 16:06:45 raspberrypi npm[28043]: internal/fs/utils.js:220
Dec 29 16:06:45 raspberrypi npm[28043]:     throw err;
Dec 29 16:06:45 raspberrypi npm[28043]:     ^
Dec 29 16:06:45 raspberrypi npm[28043]: Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/data/log/2019-12-29.16-06-45'
Dec 29 16:06:45 raspberrypi npm[28043]:     at Object.mkdirSync (fs.js:823:3)
Dec 29 16:06:45 raspberrypi npm[28043]:     at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:199:16)
Dec 29 16:06:45 raspberrypi npm[28043]:     at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
Dec 29 16:06:45 raspberrypi npm[28043]:     at Object.mkdirSync (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:82:10)
Dec 29 16:06:45 raspberrypi npm[28043]:     at Object.<anonymous> (/opt/zigbee2mqtt/lib/util/logger.js:19:8)
Dec 29 16:06:45 raspberrypi npm[28043]:     at Module._compile (internal/modules/cjs/loader.js:959:30)
Dec 29 16:06:45 raspberrypi npm[28043]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:995:10) {
Dec 29 16:06:45 raspberrypi npm[28043]:   errno: -13,
Dec 29 16:06:45 raspberrypi npm[28043]:   syscall: 'mkdir',
Dec 29 16:06:45 raspberrypi npm[28043]:   code: 'EACCES',
Dec 29 16:06:45 raspberrypi npm[28043]:   path: '/opt/zigbee2mqtt/data/log/2019-12-29.16-06-45'
Dec 29 16:06:45 raspberrypi npm[28043]: }
Dec 29 16:06:45 raspberrypi npm[28043]: npm ERR! code ELIFECYCLE
Dec 29 16:06:45 raspberrypi npm[28043]: npm ERR! errno 1
Dec 29 16:06:45 raspberrypi npm[28043]: npm ERR! zigbee2mqtt@1.8.0 start: `node index.js`
Dec 29 16:06:45 raspberrypi npm[28043]: npm ERR! Exit status 1
Dec 29 16:06:45 raspberrypi npm[28043]: npm ERR!
Dec 29 16:06:45 raspberrypi npm[28043]: npm ERR! Failed at the zigbee2mqtt@1.8.0 start script.
Dec 29 16:06:45 raspberrypi npm[28043]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Dec 29 16:06:45 raspberrypi npm[28043]: npm ERR! A complete log of this run can be found in:
Dec 29 16:06:45 raspberrypi npm[28043]: npm ERR!     /home/pi/.npm/_logs/2019-12-29T10_36_45_673Z-debug.log
Dec 29 16:06:45 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Dec 29 16:06:45 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Dec 29 16:06:45 raspberrypi systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Dec 29 16:06:45 raspberrypi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 8.
Dec 29 16:06:45 raspberrypi systemd[1]: Stopped zigbee2mqtt.
Dec 29 16:06:45 raspberrypi systemd[1]: Started zigbee2mqtt.
Dec 29 16:06:48 raspberrypi npm[28426]: > zigbee2mqtt@1.8.0 start /opt/zigbee2mqtt
Dec 29 16:06:48 raspberrypi npm[28426]: > node index.js

ayushjain4484, 2019-12-29:
> zigbee2mqtt@1.8.0 start /opt/zigbee2mqtt
> node index.js

internal/fs/utils.js:220
    throw err;
    ^

Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/data/log/2019-12-29.17-06-08'
    at Object.mkdirSync (fs.js:823:3)
    at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:199:16)
    at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
    at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
    at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
    at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
    at Object.mkdirSync (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:82:10)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/util/logger.js:19:8)
    at Module._compile (internal/modules/cjs/loader.js:959:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:995:10) {
  errno: -13,
  syscall: 'mkdir',
  code: 'EACCES',
  path: '/opt/zigbee2mqtt/data/log/2019-12-29.17-06-08'
}
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.8.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.8.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-12-29T11_36_08_430Z-debug.log

ayushjain4484, 2019-12-29:
Thanks alot brother, its working now :)
 
I tried coping all file to data dir (configuration.yaml database.db log state.json), had permission issue but after granting permissions to these file its worked.

closed 2020-01-01 by Koenkk

simonwood0609, 2020-11-15:
I'm having trouble restoring a backup using this method and I'm wondering if the permissions issue you described could be the cause.

I've restored the database.db and my configuration.yaml has the same network key, on boot of zigbee2mqtt I can see the devices added, however when I try to toggle one of the light switches I get error **'no network route' (205)**

:(

LuisCanossa, 2021-05-14:
> I'm having trouble restoring a backup using this method and I'm wondering if the permissions issue you described could be the cause.
>
> I've restored the database.db and my configuration.yaml has the [...]

I have the same problem, do you have a solution?