Koenkk/zigbee2mqtt #2632: I can't set sensitivity in Xiaomi Vibration Sensor

kitopopo, 2019-12-29:
Hi, I can't set any value of sensitivity in the xiaomi vibration sensor. I set the sensitivity in the same way of the vibration sensor:

"sudo mosquitto_pub -h 192.168.1.60 -t zigbee2mqtt/0x00158d0002a51xxx/set/ -m "{\"sensitivity\": \"low\"}""

I receive the following error message:

Publish 'set' 'sensitivity' to '0x00158d0002a51xxx' failed: 'Error: Timeout - 48696 - 1 - 28 - 0 - 4 after 10000ms'

I can adjust the sensitivity correctly in the smoke detector but i can not set this value in the vibration sensor. I tried remove the vibration sensor from the zigbe2mqtt network and add this device again. 
My vibration sensor is linked to xiaomi repeater plug, no is directly linked to the coordinator.

Additionally I do not understand why the vibrations of this device are not registered by the log of this device in domoticz but they are registered in the log of the zigbee2mqtt and also in the main log of domoticz, but they are not stored in the log of the device itself in Domoticz

Thanks for you job.

## What happened

I receive a error timeout message 

## What did you expect to happen

Adjust de sensitivity correctly

## How to reproduce it (minimal and precise)

Sending the sensitivity message

## Debug Info

zigbee2mqtt version: 20190608.zip
CC253X firmware version: 1.8.0

Sineos, 2019-12-29:
Worked for me but took me multiple attempts. Try to be very close to your coordinator and push the button every second to keep the device awake.

kitopopo, 2019-12-30:
Hi,

Thanks for your help-. It work for me. Thanks.

Regards

closed 2019-12-30 by kitopopo