Koenkk/zigbee2mqtt #2633: [Feature request] Support for E-Byte E800-DTU Adapter

mcmukkel, 2019-12-29:
Since the E-Byte E800-DTU (http://www.ebyte.com/en/product-view-news.aspx?id=524) ZigBee modem runs on a CC2530 and uses RS-485 interface, i would imagine it could be supported by Zigbee2mqtt on a Raspberry Pi. 
It's a very powerful and fairly priced ($24) adapter, that i believe would be an interesting setup in a Zigbee network.
However, i'm unsure how to set it up at this stage.
I have ordered the hardware, and would like to help with testing anything if needed.

closed 2020-03-05 by stale[bot]

Uozzon, 2020-04-21:
Hi mcmukkel, have you made some progress on the topic ? We are evaluating the same gateway for a project and the integration with zigbee2mqtt would be a great bonus.

mcmukkel, 2020-05-13:
No, unfortunately i didn't look more into it. I however got the hardware and i'm still planning on trying to make it work. 
Did you do an progress?

papperone, 2020-08-19:
sorry to revamp this post but I've seen as well an E800-DTU with ethernet connection so I was assuming it could be used wirth a serial/TCP fetaure of zigbee2mqtt but cannot find any info about it before to purchase; has anyone tested/tried it?

whc2001, 2020-12-01:
Hi, do you still have the device on your hand? I tried to compile the ZStack firmware for it and testing is required.

mcmukkel, 2020-12-15:
Yes i do. I can test it for you.

whc2001, 2020-12-20:
> Yes i do. I can test it for you.

Thanks very much! Unfortunately I am currently on travel and don't have the firmware on my laptop. I will contact you as soon as I go back home.

whc2001, 2021-02-05:
> Yes i do. I can test it for you.

Hi, sorry for the delay, I summerized the modifications to the firmware and compiled the HEX file. Please try it out: https://github.com/whc2001/zigbee2mqtt_E800