Koenkk/zigbee2mqtt #2634: Drayton Wiser TRV home assistant auto discovery

arcanefoam, 2019-12-29:
# Bug Report
## Adding support for DW iTRV home assistant autodiscovery

Since the work on #2521 is now complete, I wanted to add support for home assistant auto discovery. I added the following to homeassistant.js (in lib/extensions/):

```js
// Drayton iTRV
'itrv': {
    type: 'climate',
    object_id: 'climate',
    discovery_payload: {
        state_topic: false,
        current_temperature_topic: true,
        current_temperature_template: '{{ value_json.local_temperature }}',
        temperature_state_topic: true,
        temperature_state_template: '{{ value_json.occupied_heating_setpoint }}',
        temperature_command_topic: 'occupied_heating_setpoint',
        min_temp: 5,
        max_temp: 30,
        temp_step: 0.5,
        modes: ['off','auto'],
        mode_state_topic: true,
          // mode_state_template: "{% set values = { 'Stall':'auto'} %} {{ values[value_json.MOT] if value_json.MOT in values.keys() else 'off' }}",
            // mode_state_template: "{{ % if value_json.MOT == 'Stall' %} auto {% else %} off {% endif %}",
            mode_state_template: "{{ 'auto' }}",
        },
    },
    'binary_sensor_keypad_lock': {
        type: 'lock',
        object_id: 'keypad_lock',
        discovery_payload: {
            state_topic: true,
            command_topic: true,
            payload_lock: 'true',
            payload_unlock: 'false',
            value_template: '{{ value_json.keypad_lockout }}',
        },
    },
    'sensor_boost': {
        type: 'sensor',
        object_id: 'boost',
        discovery_payload: {
            state_topic: true,
            icon: 'mdi:thermometer-alert',
            value_template: '{{ value_json.boost }}',
        },
    },
    'sensor_pi_heating_demand': {
        type: 'sensor',
        object_id: 'heat_demand',
        discovery_payload: {
            state_topic: true,
            icon: 'mdi:fire',
            value_template: '{{ value_json.pi_heating_demand }}',
            unit_of_measurement: '%',
        },
    },
```

## What did you expect to happen

I expected to see the valve as a device with the associated entities, which indeed happened. However, I could not get the climate entity to work. Basically, it seems the mode intermittently changes between `auto` and `unknown`. I tried different approaches for the template (see the commented versions) with no avail.

If I create the different entities manually (via the home assistant configuration.yaml file) it works correctly. I don't know what I am missing in the discovery settings.  

## Debug Info
zigbee2mqtt version: 1.8+

sjorge, 2020-01-02:
> I expected to see the valve as a device with the associated entities, which indeed happened. However, I could not get the climate entity to work. Basically, it seems the mode intermittently changes [...]

It might be it reports between 'off' and 'heat', as this unit does not support 'cool' so it might not end up in 'auto' to indicate it will do either 'heat' or 'cool' depending on the set point.

sjorge, 2020-03-03:
Looking at the to/from zigbee converters, there is no system_mode for this device via hvacThermostat.

Hopefully we can extract a mode of sorts form ALG or ADC in the future

arcanefoam, 2020-03-04:
I don't think the valves report a system mode. I think wiser decided the
valves are by escence heat mode only. What I did in my home assistant is to
use the UI information with the valve state information. "Stall" is mode
"heating" and any other (I think is something like moving to install
position) is mode "Off" as in installation mode no control is done.

closed 2020-05-10 by stale[bot]