Koenkk/zigbee2mqtt #2635: Yale Assure lock not providing status

CanuckianOz, 2019-12-29:
Hey @Koenkk,

I'm still having issues with my Yale lock. It's receiving and executing lock/unlock commands, but not providing status so I'm unable to make much use of it without lock/unlock status. 

Here's the debug log. I've scraped it for just the lock joining and "lock" execution command.

https://hastebin.com/vacavebeya.apache

-CanuckianOz

Koenkk, 2020-01-07:
In your `devices.js` can you change https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L4924 to `meta: {options: {disableDefaultResponse: true}, configureKey: 1},`?

CanuckianOz, 2020-01-07:
Before I do, I’m running Hassio. Do changes to devices.js in the zigbee2mqtt/share folder affect it? I noticed the old zigbee-Shepard-converters config option doesn’t exist anymore.

moistgun, 2020-01-08:
> 
> 
> Before I do, I’m running Hassio. Do changes to devices.js in the zigbee2mqtt/share folder affect it? I noticed the old zigbee-Shepard-converters config option doesn’t exist anymore.

I was able to fix my issue with the yrd226ha2619 by copying the devices.js koenkk linked into my share/zigbee2mqtt folder and making the edit needed, then added `"zigbee_shepherd_devices": true` to the default add-on config.

CanuckianOz, 2020-01-08:
@moistgun thanks! I assume you’re running 1.8.0? Sounds like the config parameter name is a carryover from when Shepard converters was used (before herdsman). 

I’ll give it a go and report back.

moistgun, 2020-01-08:
> 
> 
> @moistgun thanks! I assume you’re running 1.8.0? Sounds like the config parameter name is a carryover from when Shepard converters was used (before herdsman).
> 
> I’ll give it a go and report back.

1.8.0-dev on zigbee2mqtt-edge
If it helps, from the readme on the addon's github 

"set the optional, top-level zigbee_shepherd_devices option to true in your configuration. When set, the add-on will scan your data_path for a devices.js file, and will run zigbee2mqtt using this custom file."

I'm not sure about the name carry over. Zigbee2mqtt has been so stable for me for so long, I haven't had a need to touch it in months 😄

CanuckianOz, 2020-01-08:
@moistgun @Koenkk Great guys, thanks that worked. Issue can be closed. 

Does not work if you do not leave, rejoin the zigbee network and allow the  lock to be configured. Otherwise it receives commands but assumes lock/unlock status based on the command (ie, if you manually operate it, status isn't updated). 

Note the solution is:
1. leave the zigbee network on your lock
2. stop Zigbee2MQTT add-on
3. change the config to include the "zigbee_shepherd_devices": true
4. Update devices.js with the correct line per Koenkk
5. open database.db in a text editor. Delete all references to your lock by searching for its zigbee ID
6. restart zigbee2MQTT
7. enable join devices
8. rejoin your lock to the network

Should now lock/unlock and provide the correct status.

closed 2020-01-08 by CanuckianOz

moistgun, 2020-01-08:
This would constitute as a workaround, but is not directly fixed in the source, no?

Koenkk, 2020-01-08:
@moistgun I've pushed a permanent fix now, will be available in the dev branch in a few hours.

moistgun, 2020-01-12:
A bit of rushed testing led me to believe I was getting all necessary state updates but it seems like I'm only getting state when following a lock or unlock command through the bridge.  If I lock or unlock it manually I get nothing.  @CanuckianOz any chance you're seeing the same thing?   This used to work on this lock just fyi

CanuckianOz, 2020-01-12:
@moistgun yes I had that issue. See my last comment. You need to leave, rejoin and reconfigure the lock. After that it’ll be good.

moistgun, 2020-01-12:
@CanuckianOz thanks I'll try it again. I'm pretty sure I've done it but we'll see!