Koenkk/zigbee2mqtt #2637: Donations

tyler-vincent, 2019-12-30:
I can gather this project is a labor of love for you @Koenkk but aside from the generic sticker at the top of the README there really isn't much direction on how to send some money your way for your hard work. I watch this repo and the related herdsman repos and you put a ton of work into this. I run my entire house on this project and it gets better each time I update.

I'm not suggesting you spam the gh_pages with donation links but I'd suggest that you give appreciative folks like myself a nice and direct way to throw you some cash. If you hate this idea please feel free to close this immediately, and I understand. If you do though let it be known your work is appreciated.

cron0mat, 2019-12-30:
[https://www.paypal.me/koenkk](https://www.paypal.me/koenkk)

sjorge, 2019-12-30:
For me there is also a sponser button on this github project, clicking that will get you Koen's paypal link too.

tyler-vincent, 2019-12-30:
That's what I meant by "generic sticker". I mean if @Koenkk thinks that's enough then that's totally cool. I just felt that something more in the docs would be nice. Took me a while even to find the button.

Koenkk, 2020-01-07:
Added a link to the index of zigbee2mqtt.io, thanks!

closed 2020-01-07 by Koenkk