Koenkk/zigbee2mqtt #2640: Battery info missing after setting cache_state to false

jvdmeent, 2019-12-30:
**zigbee2mqtt version 1.8.0-dev (commit #f04e01f)**

After setting **cache_state** to **false** i noticed that messages are incosistent.
For example i had to set **debounce 2** on my xiaomi temperature config and even then messages are incosistent. Sometimes it has battery info, but most of the time not.

**Log of last 2 hours from temperature sensor:**

info  2019-12-30 11:53:11: MQTT publish: topic 'zigbee2mqtt/0x00158d0001fa6ed0', payload '{"temperature":20.86,"linkquality":131,"humidity":39.56,"last_seen":1577703190252}'

info  2019-12-30 11:53:12: MQTT publish: topic 'zigbee2mqtt/0x00158d0001fa6ed0', payload '{"pressure":1026,"linkquality":131,"last_seen":1577703191573}'

info  2019-12-30 11:53:23: MQTT publish: topic 'zigbee2mqtt/0x00158d0001fa6ed0', payload '{"temperature":20.88,"linkquality":131,"humidity":39.54,"pressure":1026,"last_seen":1577703202001}'

info  2019-12-30 12:11:15: MQTT publish: topic 'zigbee2mqtt/0x00158d0001fa6ed0', payload '{"temperature":20.86,"linkquality":131,"humidity":39.46,"pressure":1026,"last_seen":1577704274824}'

info  2019-12-30 12:19:11: MQTT publish: topic 'zigbee2mqtt/0x00158d0001fa6ed0', payload '{"battery":20,"voltage":2755,"temperature":20.89,"humidity":39.38,"pressure":1026.7,"linkquality":131,"last_seen":1577704750725}'

info  2019-12-30 12:38:56: MQTT publish: topic 'zigbee2mqtt/0x00158d0001fa6ed0', payload '{"temperature":20.85,"linkquality":131,"humidity":39.33,"pressure":1026,"last_seen":1577705935372}'

info  2019-12-30 12:57:19: MQTT publish: topic 'zigbee2mqtt/0x00158d0001fa6ed0', payload '{"temperature":20.88,"linkquality":128,"humidity":39.16,"pressure":1026,"last_seen":1577707038735}'

info  2019-12-30 13:12:10: MQTT publish: topic 'zigbee2mqtt/0x00158d0001fa6ed0', payload '{"battery":20,"voltage":2755,"temperature":20.93,"humidity":39.17,"pressure":1025.9,"linkquality":131,"last_seen":1577707929612}'

info  2019-12-30 13:14:53: MQTT publish: topic 'zigbee2mqtt/0x00158d0001fa6ed0', payload '{"temperature":20.92,"linkquality":128,"humidity":39.06,"pressure":1025,"last_seen":1577708092228}'

info  2019-12-30 13:44:54: MQTT publish: topic 'zigbee2mqtt/0x00158d0001fa6ed0', payload '{"temperature":20.98,"linkquality":131,"humidity":38.96,"pressure":1025,"last_seen":1577709893786}'

sjorge, 2019-12-30:
IIRC the aqara sensors only publish battery info every 24 hours, so if you restart you will have to wait for 24h unless you have cache_state set to true.

jvdmeent, 2020-01-01:
Ok thanks for the info.
I'll set cache_state to true.

closed 2020-01-01 by jvdmeent