Koenkk/zigbee2mqtt #2642: Current docker images with new tags and auto(ish) architecture selection working?

tanus10, 2019-12-30:
Docker image tags have been vastly changed recently, so one cannot choose a image for a specific architecture (ex. linux/arm/v6) 

Problem:
I suspect my Raspberry Pi 1s and zeros are not getting correct image for its armv6 architecture with 
````
docker pull koenkk/zigbee2mqtt:latest
````

Thoughts:
Probably they are getting armv7 image which happens very often.

I don't know about how docker hub distinguishes various architectures, but I know it has a history of not working well, especially between armv6 and armv7 as they have been simply labeled as arm32. That has caused much problem with Rpi zeros and 1s (arm32 v6)

Since there are no tags to specify architecture now days, that is zigbee2mqtts after 1.8.0, I am not sure what else I can try. They are working fine with 1.8.0 and bellow where I can specify image's architecture manually. Rpi2,3 (armv7,8) and Armbian/arm64 works fine with above pull command.

Is there anybody who can confirm this with  Raspberry Pi 1?

Koenkk, 2020-01-05:
This is a known docker bug selecting the wrong image on pi 1 and zero, adding ` --platform arm/v6` to docker pull should fix it.

tanus10, 2020-01-05:
Thank you for the solution.

Is there any  docker-compose equivalent of docker --platform option?

Koenkk, 2020-01-08:
It's definitely a docker issue, I would recommend creating an issue here https://github.com/docker/compose/issues (they should fix it).

tanus10, 2020-01-09:
````
"--platform" is only supported on a Docker daemon with experimental features enabled
````

They consider the "--platform" option as an experiment. I don't think it would be fixed promptly.
Could you please revive the tag arm32/v6 for now?

Koenkk, 2020-01-12:
great that you managed to get it working, I don't prefer to publish to a separate tag as it would increase the zigbee2mqtt build time for 10 minutes (unless may people have this issue but this seems not to be the case atm). Would you mind making a pr for https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/information/docker.md ?

tanus10, 2020-01-12:
I agree with you. As long as there is a way that works, no need for the extra tag.

Koenkk, 2020-01-13:
I've added it to the docs (https://www.zigbee2mqtt.io/information/docker.html) assuming this can be closed now. Thanks!

closed 2020-01-13 by Koenkk