Koenkk/zigbee2mqtt #2645: Can't use friendly names for ikea traadfri outlets

tbowmo, 2019-12-31:
# Bug Report
## What happened
I can not control ikea outlets using frinedly name

## What did you expect to happen

That I can control my ikea outlets using their friendly name
## How to reproduce it (minimal and precise)

I have given my outlet a friendly name of 'outlet/1' and 'outlet/2' (the ikea bulbs have similair naming conventions with slashes like 'light/diningtable')

if I try to control the outlets by publishing ```{"state":"on"}``` to ```zigbee2mqtt/outlet/1/set``` nothing happens.

If I use the deviceId directly, that is f.ex. ```zigbee2mqtt/0x000d6ffffe9c0d6c/set``` as topic, then the outlet responds correctly, and zigbee2mqtt reports back on the topic ```zigbee2mqtt/outlet/1```

If I do the same with my lights, that is publishing ```{"state": "on"}``` to ```zigbee2mqtt/light/diningtable/set``` then it works as expected.

## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version: 20190608 (I think this is the firmware version, it is noted as revision under coordinator meta data)

pcb1962, 2020-01-01:
Just set the payload to 'on' or 'off' instead of '{"state": "on"}'

$ mosquitto_pub -h xxxxx -p 1883 -u xxxxx -P xxxxx -t zigbee2mqtt/study/outlet/set -m on

z2m log:
Jan  1 20:39:08 manatee npm[426]: zigbee2mqtt:info 2020-01-01 20:39:08: MQTT publish: topic 'zigbee2mqtt/study/outlet', payload '{"state":"ON","linkquality":68,"last_seen":"2020-01-01T20:39:08.248Z"}'

tbowmo, 2020-01-01:
Hmm.. it doesn't cut it for me.. 

```
>mosquitto_pub -h 192.168.1.64 -t zigbee/outlet/2/set -m on
```
I get an error in the log 
```
info  2020-01-01 23:04:40: MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"entity_not_found","message":{"friendly_name":"outlet/2"}}'
error 2020-01-01 23:04:40: Entity 'outlet/2' is unknown
```

If I send to the device ID instead:

```
mosquitto_pub -h 192.168.1.64 -t zigbee/0x086bd7fffe59a34d/set -m off
```

I get the following in the log
```
info  2020-01-01 23:03:22: MQTT publish: topic 'zigbee/outlet/2', payload '{"state":"OFF","linkquality":0,"last_seen":"2020-01-01T23:03:22+01:00"}'
```

And yes, I have changed the root topic to zigbee, instead of zigbee2mqtt

tbowmo, 2020-01-01:
Tried changing the friendly name to ```outlet/outside```, instead of ```outlet/2```

Now it works, so it seems that zigbee2mqtt doesn't like number only parts in it's topic "path"

pcb1962, 2020-01-01:
OK, I can confirm this issue. I renamed an outlet:

'0x000d6ffffedab29c':
  friendly_name: study/outlet/1

Send a set command:
$ mosquitto_pub -h xxx -p 1883 -u xxx -P xxx -t zigbee2mqtt/study/outlet/1/set -m on

In the log I see this:
Jan  1 22:31:16 manatee npm[2215]: zigbee2mqtt:debug 2020-01-01 22:31:16: Received MQTT message on 'zigbee2mqtt/study/outlet/1/set' with data 'on'
Jan  1 22:31:16 manatee npm[2215]: zigbee2mqtt:info 2020-01-01 22:31:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":{"friendly_name":"study/outlet/1"}}'
Jan  1 22:31:16 manatee npm[2215]: zigbee2mqtt:error 2020-01-01 22:31:16: Entity 'study/outlet/1' is unknown

However, I use a digit suffix on many of my devices without any issue, but I don't send set commands to these, they are sensor data source only:

'0x00158d000314a645':
  friendly_name: hall/presence/1
  retain: false
'0x00158d000313cfd4':
  friendly_name: utrm/presence/1
  retain: false
'0x00158d00032c28d4':
  friendly_name: study/presence/2
  retain: false
'0x00158d0002ec7dae':
  friendly_name: study/presence/1
  retain: false

Koenkk, 2020-01-05:
You cannot control devices when the friendly name ends with /NUMBER. However currently this is not clear, zigbee2mqtt should refuse to start when this is set.

tbowmo, 2020-01-05:
@Koenkk is this a limitation of mqtt protocol, or an implementation issue in zigbee2mqtt?

Koenkk, 2020-01-08:
`/NUMBER` is used to select a certain endpoint, therefore it cannot be reused (this results in ambiguity)

Boldie, 2020-03-04:
Bumped into the same issue. This configuration shall throw an error on startup.

Koenkk, 2020-03-07:
@Boldie implemented, zigbee2mqtt will refuse to start when a friendly_name ends with /DIGIT.

closed 2020-03-07 by Koenkk