Koenkk/zigbee2mqtt #2646: Xiaomi/Aqara WXKG11LM not working

artekw, 2019-12-31:
# Bug Report
Xiaomi/Aqara WXKG11LM not working
## What happened
WXKG11LM is not pair with zigbee2mqt 0.17 and 0.18-dev. It is working with 0.15 

Debug log after one click:
```
zigbee2mqtt:debug 2019-12-31 11:02:55: Received Zigbee message from '0x00158d0001e79c59', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2019-12-31 11:02:55: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2019-12-31 11:02:55: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

## Debug Info
zigbee2mqtt version: zigbee2mqtt@1.8.0-dev
CC253X firmware version: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

pcb1962, 2020-01-01:
Have you tried removing it an re-pairing?
I have multiple of these in my z2m network all working perfectly:
`Dec 31 19:27:26 manatee npm[3450]: #033[32mzigbee2mqtt:info #033[39m 2019-12-31 19:27:26: button/spare (0x00158d00023e40d3): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
Dec 31 19:27:26 manatee npm[3450]: #033[32mzigbee2mqtt:info #033[39m 2019-12-31 19:27:26: hall/button (0x00158d00023336ec): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
Dec 31 19:27:26 manatee npm[3450]: #033[32mzigbee2mqtt:info #033[39m 2019-12-31 19:27:26: landing/button (0x00158d0002043e68): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
Dec 31 19:27:26 manatee npm[3450]: #033[32mzigbee2mqtt:info #033[39m 2019-12-31 19:27:26: bedroom/button (0x00158d0002337822): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
Dec 31 19:27:26 manatee npm[3450]: #033[32mzigbee2mqtt:info #033[39m 2019-12-31 19:27:26: study/button/lights (0x00158d0001e885dd): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
Dec 31 19:27:26 manatee npm[3450]: #033[32mzigbee2mqtt:info #033[39m 2019-12-31 19:27:26: ensuite/button (0x00158d000204b26a): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
Dec 31 19:27:26 manatee npm[3450]: #033[32mzigbee2mqtt:info #033[39m 2019-12-31 19:27:26: kitchen/button (0x00158d00020088e4): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
Dec 31 19:27:26 manatee npm[3450]: #033[32mzigbee2mqtt:info #033[39m 2019-12-31 19:27:26: utrm/button/1 (0x00158d0002337f9f): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
`

877dev, 2020-01-01:
Same issue here, my existing paired devices continue to work. But if I try and re-pair a device I get "device removed", then it's impossible to re-pair. Tried with WXKG11LM and DJT11LM vibration sensor..

Here's the log after restarting the container (using Docker with Portainer);
```
zigbee2mqtt:info  2020-01-01 13:30:59: Disconnecting from MQTT server

zigbee2mqtt:info  2020-01-01 13:30:59: zigbee-herdsman stopped

Using '/app/data' as data directory

> zigbee2mqtt@1.8.0 start /app

> node index.js

zigbee2mqtt:info  2020-01-01 13:31:02: Logging to console and directory: '/app/data/log/2020-01-01.13-31-01'

zigbee2mqtt:info  2020-01-01 13:31:02: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)

zigbee2mqtt:info  2020-01-01 13:31:02: Starting zigbee-herdsman...

zigbee2mqtt:info  2020-01-01 13:31:03: zigbee-herdsman started

zigbee2mqtt:info  2020-01-01 13:31:03: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

zigbee2mqtt:info  2020-01-01 13:31:03: Currently 7 devices are joined:

zigbee2mqtt:info  2020-01-01 13:31:03: Xiaomi_1_kitchen_vibration (0x00158d00031b19f8): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)

zigbee2mqtt:info  2020-01-01 13:31:03: Xiaomi_3_weather (0x00158d0003f0ff32): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)

zigbee2mqtt:info  2020-01-01 13:31:03: Xiaomi_5_door_contact (0x00158d0004237b89): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)

zigbee2mqtt:info  2020-01-01 13:31:03: Xiaomi_2_weather (0x00158d0003fb08dc): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)

zigbee2mqtt:info  2020-01-01 13:31:03: Xiaomi_4_door_contact (0x00158d0004240999): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)

zigbee2mqtt:info  2020-01-01 13:31:03: cc2531_router_1 (0x00124b00193e5154): CC2530.ROUTER - Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (Router)

zigbee2mqtt:info  2020-01-01 13:31:03: Ikea_1_dimmer (0xccccccfffe2f9cb8): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)

zigbee2mqtt:warn  2020-01-01 13:31:03: `permit_join` set to  `true` in configuration.yaml.

zigbee2mqtt:warn  2020-01-01 13:31:03: Allowing new devices to join.

zigbee2mqtt:warn  2020-01-01 13:31:03: Set `permit_join` to `false` once you joined all devices.

zigbee2mqtt:info  2020-01-01 13:31:03: Zigbee: allowing new devices to join.

zigbee2mqtt:info  2020-01-01 13:31:03: Connecting to MQTT server at mqtt://localhost

zigbee2mqtt:info  2020-01-01 13:31:04: Connected to MQTT server

zigbee2mqtt:info  2020-01-01 13:31:04: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'

zigbee2mqtt:info  2020-01-01 13:31:04: MQTT publish: topic 'zigbee2mqtt/Xiaomi_1_kitchen_vibration', payload '{"battery":51,"voltage":2915,"linkquality":89,"angle":75,"angle_x":1,"angle_y":0,"angle_z":89,"angle_x_absolute":89,"angle_y_absolute":90}'

zigbee2mqtt:info  2020-01-01 13:31:04: MQTT publish: topic 'zigbee2mqtt/Xiaomi_3_weather', payload '{"battery":68,"voltage":2945,"temperature":19.93,"humidity":61.56,"pressure":1006,"linkquality":73}'

zigbee2mqtt:info  2020-01-01 13:31:04: MQTT publish: topic 'zigbee2mqtt/Xiaomi_5_door_contact', payload '{"battery":100,"voltage":3005,"contact":true,"linkquality":57}'

zigbee2mqtt:info  2020-01-01 13:31:04: MQTT publish: topic 'zigbee2mqtt/Xiaomi_2_weather', payload '{"battery":86,"voltage":2975,"temperature":23.03,"humidity":44.52,"pressure":1006,"linkquality":44}'

zigbee2mqtt:info  2020-01-01 13:31:04: MQTT publish: topic 'zigbee2mqtt/Xiaomi_4_door_contact', payload '{"battery":100,"voltage":3015,"contact":false,"linkquality":68}'

zigbee2mqtt:info  2020-01-01 13:31:04: MQTT publish: topic 'zigbee2mqtt/cc2531_router_1', payload '{"state":true,"led_state":false,"linkquality":84}'

zigbee2mqtt:info  2020-01-01 13:31:04: MQTT publish: topic 'zigbee2mqtt/Ikea_1_dimmer', payload '{"linkquality":68,"battery":100}'

zigbee2mqtt:info  2020-01-01 13:31:04: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.8.0","commit":"da4d26a","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":false}'
```

artekw, 2020-01-02:
> Have you tried removing it an re-pairing?

Yes, I can't pair. It react only on click as on first post.

closed 2020-03-10 by stale[bot]