Koenkk/zigbee2mqtt #2647: npm install: ImportError: No module named ast

ferbar, 2019-12-31:
# Bug Report + fix
Maybe this bug report will help others. See fix below.

## What happened
Following the installation documentation on https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#2-installing the npm install step fails with

```
prebuild-install WARN install No prebuilt binaries found (target=12.14.0 runtime=node arch=arm libc= platform=linux)
Traceback (most recent call last):
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py", line 47, in <module>
    import gyp
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 10, in <module>
    import gyp.input
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 7, in <module>
    import ast
ImportError: No module named ast
```

[snip]

```
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
```

## What did you expect to happen
Successful installation

## How to reproduce it (minimal and precise)
Distribution: DietPi python2.7-minimal
`npm install
`
## How to fix it
On DietPi only python2.7-minimal is installed. This package doesn't include the ast library.
`apt-get install python2.7
`

closed 2019-12-31 by ferbar

freynder, 2020-08-11:
Thank you, I experienced the same issue with armbian buster mainline kernel 5.4.y

bremme, 2020-11-17:
Thanks, I had the same issue on Armbian Buster mainline based kernel 5.8.y. This was the first Google hit:).