Koenkk/zigbee2mqtt #2648: Trying to support Hive SLR1b (Boiler Receiver)

markuspf, 2019-12-31:
Hi,

I am trying to support the Hive Boiler Receiver (SLR1b). I followed the how to support new devices. The device pairs, but doesn't do much yet (unfortunately). So here's what I put into devices.js (Its probably obvious that I don't quite know what I am doing here).

```
   {                                                                                             
        zigbeeModel: ['SLR1b'],                                                                   
        model: 'SLR1b',                                                                           
        vendor: 'Hive',                                                                           
        description: 'Boiler control',                                                            
        supports: 'thermostat',                                                                   
        fromZigbee: [fz.generic_power, fz.thermostat_att_report],                                 
        toZigbee: [                                                                               
            tz.thermostat_occupied_heating_setpoint, tz.thermostat_local_temperature_calibration, 
            tz.thermostat_setpoint_raise_lower, tz.thermostat_remote_sensing,                     
            tz.thermostat_system_mode, tz.thermostat_running_state,                               
        ],                                                                                        
        meta: {configureKey: 2},                                                                  
        configure: async (device, coordinatorEndpoint) => {                                       
            const endpoint5 = device.getEndpoint(5);                                              
            const endpoint6 = device.getEndpoint(6);                                              
            const endpoint7 = device.getEndpoint(7);                                              
            const endpoint8 = device.getEndpoint(8);                                              
                                                                                                  
            await bind(endpoint5, coordinatorEndpoint, [                                          
                'genBasic', 'genPowerCfg', 'genIdentify', 'genTime', 'genPollCtrl', 'hvacThermosta            ]);                                                                                   
            await bind(endpoint6, coordinatorEndpoint, [                                          
                'hvacThermostat'                                                                              ]);                                                                                               await bind(endpoint7, coordinatorEndpoint, [                                                          'msTemperatureMeasurement'                                                                    ]);                                                                                               await bind(endpoint8, coordinatorEndpoint, [                                                          'msTemperatureMeasurement'                                                                    ]);                                                                                                                                                                                                 await configureReporting.thermostatTemperature(endpoint5);                                        await configureReporting.thermostatPIHeatingDemand(endpoint5);                                    await configureReporting.thermostatTemperature(endpoint6);                                        await configureReporting.thermostatPIHeatingDemand(endpoint6);                                    await configureReporting.temperature(endpoint7);                                      
            await configureReporting.temperature(endpoint8);                                      
        },                                                                                        
        onEvent: async (type, data, device) => {                                                  
        },                                                                                        
    },                                    
```

The relevant line in database.db is

```
{"id":21,"type":"Router","ieeeAddr":"0x001e5e09024ab4f3","nwkAddr":45339,"manufId":4153,"manufName
":"Computime","powerSource":"Mains (single phase)","modelId":"SLR1b","epList":[5,6,7,8],"endpoints
":{"5":{"profId":260,"epId":5,"devId":769,"inClusterList":[0,3,9,10,513,64768],"outClusterList":[0
,10,25,1026,64768],"clusters":{"genBasic":{"attributes":{"modelId":"SLR1b","manufacturerName":"Com
putime","powerSource":1,"zclVersion":1,"appVersion":33,"stackVersion":2,"hwVersion":1,"dateCode":"
20180504"}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b001938cb23","end
pointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001938cb23","endpointID":1
},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b001938cb23","endpointID":1},{"cluste
r":10,"type":"endpoint","deviceIeeeAddress":"0x00124b001938cb23","endpointID":1},{"cluster":32,"ty
pe":"endpoint","deviceIeeeAddress":"0x00124b001938cb23","endpointID":1},{"cluster":513,"type":"end
point","deviceIeeeAddress":"0x00124b001938cb23","endpointID":1}]},"6":{"profId":260,"epId":6,"devI
d":769,"inClusterList":[513],"outClusterList":[],"clusters":{},"binds":[{"cluster":513,"type":"end
point","deviceIeeeAddress":"0x00124b001938cb23","endpointID":1}]},"7":{"profId":260,"epId":7,"devI
d":769,"inClusterList":[1026],"outClusterList":[],"clusters":{},"binds":[{"cluster":1026,"type":"e
ndpoint","deviceIeeeAddress":"0x00124b001938cb23","endpointID":1}]},"8":{"profId":260,"epId":8,"de
vId":769,"inClusterList":[1026],"outClusterList":[],"clusters":{},"binds":[{"cluster":1026,"type":
"endpoint","deviceIeeeAddress":"0x00124b001938cb23","endpointID":1}]}},"appVersion":33,"stackVersi
on":2,"hwVersion":1,"dateCode":"20180504","zclVersion":1,"interviewCompleted":true,"meta":{"config
ured":2},"lastSeen":1577801968500}
```

The first thing that happens is that the device tries to 'read' 'timeStatus' from the 'genTime' cluster, how does one appropriately reply to this request? I see the following in the logs

```
2019-12-31.14-19-49/log.txt:debug 2019-12-31 14:33:37: No converter available for 'SLR1b' with cluster 'genTime' and type 'read' and data '["timeStatus"]'
```

I tried writing a converter, but it does not seem to be doing the correct thing (or, for that matter, anything).

I do realize that this device might need proprietary commands, so I might need to get a hive hub and sniff traffic :/.

dmpolukhin, 2020-01-08:
I'm also trying to connect the same device to zigbee2mqtt but in my case actuator connected without name (i.e. name is 'undefined' in log messages) but remote thermostat identified itself as 'WPT1'. Any suggestion how to setup device without modelID?

@markuspf it seems that your config from devices.js has absolutely broken formatting...

Relevant lines from database.db:
`{"id":4,"type":"Router","ieeeAddr":"0x001e5e0902478585","nwkAddr":54035,"manufId":4153,"epList":[6,7,8,5],"endpoints":{"5":{"profId":260,"epId":5,"devId":769,"inClusterList":[0,3,9,10,513,64768],"outClusterList":[0,10,25,1026,64768],"clusters":{},"binds":[]},"6":{"profId":260,"epId":6,"devId":769,"inClusterList":[513],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"7":{"profId":260,"epId":7,"devId":769,"inClusterList":[1026],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":769,"inClusterList":[1026],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":true,"meta":{}}

{"id":5,"type":"EndDevice","ieeeAddr":"0x001e5e090240ba95","nwkAddr":38680,"manufId":4153,"manufName":"Computime","powerSource":"Battery","modelId":"WPT1","epList":[9],"endpoints":{"9":{"profId":260,"epId":9,"devId":770,"inClusterList":[0,1,3,9,10,513,516,1026,32,64768],"outClusterList":[3,10,513,25,64768],"clusters":{"genBasic":{"attributes":{"modelId":"WPT1","manufacturerName":"Computime","powerSource":3,"zclVersion":1,"appVersion":54,"stackVersion":2,"hwVersion":0,"dateCode":"20170914"}}},"binds":[]}},"appVersion":54,"stackVersion":2,"hwVersion":0,"dateCode":"20170914","zclVersion":1,"interviewCompleted":true,"meta":{}}`

markuspf, 2020-01-09:
The actuator takes a few tries before it comes across with an identification for me; did you try adding the above code to the `devices.js` file?

I can get status updates from the actuator (it tells me whether the relais is on or off), but I can only switch it using the physical button as of now. I'll receive a hub and a thermostat soon and hope some sniffing will help things along a bit...

dmpolukhin, 2020-01-09:
@marcusaram "as as" your code fragment cannot work because it has string literal that is not properly terminated see:
`'hvacThermosta            ]);` 
also it has duplicates (you can see it if you scroll you code snippet to the right). I tried to fix all of this but my actuator is not recognized because modelId is missing in my database so cannot be matched with your code expects model 'SLR1b'.

markuspf, 2020-01-09:
Do you have the same actuator as I do, i.e. the SLR1b? There are other models such as the 2-channel one for hot water and heating at the very least.

For me the acuator tried pairing once or twice with the model being `undefined` and then paired giving the `SLR1b`. How do you pair it? Do you hold down the button until it goes into stand-alone mode (white LED indication), and then hold down the button again for it to go into pairing mode (amber double flash)?

This is the devices.js entry I am currently using:

```
   {
        zigbeeModel: ['SLR1b'],
        model: 'SLR1b',
        vendor: 'Hive',
        description: 'Boiler control',
        supports: 'thermostat',
        fromZigbee: [fz.generic_power, fz.thermostat_att_report, fz.time_status],
        toZigbee: [
            tz.thermostat_occupied_heating_setpoint, tz.thermostat_local_temperature_calibration,
            tz.thermostat_setpoint_raise_lower, tz.thermostat_remote_sensing,
            tz.thermostat_system_mode, tz.thermostat_running_state,
        ], 
        meta: {configureKey: 4},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint5 = device.getEndpoint(5);
            const endpoint6 = device.getEndpoint(6);
            const endpoint7 = device.getEndpoint(7);
            const endpoint8 = device.getEndpoint(8);

            await bind(endpoint5, coordinatorEndpoint, [
                'genBasic', 'genPowerCfg', 'genIdentify', 'genTime', 'genPollCtrl', 'hvacThermostat',
            ]); 
            await bind(endpoint6, coordinatorEndpoint, [
                'hvacThermostat'
            ]);
            await bind(endpoint7, coordinatorEndpoint, [
                'msTemperatureMeasurement'
            ]);
            await bind(endpoint8, coordinatorEndpoint, [
                'msTemperatureMeasurement'
            ]);

            await configureReporting.thermostatTemperature(endpoint5);
            await configureReporting.thermostatRunningState(endpoint5);
        },
    },
```

the actuator pairs alright, but then expects something to happen, of which I don't know yet what it is. It isn't getting `timeStatus` (alone), because I also paired this device to a deCONZ platform which supports `timeStatus` and `time`, and it's still just amber flashing around. I suspect that it wants to hook up with the thermostat, hence I'll wait for that to arrive before I can try anything else.

dmpolukhin, 2020-01-09:
I do have SLR1b. Perhaps I didn't wait long enough to receive name 'SLR1b'. I pressed button again to get double amber flashes and paired actuator. But as soon as I paired I turned on remote thermostat it also paired as WPT1 but since then I haven't received any message from it. In my case remote thermostat can control actuator and I keep receiving messages from actuator:
`zigbee2mqtt:debug 2020-01-09 23:07:53: Received Zigbee message from '0x001e5e0902478585', type 'read', cluster 'genTime', data '["timeStatus"]' from endpoint 5 with groupID 0
zigbee2mqtt:warn  2020-01-09 23:07:53: No converter available for 'SLR1b' with cluster 'genTime' and type 'read' and data '["timeStatus"]'`

To get this messages I had to assign modelID to the actuator directly in  database.db.

I'm waiting for another CC2531 to use it as sniffer and be able to see all messages.

markuspf, 2020-01-10:
regarding the `timeStatus` I have opened another issue on `zigbee-herdsman`(https://github.com/Koenkk/zigbee-herdsman/issues/122), haven't managed to properly reply to the `timeStatus` messages yet, though :/

markuspf, 2020-01-10:
It is good to know that having a thermostat makes you able to control the actuator, so there's hope I'll get this thing to work properly.

dmpolukhin, 2020-01-11:
@markuspf due to errors in log about genTime every 10 seconds I didn't notice initially but I receive status updates from the thermostat and it can be controlled by posting MQTT commands. Not everything from your toZigbee list is supported but basic things like system_mode and occupied_heating_setpoint work fine.

dmpolukhin, 2020-01-11:
zigbee-herdsman actually responses to time request but doesn't include all required fields in the response. I replied with "timeStatus: 3" (timeStatus is mandatory field according to the specification), device requested localTime and after that stopped asking for time every 10 seconds. I think response should looks something like this:
`yield endpoint.readResponse(frame.Cluster.ID, frame.Header.transactionSequenceNumber, {                                                                                                               
  time: time,                                                                                                                                                                                       
  timeStatus: 3,                                                                                                                                                                                    
  localTime: time,                                                                                                                                                                              });`

markuspf, 2020-01-17:
@dmpolukhin just to be clear: You use the SLR1b together with a thermostat unit?

Once I got mine delivered and paired, the basics work for me too. I would of course rather be able to just switch the relay on the SLR1b between on and off and do the heating demand control in my own software. It's a bit pointless having the dead-weight thermostat lying around :/ 

Sure, this can be hacked by just setting the occupied heating setpoint to 30 degrees whenever there's demand...

dmpolukhin, 2020-01-18:
@markuspf as far as I understand real thermostat is SLR1b (i.e. it supports hvac interface instead of simple relay of/off interface). Remote control is just a temperature sensor and UI for the thermostat. Thermostat cannot work without remote temperature sensor and simple on/off interface is not supported. But via Zigbee you can fully control it as a thermostat i.e. set target temperature, change weekly schedule, switch mode (manual or weekly schedule) etc. Remote module doesn't send on/off commands instead it sends current temperature and all on/off logic is inside the thermostat (I checked it with sniffer). Perhaps you can emulate temperature sensor with Z2M but still on/off logic will be in SLR1b.

My changes are in https://github.com/Koenkk/zigbee-herdsman-converters/pull/923 and https://github.com/Koenkk/zigbee-herdsman/pull/128

markuspf, 2020-01-30:
Thanks for your work! I'll try it out. 

Who would've guessed that the assumption is that every house has just one room that one needs to schedule heating demands for, eh?

closed 2020-04-07 by stale[bot]

robertalexa, 2021-08-02:
Hey @dmpolukhin,

I was reading this thread and noticed that you own a WPT1 remote control? I am currently trying to add battery support for all the Hive remotes but i would need some help with people testing before i call it done.

If you still have it, and you are willing to offer me 5 minutes of your time, i can give you some steps of what to do, and if the battery percent starts showing up then I will submit a PR. Thank you

dmpolukhin, 2021-08-03:
Hi @robertalexa , unfortunately I don't have access to the device anymore (it was in the appointment that I rented).