Koenkk/zigbee2mqtt #2649: Eurotronic Spirit disconnecting and impossible to re-pair

gaggio, 2019-12-31:
# Bug Report
An Eurotronic spirit has been correctly working so far, but then it stopped being controllable (couldn't switch off or change setpoint from home assistant). Note that the coordinator continued receiving updates because the setpoint manually modified from the device got updated into home assistant.
The device was marked as unreachable, in the zigbee network map it was not connected to any other node.
I decided to re-pair the device, so I reset and despite several attempts the pairing always fails.
Tried:
- keeping the device 10cm from the coordinator, or from other routers
- deleting the device's line in database
- resetting multiple times

## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version: 20190608

Thanks

gaggio, 2020-01-01:
Made some progress by removing the stick from the host, rebooting and connecting the stick back to the usb port. The device has been paired, but still not working. In Zigbee2MqttAssistant I see it in the network map, but it is marked as "no components".

TilmanK, 2020-01-05:
I had that problem yesterday. Stopping Zigbee2Mqtt, reconnecting the stick physically and then starting Zigbee2Mqtt resolved the problem. Instantly all devices in pairing mode connected.

gaggio, 2020-01-06:
I confirm this beahviour, tried with another device as well. But still the Eurotronic valve is not correctly working although correctly re-paired. While reported as connected with good link quality, it is reported as "not supported" in the log, and "no components" in zigbee2mqttAssistant.

Koenkk, 2020-01-08:
@gaggio can you share your `database.db` and log?

gaggio, 2020-01-10:
> @gaggio can you share your `database.db` and log?

Sure. Today the device started automagically to work again after several days, without any intervention from my side.
If you're still interested to investigate, see below:

`database.db`: https://pastebin.com/DuYDCvV0
Log for the first 2 hours after starting Zigbee2Mqtt: https://pastebin.com/jXzktHwk
Log at the time the device magically started to work again: https://pastebin.com/tMnaeVZe

Interested device id is 0x00158d000192dce7

Koenkk, 2020-01-13:
I think initially the interview failed:
```
info  2020-01-06 22:44:10: 0x00158d000192dce7 (0x00158d000192dce7): Not supported (EndDevice)
```

But in the second log it interviews correctly, not sure why but for that you need to remove and re-pair the device again (to check if the issue is reproducible) https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

nerdosity, 2020-01-14:
Today I got 4 Spirit ZigBee valves.
I managed to pair 2 of them, just randomly stopping service, rebooting, changing port on Synology nas, resetting the valve, trying again.

Meanwhile, I paired *instantly* 2 IKEA repeaters. Just pressing the reset button and devices were paired.

I tried to pair the valves near the ZigBee cc2531 stick, near the repeater, before of everything I tried to pair them directly from the radiator (silly me).

I will attach the log in a while. I need to set it more verbose than usual?

nerdosity, 2020-01-14:
`

2020-01-14 19:26:50 | stdout | > node index.js
-- | -- | --
2020-01-14 19:26:50 | stdout | > zigbee2mqtt@1.8.0-dev start /app
2020-01-14 19:26:50 | stdout |  
2020-01-14 19:26:50 | stdout | Using '/app/data' as data directory
2020-01-14 19:26:48 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:26:48: zigbee-herdsman stopped
2020-01-14 19:26:48 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:26:48: Disconnecting from MQTT server
2020-01-14 19:26:48 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:26:48: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
2020-01-14 19:26:22 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:26:22: MQTT publish: topic 'zigbee2mqtt/0x00158d0003c475a9', payload '{"local_temperature":19.5,"occupied_heating_setpoint":19,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"linkquality":76,"current_heating_setpoint":19,"eurotronic_system_mode":5,"system_mode":"heat","eurotronic_error_status":0,"eurotronic_host_flags":{"mirror_display":false,"boost":true,"window_open":false,"child_protection":false}}'
2020-01-14 19:26:22 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:26:22: MQTT publish: topic 'zigbee2mqtt/0x00158d0003c475a9', payload '{"local_temperature":19.5,"occupied_heating_setpoint":19,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"linkquality":76,"current_heating_setpoint":19,"eurotronic_system_mode":5,"system_mode":"heat","eurotronic_error_status":0,"eurotronic_host_flags":{"mirror_display":false,"boost":true,"window_open":false,"child_protection":false}}'
2020-01-14 19:26:22 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:26:22: MQTT publish: topic 'zigbee2mqtt/0x00158d0003c475a9', payload '{"local_temperature":19.5,"occupied_heating_setpoint":19,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"linkquality":76,"current_heating_setpoint":19,"eurotronic_system_mode":5,"system_mode":"heat","eurotronic_error_status":0,"eurotronic_host_flags":{"mirror_display":false,"boost":true,"window_open":false,"child_protection":false}}'
2020-01-14 19:26:16 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:26:16: MQTT publish: topic 'zigbee2mqtt/0x00158d0003c475a9', payload '{"local_temperature":19.5,"occupied_heating_setpoint":19,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"linkquality":78,"current_heating_setpoint":19,"eurotronic_system_mode":5,"system_mode":"heat","eurotronic_error_status":0,"eurotronic_host_flags":{"mirror_display":false,"boost":true,"window_open":false,"child_protection":false}}'
2020-01-14 19:25:59 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:25:59: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ffd6ba', payload '{"occupied_heating_setpoint":19,"unoccupied_heating_setpoint":16,"pi_heating_demand":40,"linkquality":36,"local_temperature":17,"current_heating_setpoint":19,"eurotronic_system_mode":1,"system_mode":"auto","eurotronic_error_status":0,"battery":70,"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":false}}'
2020-01-14 19:21:14 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:21:14: MQTT publish: topic 'zigbee2mqtt/0x00158d0003c475a9', payload '{"local_temperature":19.5,"occupied_heating_setpoint":19,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"linkquality":86,"current_heating_setpoint":19,"eurotronic_system_mode":5,"system_mode":"heat","eurotronic_error_status":0,"eurotronic_host_flags":{"mirror_display":false,"boost":true,"window_open":false,"child_protection":false}}'
2020-01-14 19:20:38 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:20:38: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.8.0-dev","commit":"5c18734","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
2020-01-14 19:20:38 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:20:38: Zigbee: allowing new devices to join.
2020-01-14 19:20:14 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:20:14: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.8.0-dev","commit":"5c18734","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":false}'
2020-01-14 19:20:14 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:20:14: Zigbee: disabling joining new devices.
2020-01-14 19:19:31 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:19:31: MQTT publish: topic 'zigbee2mqtt/0x00158d0003c475a9', payload '{"local_temperature":19.5,"occupied_heating_setpoint":19,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"linkquality":86,"current_heating_setpoint":19,"eurotronic_system_mode":5,"system_mode":"heat","eurotronic_error_status":0,"eurotronic_host_flags":{"mirror_display":false,"boost":true,"window_open":false,"child_protection":false}}'
2020-01-14 19:19:30 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:19:30: MQTT publish: topic 'zigbee2mqtt/0x00158d0003c475a9', payload '{"local_temperature":19.5,"occupied_heating_setpoint":19,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"linkquality":84,"current_heating_setpoint":19,"eurotronic_system_mode":1,"system_mode":"auto","eurotronic_error_status":0,"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":false}}'
[... cut ...]
2020-01-14 19:15:21 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:15:21: Starting zigbee2mqtt version 1.8.0-dev (commit #5c18734)
2020-01-14 19:15:21 | stdout | [32mzigbee2mqtt:info [39m 2020-01-14 19:15:21: Logging to console and directory: '/app/data/log/2020-01-14.19-15-21'
2020-01-14 19:15:20 | stdout |  
2020-01-14 19:15:20 | stdout | > node index.js
2020-01-14 19:15:20 | stdout | > zigbee2mqtt@1.8.0-dev start /app
2020-01-14 19:15:20 | stdout |  
2020-01-14 19:15:19 | stdout | Using '/app/data' as data directory

`

nerdosity, 2020-01-14:
Batteries. Never install your new valves with half-used batteries.
After changing batteries even the other two valves paired.

Koenkk, 2020-01-15:
Great! I assume this can be closed now.

closed 2020-01-15 by Koenkk

nerdosity, 2020-01-15:
IDK if it's the same issue for everyone here, it was surely for me since I was installing new Spirit valves with old batteries I had on my old valves.
Maybe we should wait for an update from someone else.