Koenkk/zigbee2mqtt #2650: Smart ZigBee 3.0 inline Light Dimmer

ayushjain4484, 2020-01-01:
Happy Newyear All,

I just brought Smart Zigbee dimmer switch from ali express

[Aliexpress link:](https://www.aliexpress.com/item/4000270950615.html?spm=a2g0o.productlist.0.0.1a80437dLrI0hs&algo_pvid=3b0b8f8c-250c-423a-ad61-f0ce05a6523c&algo_expid=3b0b8f8c-250c-423a-ad61-f0ce05a6523c-25&btsid=2fe2d552-4d1c-418a-a4c6-aabc95b7529c&ws_ab_test=searchweb0_0,searchweb201602_6,searchweb201603_55)

This device got paired very easily and working fine as ON/OFF Switch but i wanted to use it as a dimmer switch since main purpose to but this was to use it as a Dimmer.

I can see below response from device

Jan 01 14:01:13 raspberrypi npm[29382]: zigbee2mqtt:info 2020-01-01 14:01:13: MQTT publish: topic ‘zigbee2mqtt/Dimmer’, payload ‘{“state”:“ON”,“linkquality”:115}’

Since it as State as topic it ccan be used as ON/OFF device,
Is there any option to add brightness topic aswell since it is supported by hardware.

PLease assist.

NewFolk, 2020-01-05:
Hi every new for zigbee2mqtt device should be configured first.
Try to follow this manual:
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

ayushjain4484, 2020-01-05:
Appologies, 
Acutally issue here is device is working but not not working fine since it is a dimmer switch it should have on/off and brightnee

but currently it is only having state on/off
Jan 05 19:02:21 raspberrypi npm[8046]: zigbee2mqtt:info  2020-01-05 19:02:21: MQTT publish: topic 'zigbee2mqtt/Dimmer', payload '{"state":"ON","linkquality":76}'

Somehow this device got paired by its own.

When i pass set topic On and Off it works fine i am not user under which device this got paired.

ayushjain4484, 2020-01-05:
Or please let me know if there is any way i can check under what device type it got paired to that i can just make changes accordingly under devices.js

since this device is working fine on passing on/off there has to be some device type under which it got paired.

closed 2020-03-12 by stale[bot]