Koenkk/zigbee2mqtt #2652: Zen Thermostat

peter1960, 2020-01-01:
Using latest firmware & zigbee2mqtt

I have a ZEN thermostat and when paired it shows error `Cannot read property 'bind' of undefined` :

`zigbee2mqtt:debug 2020-01-01 12:41:12: Device 'Heater' announced itself
zigbee2mqtt:debug 2020-01-01 12:41:13: Received Zigbee message from 'Heater', type 'readResponse', cluster 'genBasic', data '{"modelId":"Zen-01","manufacturerName":"Zen Within","powerSource":129}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-01-01 12:41:13: No converter available for 'Zen-01-W' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"Zen-01","manufacturerName":"Zen Within","powerSource":129}'
zigbee2mqtt:debug 2020-01-01 12:41:13: Received Zigbee message from 'Heater', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"appVersion":23,"stackVersion":2}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-01-01 12:41:13: No converter available for 'Zen-01-W' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1,"appVersion":23,"stackVersion":2}'
zigbee2mqtt:debug 2020-01-01 12:41:13: Received Zigbee message from 'Heater', type 'readResponse', cluster 'genBasic', data '{"hwVersion":2}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-01-01 12:41:13: No converter available for 'Zen-01-W' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":2}'
zigbee2mqtt:info  2020-01-01 12:41:13: Successfully interviewed 'Heater', device has successfully been paired
zigbee2mqtt:info  2020-01-01 12:41:13: Device 'Heater' is supported, identified as: Zen Thermostat (Zen-01-W)
zigbee2mqtt:info  2020-01-01 12:41:13: MQTT publish: topic 'zigbeeTest/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"Heater","model":"Zen-01-W","vendor":"Zen","description":"Thermostat","supported":true}}'
zigbee2mqtt:error 2020-01-01 12:41:13: Failed to configure 'Heater', attempt 2 (TypeError: Cannot read property 'bind' of undefined
    at bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:17:24)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:5704:19)
    at DeviceConfigure.configure (/opt/zigbee2mqtt/lib/extension/deviceConfigure.js:96:32)
    at DeviceConfigure.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/deviceConfigure.js:77:18)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:322:44))
`

I can see in the devices.js it has
`    {
        zigbeeModel: ['Zen-01'],
        model: 'Zen-01-W',
        vendor: 'Zen',
        description: 'Thermostat',
`
If I then change to
`    {
        zigbeeModel: ['Zen-01-W'],
        model: 'Zen-01-W',
        vendor: 'Zen',
        description: 'Thermostat',
`
it will then not have the error, is the zigbeeModel not correct for mine  and  a new entry in the devices.js is needed?

peter1960, 2020-01-02:
I found the issue
`const endpoint = device.getEndpoint(3);`
Changed it to 1 and it works

rick-cross, 2020-02-08:
> I found the issue
> `const endpoint = device.getEndpoint(3);`
> Changed it to 1 and it works

Hi @peter1960 ,
Would you mind explaining where you changed that config/parameter?
I have the same thermostat and same exact issue as you.
I'm running Hass.io on docker on Synology.
Greatly appreciated.

peter1960, 2020-02-08:
It is in the devices.js, look at the docs for adding new devices on how to
find the file.
I think a update is needed as I cannot see why it's set to 3, maybe a typo.

rick-cross, 2020-02-08:
Thanks @peter1960 for the info. 

@Koenkk is the issue described by Peter a mistake/typo? If it is a mistake, would it be possible to have an update?  

I ask, selfishly, because it looks like the way I have my setup prevents me from fixing the issue (edit devices.js) by myself. I'm running hass.io as a docker via a synology native package from [here](https://community.home-assistant.io/t/hass-io-on-synology-dsm-native-package/125559) and essentially I'm unable to access the internal docker directory to modify the file. 

I was following the documentation to add [new devices via hass.io SSH addon](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#hassio-addon) and [hassio debugging page](https://developers.home-assistant.io/docs/en/hassio_debugging.html) to access the devices.js. And it looks like the hass.io SSH add-on doesn't work on my setup according to the developer of the add-on in this issue [which kills/doesn't start the SSH server](https://github.com/hassio-addons/addon-ssh/issues/110). Its the same issue [I'm encountering (has a bit more detail)](https://github.com/hassio-addons/addon-ssh/issues/119)

So it looks like I am not able to get access to the internal docker files to make the correction as Peter did. Hence the selfish request :(

rick-cross, 2020-02-11:
FFFFF, finally figured it out, I'm posting my steps here in case someone else encounters the same issue as me.
I used docker cp command to copy the file to my local synology drive:

`docker cp <container-name>:</app/node_modules/zigbee-herdsman-converters/devices.js> <path/on/host/>` 

Then modified the devices.js file as Peter described above with vi (using the native synology text editor wouldn't keep the changes persistently).
Then copied the file back to the directory within synology:

`docker cp <path/on/host/devices.js> <container-name>:</app/node_modules/zigbee-herdsman-converters> ` 

And then restart the zigbee2mqtt. That did the trick. I'll have to wait and see if an update to the z2mqtt docker overwrites my changes back to (3) or keeps my mods.
In the end it was pretty simple, it just took forever to figure out because it was tough (for me) to figure out how to modify the internal files inside docker... Found out about the cp command from here:
[about half way down from Julius Printz](https://stackoverflow.com/questions/20813486/exploring-docker-containers-file-system)

Also just as a helpful (hopefully) information/feedback to the developers, I tried to follow the steps described [here](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#docker) to have a local copy of the zigbee herdsman converters folder 

```
cd /opt
git clone https://github.com/Koenkk/zigbee-herdsman-converters.git
# If you also want to add Home Assistant integration for this devices
wget https://raw.githubusercontent.com/Koenkk/zigbee2mqtt/master/lib/extension/homeassistant.js
```

and then link that volume/directory to the docker container, via 

`-v /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters -v /opt/homeassistant.js:/app/lib/extension/homeassistant.js`

but the docker wouldn't start and kept giving me an error like "index.js module_not_found".
That's why I had to end up doing it with the cp docker command.

peter1960, 2020-02-11:
Will be interesting to see how it behaves for you.
After a few days testing I have noticed that the values returned in the json may not be the real values unless you have ask for them.
If you ask for say the state with zigbee2mqtt/Thermostat/get/system_mode it returns all the values but but only system_mode will be correct, other values are what you had when last asked so occupied_heating_setpoint will not be what the device is actually set to.

Often gets fail with:

{"type":"zigbee_publish_error","message":"Publish 'get' 'system_mode' to 'Thermostat' failed: 'Error: Read 0x00244600000e5fef/1 hvacThermostat([\"systemMode\"], {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":true}) failed (Error: Timeout - 13480 - 1 - 85 - 513 - 1 after 10000ms)'","meta":{"friendly_name":"Thermostat"}}

rick-cross, 2020-02-11:
Yeah that was the next thing for me to look at. Haven't had much time to investigate since I "connected" it. But immediately after it connected, I noticed that home assistant widget was/is reporting the value of about 20 degrees F inside my house :). The thermostat is set to and reads about 70 F.
I also tried to move the "scroll bar" on the widget up and down a bit and noticed that the thermostat changed its "set to" value to something like 45 F, so the heat turned off obviously. Again, these are just the few initial observations, I haven't had much time to look into it since then.

sjorge, 2020-02-12:
70 F is about 21 C... so it might be reporting in celcius

peter1960, 2020-02-13:
I created https://github.com/Koenkk/zigbee-herdsman-converters/issues/984 about the endpoint

rick-cross, 2020-02-16:
> 70 F is about 21 C... so it might be reporting in celcius

yes i think so too and it looks like thats what is happening.
The one time the thermostat responded to inputs from HA, I set the temps to min and max and the thermostat got set to the 44 which equals 7 C and 86 which equals 30 C which are the min and max values in the config as described [here](https://www.zigbee2mqtt.io/devices/Zen-01-W.html).

@peter1960 
Thanks for creating the issue/ticket in the correct place (zigbee-herdsman-converters) unlike me (above). 
BTW I saw that someone else had the [same issue](https://github.com/Koenkk/zigbee-herdsman-converters/issues/524) that you also commented on and got a bit further but still wasn't successful and eventually gave up. Unfortunately I'm not able to follow very well what he was changing/modifying  to try to understand what the issues are or could be. I'm pretty new to this. Hopefully a future update can resolve the issue since the thermostat seems to work with the smarthings hub as the guy indicated.
Its strange that for me the thermostat seems to be communicating the current temperature to HA (in 'sleep' mode), but it doesn't respond to inputs from HA in sleep mode. Only sometimes when its awake.

closed 2020-04-25 by stale[bot]

crixyd, 2020-07-09:
I too have the ZEN thermostat, and am experiencing several issues inline with those detailed above, but specifically
- Thermostat doesn't report temperature back to HA
- Thermostat doesn't report mode back to HA if change initiated on thermostat
- Thermostat accepts mode changes, however only one in every ~5 attempts when the device is asleep, or every time when it is awake

On review of Z2M logs I see the following (note I haven't modified zigbee-herdsman-converters/devices.js) 

2020-07-09T10:22:16.407Z zigbee-herdsman:deconz:adapter BIND FAILED - addr: 0x42e1 openRequest TIMEOUT
2020-07-09T10:22:16.408Z zigbee-herdsman:controller:endpoint Bind 0x00244600000ee462/1 genBasic from '0x00212effff05c0df/1' failed (undefined)
zigbee2mqtt:error 2020-07-09 20:22:16: Failed to configure 'Universal - ZEN Thermostat', attempt 1 (Error: Bind 0x00244600000ee462/1 genBasic from '0x00212effff05c0df/1' failed (undefined)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:255:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
winston:file: written true false
winston:file: logged 65691 error 2020-07-09 20:22:16: Failed to configure 'Universal - ZEN Thermostat', attempt 1 (Error: Bind 0x00244600000ee462/1 genBasic from '0x00212effff05c0df/1' failed (undefined)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:255:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

For context I am using the experimental DeCONZ adapter Raspbee II, and Z2M is working with all other supported devices in my household. When using the DeCONZ integration rather than Z2M (which also has partial ZEN support) the temperature is reported, however device mode cannot be changed. So pretty much exactly the opposite issue :p

peter1960, 2020-07-09:
I have mine working ok now, as for your observations:
Even though all values are returned I found the device only reports back
updated value what you asked for
Yes you need to continually try talk and eventually get through.

   - Thermostat doesn't report temperature back to HA
   - Thermostat doesn't report mode back to HA if change initiated on
   thermostat
   - Thermostat accepts mode changes, however only one in every ~5 attempts
   when the device is asleep, or every time when it is awake

I actually use node red to control it with
https://flows.nodered.org/flow/6e4649bc6d6529078cbb731610242eac and then
the HA thermostat is correct  and used as read only an below is the graph
from HA

[image: image.png]

crixyd, 2020-07-10:
Ahh ok, I hadn't fully appreciated the need to make specific requests to
get the granular data points, i'll give that a shot! In any case it seems
that this wouldn't yet be considered fully supported? It works well via
smartthings in my experience, so the device is capable of a more typical 2
way integration.

peter1960, 2020-07-10:
Works perfect via Node Red and that controller and just ignore all the
retries/failure etc

crixyd, 2020-07-10:
Nice :) Until your message I hadn't even thought of using node red at the
device level... so far I'm only using it for HA automation. Great plan!

stegno, 2020-12-27:
Also have problem setting mode and temp for ZEN. Phoscon integration worked better, but not perfect also.

rick-cross, 2021-01-08:
@stegno 
My Zen thermostat now works fine for me with the new-ish ZHA integration thats part of home assistant. Haven't needed MQTT.  
I'm able to control the mode, the temp and on/off also. Even have some simple time based automations.