Koenkk/zigbee2mqtt #2654: z2m not starting

nickrout, 2020-01-02:
# Bug Report
## What happened

New install fails to start

## What did you expect to happen

z2m starting

## How to reproduce it (minimal and precise)

npm start in /opt/zigbee2mqtt

## Debug Info
zigbee2mqtt version:  1.8.0 (commit #da4d26a)
CC253X firmware version: CC2530ZNP-Prod.hex from CC2530_CC2591_DEFAULT_20190608.zip

log: 

/opt/zigbee2mqtt$ npm start

> zigbee2mqtt@1.8.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-01-02 14:54:05: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-01-02.14-54-05'
zigbee2mqtt:info  2020-01-02 14:54:05: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
zigbee2mqtt:info  2020-01-02 14:54:05: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-01-02 14:54:13: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-01-02 14:54:13: Failed to start zigbee
zigbee2mqtt:error 2020-01-02 14:54:13: Exiting...
zigbee2mqtt:error 2020-01-02 14:54:13: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.8.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.8.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/nick/.npm/_logs/2020-01-02T01_54_43_065Z-debug.log

nickrout, 2020-01-02:
full debug log
```
/opt/zigbee2mqtt$ DEBUG=* npm start

> zigbee2mqtt@1.8.0 start /opt/zigbee2mqtt
> node index.js

winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
zigbee2mqtt:info  2020-01-02 15:14:05: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-01-02.15-14-05'
winston:file written true false
zigbee2mqtt:debug 2020-01-02 15:14:05: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
winston:file written true false
winston:file logged 109 info  2020-01-02 15:14:05: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-01-02.15-14-05'

winston:file logged 212 debug 2020-01-02 15:14:05: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)

winston:file ENOENT ok /opt/zigbee2mqtt/data/log/2020-01-02.15-14-05/log.txt
winston:file stat done: log.txt { size: 0 }
winston:file create stream start /opt/zigbee2mqtt/data/log/2020-01-02.15-14-05/log.txt { flags: 'a' }
winston:file create stream ok /opt/zigbee2mqtt/data/log/2020-01-02.15-14-05/log.txt
winston:file file open ok /opt/zigbee2mqtt/data/log/2020-01-02.15-14-05/log.txt
zigbee2mqtt:info  2020-01-02 15:14:05: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
winston:file written true false
zigbee2mqtt:info  2020-01-02 15:14:05: Starting zigbee-herdsman...
winston:file written true false
zigbee2mqtt:debug 2020-01-02 15:14:05: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyUSB0"}}'
winston:file written true false
  serialport/bindings loading LinuxBinding +0ms
  serialport/stream .list +0ms
winston:file logged 80 info  2020-01-02 15:14:05: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)

winston:file logged 135 info  2020-01-02 15:14:05: Starting zigbee-herdsman...

winston:file logged 564 debug 2020-01-02 15:14:05: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyUSB0"}}'

  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyUSB0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  serialport/stream opening path: /dev/ttyUSB0 +106ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +1ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyUSB0 +2ms
  serialport/stream _read reading { start: 0, toRead: 65536 } +0ms
  serialport/binding-abstract read +2ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +5ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  serialport/stream _write 1 bytes of data +1ms
  serialport/binding-abstract write 1 bytes +1ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixWrite write returned: wrote 1 bytes +2ms
  serialport/bindings/unixWrite Finished writing 1 bytes +0ms
  serialport/stream binding.write write finished +3ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +2ms
  serialport/bindings/poller Polling for "readable" +6ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +0ms
[... 41 lines cut ...]
10 silly lifecycle zigbee2mqtt@1.8.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.8.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.8.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.8.0 start: `node index.js`
13 verbose stack Exit status 1
[... cut ...]
14 verbose pkgid zigbee2mqtt@1.8.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.15.0-72-generic
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.16.2
19 verbose npm  v6.9.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.8.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.8.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

pcb1962, 2020-01-02:
Is your zigbee stick really at /dev/ttyUSB0 ?
It's more commonly found at /dev/ttyACM0 or similar.

nickrout, 2020-01-02:
Yes it is at /dev/ttyUSB0.

skoefel, 2020-01-03:
Im having the same issue

Kryzek, 2020-01-03:
You can use `port: null` to detect zigbee device automatically instead of `port: /dev/...`. This should solve the issue if the problem is due wrong port entry.

skoefel, 2020-01-03:
The port is not the issue. Ive had it working, after an update, the error came.
But to make sure, ive checked with port: null, but still the same error.

renewoensdregt, 2020-01-03:
your issue sounds exactly the same as I have reported earlier too. Same errors and same antenna and config.
https://github.com/Koenkk/zigbee2mqtt/issues/2591#issue-541669466

Normaniac, 2020-01-08:
I also have this issue with the same log output. 
I followed the install instruction (https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html) line by line on a fresh Buster Lite on a Pi 4 AND on a Pi 3. Zigbee-Stick on /dev/ttyACM0 is available.

"Fresh" means: Clean install, just updated and upgraded.

renewoensdregt, 2020-01-08:
Have a look at my issue above (2591) which was finally resolved by setting disable_led to false in the configuration. Might make sense as the CC2530 does not have a led...

Normaniac, 2020-01-08:
Thank you @renewoensdregt, I have seen your issue and tried that, but it didn't work for me.

Koenkk, 2020-01-09:
This issue means that there is something wrong with the connection to the CC253X, you can try:
- Make sure the adapter is connected OK
- Reflash firmware
- Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

closed 2020-03-16 by stale[bot]