Koenkk/zigbee2mqtt #2655: Interview failed for ShenZhen_Homa HOMA1001

kalsan, 2020-01-02:
# Bug Report
## What happened
I'm following the "How to support new devices" page for supporting the Dresden Elektronik 4in1 Dimmer 12-24V, which shows up as "ShenZhen_Homa HOMA1001" in the network map (probably it's the same device).

## What did you expect to happen
I expected zigbee2mqtt to report a successful interview followed by the message "unknown device". However the log is as follows:
```
zigbee2mqtt_1    | zigbee2mqtt:info  2020-01-02T10:17:47: Device '0x00124b001eb08d44' joined
zigbee2mqtt_1    | zigbee2mqtt:info  2020-01-02T10:17:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b001eb08d44"}}'
zigbee2mqtt_1    | zigbee2mqtt:info  2020-01-02T10:17:47: Starting interview of '0x00124b001eb08d44'
zigbee2mqtt_1    | zigbee2mqtt:info  2020-01-02T10:17:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001eb08d44"}}'
zigbee2mqtt_1    | zigbee2mqtt:error 2020-01-02T10:17:58: Failed to interview '0x00124b001eb08d44', device has not successfully been paired
zigbee2mqtt_1    | zigbee2mqtt:info  2020-01-02T10:17:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00124b001eb08d44"}}'
```

In the graphviz network map, the device sometimes shows up as follows:
[image]

However, after re-pairing the device by resetting it (turning the power on and off repetitively), the device shows up as follows:
[image]

I'm not sure what is the issue here and what steps I should take in order to attempt a resolution.

## How to reproduce it (minimal and precise)

Reset the dimmer by turning it on and off repeditively, set permit_join to true.

## Debug Info
zigbee2mqtt version: docker latest
CC253X firmware version: November 2019

kalsan, 2020-01-17:
Hmm any thoughts about this? Is more info needed?

Koenkk, 2020-01-17:
Can you provide the herdsman debug logging while pairing this device? First make sure you are on zigbee2mqtt 1.9.0

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

kalsan, 2020-01-18:
Hi and thanks for your reply. I performed the following steps:

- Upgrade the docker image to 1.9.0
- Change the run script to include herdsman log as described
- Allow device joining
- Restart zigbee2mqtt
- Power cycle the HOMA 5x in order to reset it

The log is below:

```
info  2020-01-18 09:05:06: Logging to console and directory: '/app/data/log/2020-01-18.09-05-06'
debug 2020-01-18 09:05:06: Loaded state from file /app/data/state.json
info  2020-01-18 09:05:06: Starting zigbee2mqtt version 1.9.0 (commit #287f843)
info  2020-01-18 09:05:06: Starting zigbee-herdsman...
debug 2020-01-18 09:05:06: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
info  2020-01-18 09:05:07: zigbee-herdsman started
info  2020-01-18 09:05:07: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
debug 2020-01-18 09:05:07: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
info  2020-01-18 09:05:07: Currently 25 devices are joined:
info  2020-01-18 09:05:07: 0x14b457fffe7447ec (0x14b457fffe7447ec): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
info  2020-01-18 09:05:07: 0x14b457fffe7c91e3 (0x14b457fffe7c91e3): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
info  2020-01-18 09:05:07: 0x14b457fffe2b06a8 (0x14b457fffe2b06a8): L1527 - IKEA FLOALT LED light panel, dimmable, white spectrum (30x30 cm) (Router)
info  2020-01-18 09:05:07: 0x14b457fffe47a1fe (0x14b457fffe47a1fe): L1527 - IKEA FLOALT LED light panel, dimmable, white spectrum (30x30 cm) (Router)
info  2020-01-18 09:05:07: 0x14b457fffe65eb77 (0x14b457fffe65eb77): E1744 - IKEA SYMFONISK sound controller (EndDevice)
info  2020-01-18 09:05:07: 0x14b457fffe4aadef (0x14b457fffe4aadef): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
info  2020-01-18 09:05:07: 0x14b457fffe4dc869 (0x14b457fffe4dc869): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
info  2020-01-18 09:05:07: 0x14b457fffe74fc02 (0x14b457fffe74fc02): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
info  2020-01-18 09:05:07: 0x14b457fffe7b4856 (0x14b457fffe7b4856): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
info  2020-01-18 09:05:07: 0x000d6ffffe9c56ed (0x000d6ffffe9c56ed): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
info  2020-01-18 09:05:07: 0x14b457fffe4fa91c (0x14b457fffe4fa91c): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
info  2020-01-18 09:05:07: 0x680ae2fffe2a67bf (0x680ae2fffe2a67bf): LED1733G7 - IKEA TRADFRI LED bulb E14 600 lumen, dimmable, white spectrum, opal white (Router)
info  2020-01-18 09:05:07: 0xec1bbdfffede2176 (0xec1bbdfffede2176): LED1733G7 - IKEA TRADFRI LED bulb E14 600 lumen, dimmable, white spectrum, opal white (Router)
info  2020-01-18 09:05:07: 0x14b457fffe744143 (0x14b457fffe744143): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
info  2020-01-18 09:05:07: 0xccccccfffe37b493 (0xccccccfffe37b493): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
info  2020-01-18 09:05:07: 0x14b457fffe46be5f (0x14b457fffe46be5f): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
info  2020-01-18 09:05:07: 0xccccccfffe39f77b (0xccccccfffe39f77b): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
info  2020-01-18 09:05:07: 0xccccccfffe6cec9b (0xccccccfffe6cec9b): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
info  2020-01-18 09:05:07: 0xccccccfffe56adfa (0xccccccfffe56adfa): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
info  2020-01-18 09:05:07: 0x14b457fffe41a0fd (0x14b457fffe41a0fd): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
info  2020-01-18 09:05:07: 0x14b457fffe7c75b1 (0x14b457fffe7c75b1): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
info  2020-01-18 09:05:07: 0x14b457fffe7b2878 (0x14b457fffe7b2878): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
info  2020-01-18 09:05:07: 0x0017880106a7d604 (0x0017880106a7d604): 324131092621 - Philips Hue dimmer switch (EndDevice)
info  2020-01-18 09:05:07: 0x0017880106344733 (0x0017880106344733): 324131092621 - Philips Hue dimmer switch (EndDevice)
info  2020-01-18 09:05:07: 0xec1bbdfffe1bb0e6 (0xec1bbdfffe1bb0e6): ICPSHC24-30EU-IL-1 - IKEA TRADFRI driver for wireless control (30 watt) (Router)
warn  2020-01-18 09:05:07: `permit_join` set to  `true` in configuration.yaml.
warn  2020-01-18 09:05:07: Allowing new devices to join.
warn  2020-01-18 09:05:07: Set `permit_join` to `false` once you joined all devices.
info  2020-01-18 09:05:07: Zigbee: allowing new devices to join.
info  2020-01-18 09:05:07: Connecting to MQTT server at mqtt://localhost
info  2020-01-18 09:05:07: Connected to MQTT server
[... cut ...]
info  2020-01-18 09:05:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001eb08d44"}}'
debug 2020-01-18 09:05:29: Device '0x00124b001eb08d44' announced itself
debug 2020-01-18 09:05:29: Device '0x00124b001eb08d44' announced itself
error 2020-01-18 09:05:40: Failed to interview '0x00124b001eb08d44', device has not successfully been paired
info  2020-01-18 09:05:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00124b001eb08d44"}}'
```

kalsan, 2020-01-18:
Full log by redirecting stdout:

```
Attaching to smarthome-s9_zigbee2mqtt_1
[36mzigbee2mqtt_1    |[0m Using '/app/data' as data directory
[36mzigbee2mqtt_1    |[0m 
[36mzigbee2mqtt_1    |[0m > zigbee2mqtt@1.9.0 start /app
[36mzigbee2mqtt_1    |[0m > node index.js
[36mzigbee2mqtt_1    |[0m 
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:18: Logging to console and directory: '/app/data/log/2020-01-18.09-00-18'
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:18: Starting zigbee2mqtt version 1.9.0 (commit #287f843)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:18: Starting zigbee-herdsman...
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: zigbee-herdsman started
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: Currently 26 devices are joined:
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe7447ec (0x14b457fffe7447ec): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe7c91e3 (0x14b457fffe7c91e3): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe2b06a8 (0x14b457fffe2b06a8): L1527 - IKEA FLOALT LED light panel, dimmable, white spectrum (30x30 cm) (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe47a1fe (0x14b457fffe47a1fe): L1527 - IKEA FLOALT LED light panel, dimmable, white spectrum (30x30 cm) (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe65eb77 (0x14b457fffe65eb77): E1744 - IKEA SYMFONISK sound controller (EndDevice)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe4aadef (0x14b457fffe4aadef): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe4dc869 (0x14b457fffe4dc869): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe74fc02 (0x14b457fffe74fc02): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe7b4856 (0x14b457fffe7b4856): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x000d6ffffe9c56ed (0x000d6ffffe9c56ed): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe4fa91c (0x14b457fffe4fa91c): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x680ae2fffe2a67bf (0x680ae2fffe2a67bf): LED1733G7 - IKEA TRADFRI LED bulb E14 600 lumen, dimmable, white spectrum, opal white (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0xec1bbdfffede2176 (0xec1bbdfffede2176): LED1733G7 - IKEA TRADFRI LED bulb E14 600 lumen, dimmable, white spectrum, opal white (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe744143 (0x14b457fffe744143): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0xccccccfffe37b493 (0xccccccfffe37b493): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe46be5f (0x14b457fffe46be5f): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0xccccccfffe39f77b (0xccccccfffe39f77b): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0xccccccfffe6cec9b (0xccccccfffe6cec9b): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0xccccccfffe56adfa (0xccccccfffe56adfa): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe41a0fd (0x14b457fffe41a0fd): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe7c75b1 (0x14b457fffe7c75b1): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:00:19: 0x14b457fffe7b2878 (0x14b457fffe7b2878): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
[... cut ...] d6ffffe9c56ed_linkquality","unique_id":"0x000d6ffffe9c56ed_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6ffffe9c56ed"],"name":"0x000d6ffffe9c56ed","sw_version":"Zigbee2mqtt 1.9.0","model":"TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (LED1537R6)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
[36mzigbee2mqtt_1    |[0m [32mzigbee2mqtt:info [39m 2020-01-18 09:05:08: MQTT publish: topic 'homeassistant/sensor/0x14b457fffe4fa91c/action/co

Koenkk, 2020-01-18:
`Cannot read property 'updateLastSeen' of undefined` should be fixed in the latest dev branch, please try again.

kalsan, 2020-01-22:
Indeed, the interview is now successful (using latest-dev docker image). I'll now try to figure out how to find out the parameters this device needs in order to function.
Thank you so much for fixing this issue! I can't express how much I love zigbee2mqtt.

kalsan, 2020-01-22:
reference for other HOMA owners, the log should now be:
```
zigbee2mqtt_1    | zigbee2mqtt:info  2020-01-22 20:12:25: Device '0x00124b001eb08d44' joined
zigbee2mqtt_1    | zigbee2mqtt:info  2020-01-22 20:12:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b001eb08d44"}}'
zigbee2mqtt_1    | zigbee2mqtt:info  2020-01-22 20:12:25: Starting interview of '0x00124b001eb08d44'
zigbee2mqtt_1    | zigbee2mqtt:info  2020-01-22 20:12:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001eb08d44"}}'
zigbee2mqtt_1    | zigbee2mqtt:warn  2020-01-22 20:12:36: Received message from unsupported device with Zigbee model 'HOMA1001'
zigbee2mqtt_1    | zigbee2mqtt:warn  2020-01-22 20:12:36: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt_1    | zigbee2mqtt:warn  2020-01-22 20:12:36: Received message from unsupported device with Zigbee model 'HOMA1001'
zigbee2mqtt_1    | zigbee2mqtt:warn  2020-01-22 20:12:36: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt_1    | zigbee2mqtt:warn  2020-01-22 20:12:36: Received message from unsupported device with Zigbee model 'HOMA1001'
zigbee2mqtt_1    | zigbee2mqtt:warn  2020-01-22 20:12:36: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt_1    | zigbee2mqtt:warn  2020-01-22 20:12:36: Received message from unsupported device with Zigbee model 'HOMA1001'
zigbee2mqtt_1    | zigbee2mqtt:warn  2020-01-22 20:12:36: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt_1    | zigbee2mqtt:warn  2020-01-22 20:12:36: Received message from unsupported device with Zigbee model 'HOMA1001'
zigbee2mqtt_1    | zigbee2mqtt:warn  2020-01-22 20:12:36: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt_1    | zigbee2mqtt:info  2020-01-22 20:12:36: Successfully interviewed '0x00124b001eb08d44', device has successfully been paired
zigbee2mqtt_1    | zigbee2mqtt:warn  2020-01-22 20:12:36: Device '0x00124b001eb08d44' with Zigbee model 'HOMA1001' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt_1    | zigbee2mqtt:info  2020-01-22 20:12:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001eb08d44","supported":false}}'
```

closed 2020-01-22 by kalsan