Koenkk/zigbee2mqtt #2664: Device doesn't have "model" attribute, skipped

Furiousz23, 2020-01-03:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

Today i tried to pair a Ikea Tradfri LED1837R5. 

This is the log:
`Jan 03 19:30:23 raspberrypi npm[527]: zigbee2mqtt:info  2020-01-03 19:30:23: Starting interview of '0xec1bbdfffe94a2fa'
Jan 03 19:30:23 raspberrypi npm[527]: zigbee2mqtt:info  2020-01-03 19:30:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xec1bbdfffe94a2fa"}}'
Jan 03 19:30:23 raspberrypi npm[527]: zigbee2mqtt:info  2020-01-03 19:30:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b0018ed2449","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20180815","lastSeen":1578076223896},{"ieeeAddr":"0x00158d0001e7fb79","type":"EndDevice","networkAddress":17533,"model":"WXKG01LM","friendly_name":"0x00158d0001e7fb79","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_switch","lastSeen":null},{"ieeeAddr":"0x00158d0001e75987","type":"EndDevice","networkAddress":872,"model":"WXKG01LM","friendly_name":"0x00158d0001e75987","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_switch","lastSeen":null},{"ieeeAddr":"0x00158d000120d664","type":"EndDevice","networkAddress":55688,"model":"WXKG01LM","friendly_name":"0x00158d000120d664","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_switch","lastSeen":null},{"ieeeAddr":"0x00158d0001b97b5c","type":"EndDevice","networkAddress":18060,"model":"MCCGQ11LM","friendly_name":"0x00158d0001b97b5c","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_magnet.aq2","lastSeen":null},{"ieeeAddr":"0x00158d000232cf51","type":"EndDevice","networkAddress":56446,"model":"MCCGQ11LM","friendly_name":"0x00158d000232cf51","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_magnet.aq2","lastSeen":1578076094704},{"ieeeAddr":"0xd0cf5efffe6f6883","type":"Router","networkAddress":15081,"model":"E1603/E1702","friendly_name":"0xd0cf5efffe6f6883","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI control outlet","lastSeen":null},{"ieeeAddr":"0xd0cf5efffeea8d28","type":"Router","networkAddress":34509,"model":"E1603/E1702","friendly_name":"0xd0cf5efffeea8d28","manufacturerID":4476,"manufacturerName":"IKEA of Swed
Jan 03 19:30:23 raspberrypi npm[527]: en","powerSource":"Mains (single phase)","modelID":"TRADFRI control outlet","lastSeen":null},{"ieeeAddr":"0x0017880104ad357b","type":"EndDevice","networkAddress":14573,"model":"324131092621","friendly_name":"0x0017880104ad357b","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Battery","modelID":"RWL021","lastSeen":null},{"ieeeAddr":"0x086bd7fffe4057f5","type":"Router","networkAddress":35702,"model":"ICPSHC24-10EU-IL-1","friendly_name":"0x086bd7fffe4057f5","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI Driver 10W","lastSeen":null},{"ieeeAddr":"0x14b457fffef68f2c","type":"Router","networkAddress":19001,"model":"LED1624G9","friendly_name":"0x14b457fffef68f2c","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E27 CWS opal 600lm","lastSeen":null},{"ieeeAddr":"0xec1bbdfffe9cb750","type":"Router","networkAddress":20123,"model":"LED1837R5","friendly_name":"0xec1bbdfffe9cb750","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb GU10 WW 400lm","lastSeen":null},{"ieeeAddr":"0xec1bbdfffe94a2fa","networkAddress":17239,"friendly_name":"0xec1bbdfffe94a2fa","lastSeen":1578076223833}]}'
Jan 03 19:30:34 raspberrypi npm[527]: zigbee2mqtt:info  2020-01-03 19:30:34: Successfully interviewed '0xec1bbdfffe94a2fa', device has successfully been paired
Jan 03 19:30:34 raspberrypi npm[527]: zigbee2mqtt:info  2020-01-03 19:30:34: Device '0xec1bbdfffe94a2fa' is supported, identified as: IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1837R5)
Jan 03 19:30:34 raspberrypi npm[527]: zigbee2mqtt:info  2020-01-03 19:30:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xec1bbdfffe94a2fa","model":"LED1837R5","vendor":"IKEA","description":"TRADFRI LED bulb GU10 400 lumen, dimmable","supported":true}}'
`

githubos1, 2020-01-05:
Having the same issue....  from what I found out it that the device, in my case a heater thermostate
 valve from "Essentials" is not compatible at the moment..  i.e.  it's not in the zigbee2mqtt database...
see: https://www.zigbee2mqtt.io/information/supported_devices.html

tried to add a new entrance by copy and paste of an existing "similar" device already in the database and adding manually (zibeeModel, model, vendor, etc). which actually fixed the above error, but unfortunately the device although paired and correctly identified in "/opt/zigbee2mqtt/data/configuration.yaml " - even data is transmitted to/from Domoticz
#############################
 2020-01-05 16:50:42.833 (ZIGGY) Sending 2 bytes of data
2020-01-05 16:50:42.833 (ZIGGY) c0 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. �.
2020-01-05 16:50:42.833 (ZIGGY) Received 2 bytes of data
2020-01-05 16:50:42.834 (ZIGGY) d0 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. �.
2020-01-05 16:50:52.803 (ZIGGY) MqttClient::ping
2020-01-05 16:50:52.803 (ZIGGY) Sending 2 bytes of data
2020-01-05 16:50:52.803 (ZIGGY) c0 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. �.
2020-01-05 16:50:52.804 (ZIGGY) Received 2 bytes of data
2020-01-05 16:50:52.804 (ZIGGY) d0 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. �.
2020-01-05 16:51:02.827 (ZIGGY) MqttClient::ping
2020-01-05 16:51:02.827 (ZIGGY) Sending 2 bytes of data
2020-01-05 16:51:02.828 (ZIGGY) c0 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. �.
2020-01-05 16:51:02.828 (ZIGGY) Received 2 bytes of data
2020-01-05 16:51:02.829 (ZIGGY) d0 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. �.
2020-01-05 16:51:12.849 (ZIGGY) MqttClient::ping
2020-01-05 16:51:12.849 (ZIGGY) Sending 2 bytes of data
2020-01-05 16:51:12.850 (ZIGGY) c0 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. �. 
#####################################

I can not add or find the device in Domoticz :-(((

I'm lost..

madefordeveloping, 2020-01-20:
Same problem here.
After everything was working before, I upgraded my raspberry pi and installed everything as usual.
Unfortunately i am getting exactly the same problem as TS.

`Jan 20 03:24:48 raspberrypi npm[491]: zigbee2mqtt:info  2020-01-20 03:24:48: Successfully interviewed '0x90fd9ffffe6fb9d0', device has successfully b                                                                                                        een paired
Jan 20 03:24:48 raspberrypi npm[491]: zigbee2mqtt:info  2020-01-20 03:24:48: Device '0x90fd9ffffe6fb9d0' is supported, identified as: IKEA TRADFRI mo                                                                                                        tion sensor (E1525/E1745)
Jan 20 03:24:48 raspberrypi npm[491]: zigbee2mqtt:info  2020-01-20 03:24:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing"                                                                                                        ,"message":"interview_successful","meta":{"friendly_name":"0x90fd9ffffe6fb9d0","model":"E1525/E1745","vendor":"IKEA","description":"TRADFRI motion se                                                                                                        nsor","supported":true}}'
Jan 20 03:24:48 raspberrypi npm[491]: zigbee2mqtt:info  2020-01-20 03:24:48: Configuring '0x90fd9ffffe6fb9d0'
Jan 20 03:24:54 raspberrypi npm[491]: zigbee2mqtt:info  2020-01-20 03:24:54: Successfully configured '0x90fd9ffffe6fb9d0'
`

But in Domoticz:
`2020-01-20 03:24:22.040 (Zigbee2MQTT) Received available devices list from bridge
2020-01-20 03:24:22.040 (Zigbee2MQTT) Device SJCGQ11LM 0x00158d00044bdbf4 (0x00158d00044bdbf4)
2020-01-20 03:24:22.041 (Zigbee2MQTT) Device SJCGQ11LM 0x00158d00044bdae7 (0x00158d00044bdae7)
2020-01-20 03:24:22.041 (Zigbee2MQTT) Device E1525/E1745 0x14b457fffe4eee94 (0x14b457fffe4eee94)
2020-01-20 03:24:22.041 (Zigbee2MQTT) Device E1525/E1745 0x90fd9ffffe6dd5ff (0x90fd9ffffe6dd5ff)
2020-01-20 03:24:22.041 (Zigbee2MQTT) Device 0x90fd9ffffe6fb9d0 (0x90fd9ffffe6fb9d0) doesn't have "model" attribute, skipped`

This isn't just happening with Device 0x90fd9ffffe6fb9d0.
I have more devices with this problem.

closed 2020-03-27 by stale[bot]

microchip1974, 2020-03-29:
Same problem here with Eurotronics Thermostatic valves Domoticz Stable RPI4
2020-03-29 10:28:39.443 (Zigbee2Mqtt) Device 0x00158d0001ff8d2d (0x00158d0001ff8d2d) doesn't have "model" attribute, skipped
2020-03-29 10:28:39.443 (Zigbee2Mqtt) Device 0x00158d00032f7cad (0x00158d00032f7cad) doesn't have "model" attribute, skipped