Koenkk/zigbee2mqtt #2667: brightness_move won't return current brightness for Tradfri bulb

merlinschumacher, 2020-01-03:
## What happened
I used brightness_move on a tradfri bulb to gradually change its brightness. after the stop command the mqtt topic still reported the previous brightness value. 

## What did you expect to happen
i expected the mqtt topic to reflect the current brightness value of the bulb.

## How to reproduce it (minimal and precise)
set a brightness by number for a tradfri bulb. trigger a brightness move. stop it.

## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version: 20190619 source routing

sti0, 2020-01-10:
The brightness of my Hue lights won't get updated either.

Koenkk, 2020-01-13:
Will be fixed in the latest dev branch in a few hours, thanks!

closed 2020-01-13 by Koenkk