Koenkk/zigbee2mqtt #2670: not able to connect TS0201

arovak, 2020-01-04:
## What happened
When pairing a Zigbee 3.0 device TS0201 it does not register in zigbee2mqtt. No log output is shown of a device attempting to connect.
Device bought on Aliexpress from this link.
[https://nl.aliexpress.com/item/4000420210948.html](url)

## What did you expect to happen
Device being recognized or at least some log activity of a device trying to pair.

## How to reproduce it (minimal and precise)
Put bridge in permit_join mode and press button on device for 5 sec to start pairing.

## Debug Info
zigbee2mqtt version: 1.8.0 commit da4d26a
CC253X firmware version:zStack30x revision 20190425

closed 2020-03-11 by stale[bot]

jafd, 2020-06-09:
JFYI you just need to hold the button longer.

lug-gh, 2020-06-29:
> JFYI you just need to hold the button longer.

Longer than what? The manual says 5-10 seconds for pairing, pressing more than 10 seconds will cancel the pairing...

jafd, 2020-06-30:
Closer to 10 seconds than to 5, obviously. Also AFAIR I plain disregarded what the manual said about holding longer than 10 seconds and held the button until it freakin' paired.

Can't you just experiment and see for yourself? Are you counting those seconds with a real timer? When you have to keep holding that tiny button, the seconds in your mind seem to _crawl_. I can tell.

lug-gh, 2020-06-30:
No need to be snippy, and if you could write something more formally I might understand what you want to say, as it is now it makes no sense as a non-native speaker, even with a translator.

Of course I tried it myself, but since it didn't work for ages, I looked around on Github to see if there were any tricks. So I asked you how much "longer" is. After 30-40 minutes, the pairing finally worked for me, without me doing anything different. Keep the button pressed for about 7 seconds, then it flashes for 20 seconds.

holser, 2021-02-07:
Hi,

I had the similar issue with TS0201 and cc2538:zStack30x:coordinator revision:20200327. However I was able to analyze it a little bit further. When I have coordinator and TS0201 I was not able to pair it no matter what I did. When I connected router device (LED1903C5 in my case) I was able to add TS0201 to the system immediately. Though it was connected through the router. It means when I shut down the bulb TS0201 was disconnected without any attempts to connect to my cc2538 zigbee stick. I guess there is something in cc2538 firmware that doesn't allow to pair TS0201 directly. Hope this helps