Koenkk/zigbee2mqtt #2671: Can't (re)start z2m after turning the coordinator into a router

mr2c12, 2020-01-04:
# Bug Report
## What happened

I had a CC2530 used as a router (ieeeAddr '0x00124b001bb85676').
Today I flashed this board with a router firmware (and got a new coordinator based onCC2530+CC2591).

At first, I run z2m with the new coordinator, everything went fine (all my existing devices connected, as expected).
Then, I associate the "new router" to the existing network and I got:

> info  2020-01-04 15:06:12: Starting interview of '0x00124b001bb85676'
> info  2020-01-04 15:06:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001bb85676"}}'
> info  2020-01-04 15:06:14: Successfully interviewed '0x00124b001bb85676', device has successfully been paired
> info  2020-01-04 15:06:14: Device '0x00124b001bb85676' is supported, identified as: Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)
> info  2020-01-04 15:06:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001bb85676","model":"CC2530.ROUTER","vendor":"Custom devices (DiY)","description":"[CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/)","supported":true}}'

The new router has been identified and worked as expected.
Please note: at this point I was unable to get a complete map of my devices. The new coordinator didn't show up in the network map, only the new router (aka old coordinator).

From this moment, after I stopped z2m, I am unable to start it again since I get:

> > zigbee2mqtt@1.8.0 start /opt/zigbee2mqtt
> > node index.js
> 
> zigbee2mqtt:info  2020-01-04 15:54:58: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-01-04.15-54-58'
> zigbee2mqtt:info  2020-01-04 15:54:59: Starting zigbee2mqtt version 1.8.0 (commit #unknown)
> zigbee2mqtt:info  2020-01-04 15:54:59: Starting zigbee-herdsman...
> zigbee2mqtt:error 2020-01-04 15:55:01: Error while starting zigbee-herdsman
> zigbee2mqtt:error 2020-01-04 15:55:01: Failed to start zigbee
> zigbee2mqtt:error 2020-01-04 15:55:01: Exiting...
> zigbee2mqtt:error 2020-01-04 15:55:01: Error: Device with ieeeAddr '0x00124b001bb85676' already exists
>     at Function.create (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:159:19)
>     at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:118:32)
>     at Generator.next (<anonymous>)
>     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:5:58)

There seems to be a problem related to the device with ieee address 0x00124b001bb85676, the one that used to be the coordinator but now is a router.

## What did you expect to happen

Be able to start z2m again (without re-pairing every device - finger crossed).

## How to reproduce it (minimal and precise)

1. Use CC2530#1 as coordinator.
2. Stop z2m.
3. Disconnect CC2530#1. Use CC2530#2 as coordinator.
4. Start z2m.
5. Flash CC2530#1 with router firmware.
6. Pair CC2530#1 to the current coordinator.
7. Stop z2m.
8. Start z2m.. error!

## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version: 20190523

What can I do? I already tried to remove the entry for 0x00124b001bb85676 in the devices: section of configuration.yaml, but it's uneffective.

mr2c12, 2020-01-05:
Given that:

- CC2530#1 is the new router, previously used as coordinator
- CC2530#2 is the new coordinator

I was able to start z2m again setting ieaddr of coordinator to blank in database.db, it was set to the same ieaddr of CC2530#1

Then I tried to retrieve the ieaddr of CC2530#2 to put it into the database.db: I removed the data/ folder, create a new configuration.yaml and then startedz2m. But I got for CC2530#2 the same ieaddr of CC2530#1!! How is that possible?

At the moment, z2m is up and running but I have one ikea bulb that does not connect. I will probably try to re-pair everything in the next few days but a bit worried about this problem with ieaddr for CC2530#1 and CC2530#2..

mr2c12, 2020-01-06:
Ok, I started from zero.

1. Start z2m using CC2530#2 as coordinator (as it should be)
2. Pair CC2530#1 (the old coordinator, now a router)
3. Pair an Ikea router
4. Stop z2m
5. Start z2m fails with error:

> $ npm start
> 
> > zigbee2mqtt@1.8.0 start /opt/zigbee2mqtt
> > node index.js
> 
> zigbee2mqtt:info  2020-01-06 10:47:10: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-01-06.10-47-10'
> zigbee2mqtt:debug 2020-01-06 10:47:10: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-01-05.15-39-52'
> zigbee2mqtt:debug 2020-01-06 10:47:11: Loaded state from file /opt/zigbee2mqtt/data/state.json
> zigbee2mqtt:info  2020-01-06 10:47:11: Starting zigbee2mqtt version 1.8.0 (commit #unknown)
> zigbee2mqtt:info  2020-01-06 10:47:11: Starting zigbee-herdsman...
> zigbee2mqtt:debug 2020-01-06 10:47:11: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0"}}'
> zigbee2mqtt:error 2020-01-06 10:47:13: Error while starting zigbee-herdsman
> zigbee2mqtt:error 2020-01-06 10:47:13: Failed to start zigbee
> zigbee2mqtt:error 2020-01-06 10:47:13: Exiting...
> zigbee2mqtt:error 2020-01-06 10:47:13: Error: Device with ieeeAddr '0x00124b001bb85676' already exists
>     at Function.create (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:159:19)
>     at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:118:32)
>     at Generator.next (<anonymous>)
>     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:5:58)
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! zigbee2mqtt@1.8.0 start: `node index.js`
> npm ERR! Exit status 1
> npm ERR! 
> npm ERR! Failed at the zigbee2mqtt@1.8.0 start script.
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /home/orangepi/.npm/_logs/2020-01-06T09_47_13_641Z-debug.log

Attached the newly created configuration and database files.

[configuration.yaml.txt](https://github.com/Koenkk/zigbee2mqtt/files/4025549/configuration.yaml.txt)
[database.db.txt](https://github.com/Koenkk/zigbee2mqtt/files/4025550/database.db.txt)

mr2c12, 2020-01-06:
Quick addition: if I start z2m from a fresh installation and enabling the debug I get:

> $ npm start
> 
> > zigbee2mqtt@1.8.0 start /opt/zigbee2mqtt
> > node index.js
> 
> zigbee2mqtt:info  2020-01-06 10:57:20: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-01-06.10-57-20'
> zigbee2mqtt:debug 2020-01-06 10:57:20: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-01-05.16-11-18'
> zigbee2mqtt:debug 2020-01-06 10:57:21: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
> zigbee2mqtt:info  2020-01-06 10:57:21: Starting zigbee2mqtt version 1.8.0 (commit #unknown)
> zigbee2mqtt:info  2020-01-06 10:57:21: Starting zigbee-herdsman...
> zigbee2mqtt:debug 2020-01-06 10:57:21: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,204,221,221,204,204,221,204],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0"}}'
> zigbee2mqtt:info  2020-01-06 10:57:31: zigbee-herdsman started
> zigbee2mqtt:info  2020-01-06 10:57:31: Coordinator firmware version: '{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190523}}'
> zigbee2mqtt:debug 2020-01-06 10:57:31: Zigbee network parameters: {"panID":6755,"extendedPanID":"0x00124b001bb85676","channel":11}
> zigbee2mqtt:info  2020-01-06 10:57:31: Currently 0 devices are joined:
> zigbee2mqtt:warn  2020-01-06 10:57:31: `permit_join` set to  `true` in configuration.yaml.
> zigbee2mqtt:warn  2020-01-06 10:57:31: Allowing new devices to join.
> zigbee2mqtt:warn  2020-01-06 10:57:31: Set `permit_join` to `false` once you joined all devices.
> zigbee2mqtt:info  2020-01-06 10:57:31: Zigbee: allowing new devices to join.
> zigbee2mqtt:info  2020-01-06 10:57:31: Connecting to MQTT server at mqtt://192.168.1.6
> zigbee2mqtt:info  2020-01-06 10:57:31: Connected to MQTT server
> zigbee2mqtt:info  2020-01-06 10:57:31: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
> zigbee2mqtt:info  2020-01-06 10:57:31: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.8.0","commit":"unknown","coordinator":{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190523}},"log_level":"debug","permit_join":true}'

It seems really that my router and my coordinator share the same ieaddr?

mr2c12, 2020-01-06:
Closing this since completely invalid.
Not sure how I fixed this, but I had to re-flash the coordinator and change its secondary address (was set to default to FF FF FF..).

Now everything seems to work. Not even related to zigbee2mqtt, sorry.

closed 2020-01-06 by mr2c12

davidjb, 2021-07-23:
I had a similar issue with trying to use a second CC2652R as a router alongside another CC2652R as the coordinator.  Both devices use the same Primary IEEE Address and so when the second CC2652R joined the network, zigbee2mqtt ended up replacing the coordinator's entry (`"id": 1` in `database.db`) with that of the router, and an entry added into `devices.yaml`.  The result was the same when trying to restart zigbee2mqtt: `Error: Device with ieeeAddr '0x...' already exists`.  I was able to rollback the changes to the config, and get z2m back working again.

For me, resolving it was a case of changing the router's secondary IEEE address, given the [hierarchy of address resolution](https://github.com/Koenkk/Z-Stack-firmware/issues/72#issuecomment-486300124). Setting its MAC address is possible via its boot loader script `./cc2538-bsl.py --ieee-address 00:12:4b:aa:bb:cc:dd:ee [...]`.