Koenkk/zigbee2mqtt #2672: editing fromZigbee.js

toremick, 2020-01-04:
First of all, sorry if this is the wrong place to ask.

Im running zigbee2mqtt under hassio. installed as hassio addon.
Trying to add a new remote from namron that i found, and im having trouble adding my own things to fromZigbee.js.

found out i needed a copy of devices.js under /share/zigbee2mqtt/ ( data_path), added my remote there. But the trouble is that when i try to edit "/zigbee2mqtt-1.8.0/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js" it disapears after a needed restart of zigbee2mqtt.

tried adding a copy of fromZigbee.js to /share/zigbee2mqtt folder, but it doesnt use it.

How am i supposed to be able to do the testing i need to make this work?

carldebilly, 2020-01-06:
A small tip: install the portainer add-on and use it to restart Z2M instead of using HASS.IO. It won't recreate the container each time.

toremick, 2020-01-06:
Hi Carl,I only restart the hassio addon zigbee2mqtt, not hassio.-Tore6. jan.

toremick, 2020-01-06:
Think i misunderstood you, will try!But what after å reboot or power failure?-Tore6. jan. 2020 17:19 skrev tore@eposten.net:Hi Carl,I only restart the hassio addon zigbee2mqtt, not hassio.-Tore6. jan.

carldebilly, 2020-01-06:
You'll loose it.  But it's very useful to "test" your changes.  Once it's working correctly, you can use this instead: https://github.com/danielwelch/hassio-zigbee2mqtt#adding-support-for-new-devices

toremick, 2020-01-06:
> A small tip: install the portainer add-on and use it to restart Z2M instead of using HASS.IO. It won't recreate the container each time.

tried looking, but cant find where in portainer i can restart z2m?

toremick, 2020-01-06:
Just tried a "docker restart  <name of container>  " but that doesnt load the new settings.

toremick, 2020-01-06:
forget it, installed zigbee2mqtt on ubuntu and running it away from docker/hassio

closed 2020-03-13 by stale[bot]